espressif/esptool #332: A typo

c1r2t3, 2018-07-25:
In file esptool.py line 2072  : print('\rWriting at..........
should be print('\nWriting at.......

projectgus, 2018-07-25:
Hi @c1r2t3,

Thanks for bringing this up. This is deliberate use of the line-feed character so that the "Writing..." line will keep overwriting itself on the same line, rather than producing many lines of output.

Here's an example of it working as expected: https://asciinema.org/a/QTCOOaKnAKYkJ2vuo6CZifi2a

Is there a platform where this doesn't work as expected?

(I have noticed it can cause some funny output when esptool is redirected to a file. It may be possible to disable it when output is not a TTY, although I don't know how this behaves on Windows.)

c1r2t3, 2018-07-26:
So sorry for not being more specific.
Linux machine
Arduino IDE 1.8.5
My problem is that, when compiling  there is a window that shows progress and this line is not overwritten and goes down without effectively auto scrolling .
After the "\n" it is scrolling and I can see what is going on.
Hope I made myself understood :)

radimkarnis, 2020-12-08:
I’m closing this issue because it has been inactive for a few months and it seems like it's not relevant anymore. Please reopen if further discussion is needed.

closed 2020-12-08 by radimkarnis