espressif/esptool #333: Having a problem to run esptool.py

amitkray31, 2018-07-26:
(Please delete any lines which don't apply.)

* Operating system:ubuntu-18.04
* Python version:Python 2.7.15rc1
* ESP hardware in use:esp8266-01

# Full esptool.py command line as run: 
sudo python esptool.py --port /dev/ttyUSB0  write_flash 0x00000 /home/akr/Downloads/ai-thinker-v1.1.1.bin

# Full output from esptool.py :
 esptool.py v2.5.1-dev
Serial port /dev/ttyUSB0
Traceback (most recent call last):
  File "esptool.py", line 2851, in <module>
    _main()
  File "esptool.py", line 2844, in _main
    main()
  File "esptool.py", line 2545, in main
    esp = ESPLoader.detect_chip(each_port, initial_baud, args.before, args.trace)
  File "esptool.py", line 244, in detect_chip
    detect_port = ESPLoader(port, baud, trace_enabled=trace_enabled)
  File "esptool.py", line 213, in __init__
    self._port = serial.serial_for_url(port)
  File "/home/akr/.local/lib/python2.7/site-packages/serial/__init__.py", line 88, in serial_for_url
    instance.open()
  File "/home/akr/.local/lib/python2.7/site-packages/serial/serialposix.py", line 286, in open
    self._update_dtr_state()
  File "/home/akr/.local/lib/python2.7/site-packages/serial/serialposix.py", line 634, in _update_dtr_state
    fcntl.ioctl(self.fd, TIOCMBIS, TIOCM_DTR_str)
IOError: [Errno 71] Protocol error

# What is the expected behaviour?
Connecting...
Erasing flash... 

Writing at 0x00000000... (0 %)

# Do you have any other information from investigating this?

# Is there any other information you can think of which will help us reproduce this problem?

amitkray31, 2018-07-26:
The interfacing chip is CH340G

amitkray31, 2018-07-26:
So, is it possible to flash esp8266?

amitkray31, 2018-07-26:
Ok, after manually resetting the chip it is working but facing some other problem
esptool.py v2.5.1-dev
Serial port /dev/ttyUSB1
Connecting....
Detecting chip type... ESP8266
Chip is ESP8266EX
Features: WiFi
MAC: 5c:cf:7f:69:21:cb
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Auto-detected Flash size: 1MB
Compressed 1044480 bytes to 204643...
Wrote 1044480 bytes (204643 compressed) at 0x00000000 in 22.0 seconds (effective 379.1 kbit/s)...

A fatal error occurred: Timed out waiting for packet header

projectgus, 2018-07-26:
This might be a bug due to the high compression ratio it's timing out waiting for the ESP to finish:

* Did the flash succeed despite the error?
* Try adding the `--no-compress` option after `write_flash`

amitkray31, 2018-07-27:
Yes, despite of the error message the flash has been successful.
Output:-
esptool.py v2.5.1-dev
Serial port /dev/ttyUSB0
Connecting...
Detecting chip type... ESP8266
Chip is ESP8266EX
Features: WiFi
MAC: xx:xx:xx:xx:xx:xx
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Auto-detected Flash size: 1MB
Compressed 1044480 bytes to 204643...
Wrote 1044480 bytes (204643 compressed) at 0x00000000 in 21.3 seconds (effective 392.1 kbit/s)...

A fatal error occurred: Timed out waiting for packet header

amitkray31, 2018-07-27:
Without compression the flash also succeeded.
**output(Without compression):-**

esptool.py v2.5.1-dev
Serial port /dev/ttyUSB0
Connecting........_
Detecting chip type... ESP8266
Chip is ESP8266EX
Features: WiFi
MAC: xx:xx:xx:xx:xx:xx
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Wrote 1048576 bytes at 0x00000000 in 92.6 seconds (90.6 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...

amitkray31, 2018-07-27:
Thanks a lot  you for your cooperation .

closed 2018-12-21 by projectgus