espressif/esptool #334: esptool.py serial test

XiaofanLii, 2018-07-28:
* Operating system:Mac OS 10.13
* Python version: Python 2.7.10
* ESP hardware in use:ESP8266

esptool.py --port /dev/tty.wchusbserial1410 --before no_reset --after no_reset chip_id

esptool.py v2.5.0
Serial port /dev/tty.wchusbserial1410
Connecting........_____....._____....._____....._____....._____.....
Traceback (most recent call last):
  File "/usr/local/bin/esptool.py", line 2849, in <module>
    _main()
  File "/usr/local/bin/esptool.py", line 2842, in _main
    main()
  File "/usr/local/bin/esptool.py", line 2543, in main
    esp = ESPLoader.detect_chip(each_port, initial_baud, args.before, args.trace)
  File "/usr/local/bin/esptool.py", line 245, in detect_chip
    detect_port.connect(connect_mode)
  File "/usr/local/bin/esptool.py", line 447, in connect
    last_error = self._connect_attempt(mode=mode, esp32r0_delay=True)
  File "/usr/local/bin/esptool.py", line 424, in _connect_attempt
    self.sync()
  File "/usr/local/bin/esptool.py", line 363, in sync
    timeout=SYNC_TIMEOUT)
  File "/usr/local/bin/esptool.py", line 327, in command
    self._port.timeout = saved_timeout
  File "/Library/Python/2.7/site-packages/serial/serialutil.py", line 368, in timeout
    self._reconfigure_port()
  File "/Library/Python/2.7/site-packages/serial/serialposix.py", line 326, in _reconfigure_port
    raise SerialException("Could not configure port: {}".format(msg))
serial.serialutil.SerialException: Could not configure port: (6, 'Device not configured')

if use  esptool.py --port /dev/tty.wchusbserial1410  chip_id
esptool.py v2.5.0
Serial port /dev/tty.wchusbserial1410
Connecting........_____.....
Traceback (most recent call last):
  File "/usr/local/bin/esptool.py", line 2849, in <module>
    _main()
  File "/usr/local/bin/esptool.py", line 2842, in _main
    main()
  File "/usr/local/bin/esptool.py", line 2543, in main
    esp = ESPLoader.detect_chip(each_port, initial_baud, args.before, args.trace)
  File "/usr/local/bin/esptool.py", line 245, in detect_chip
    detect_port.connect(connect_mode)
  File "/usr/local/bin/esptool.py", line 447, in connect
    last_error = self._connect_attempt(mode=mode, esp32r0_delay=True)
  File "/usr/local/bin/esptool.py", line 410, in _connect_attempt
    self._setDTR(True)   # IO0=LOW
  File "/usr/local/bin/esptool.py", line 368, in _setDTR
    self._port.setDTR(state)
  File "/Library/Python/2.7/site-packages/serial/serialutil.py", line 599, in setDTR
    self.dtr = value
  File "/Library/Python/2.7/site-packages/serial/serialutil.py", line 469, in dtr
    self._update_dtr_state()
  File "/Library/Python/2.7/site-packages/serial/serialposix.py", line 634, in _update_dtr_state
    fcntl.ioctl(self.fd, TIOCMBIS, TIOCM_DTR_str)
IOError: [Errno 6] Device not configured

And my usb chip is ch340

projectgus, 2018-07-31:
HI @XiaofanLii,

This is an operating system level error that indicates some problem with the driver talking to the chip. It may indicate a problem with the USB connection, the port is resetting or otherwise failing to communicate. Suggest double checking the port name is correct, and swapping the USB cable and port to see if either of these are faulty.

Are you using a development board? If not, how is everything wired up?

XiaofanLii, 2018-08-01:
@projectgus 
When I change the usb chipset to FT232R, all esptool commond work fine.
I was used ch340 and cp2102, it report same error.
Maybe ch340 and cp2102 shipset and driver(on macOS) did not support re-configure.

closed 2018-08-01 by XiaofanLii

projectgus, 2018-08-01:
Hi @XiaofanLii ,

Are you closing because you resolved this problem? CP2102 & CH340 should both work with esptool.py on Mac OS.

XiaofanLii, 2018-08-02:
@projectgus No, I close this because I found another useful chipset on my macOS. About CP2102 & CH340, I will do more test.