espressif/esptool #335: Issues with esptool.py read_mem EEPROM section

mimaamsa, 2018-07-28:
I want to be able to read the EEPROM section using read_mem. I know the amount of data to be read and I would make the read_mem calls accordingly.

Tried following:
==== 
#include <EEPROM.h>

void setup() {
    EEPROM.begin(512);

    size_t i;
    char yourString[] = "Some text here.";
    for(i=0; i<=strlen(yourString); i++)
        EEPROM.write(i, (uint8_t) yourString[i]);

    EEPROM.commit();
    EEPROM.end();
}

void loop() {
}
==== 

Now, trying to read EEPROM contents from the memory location "0x405fb000" (address of last sector for nodemcu esp8266) always returns "0x00008000"

==== 
[shell] : python3.6 esptool.py --port /dev/cu.SLAB_USBtoUART --after no_reset --no-stub read_mem 0x405fb000
esptool.py v2.5.0
Serial port /dev/cu.SLAB_USBtoUART
Connecting........_
Detecting chip type... ESP8266
Chip is ESP8266EX
Features: WiFi
MAC: 68:c6:3a:ca:fc:56
Enabling default SPI flash mode...
0x405fb000 = 0x00008000
Staying in bootloader.
==== 

Please guide in figuring out, how to achieve the reading from EEPROM ?

projectgus, 2018-07-31:
Hi,

The "EEPROM" library in Arduino uses the SPI flash contents, but it reads the SPI flash via the flash access cache (MMU) which maps a chunk of flash into the address space, where it can be read like RAM. This is where this memory address 0x405fb000 comes from.

When esptool.py runs it doesn't use the flash cache, so you can't read flash by doing simple memory reads.

The flash cache maps the physical flash into the address space at offset 0x40200000. So you can find the physical address in the flash chip via (0x405fb000 - 0x40200000 = 0x3fb000).

You should be able to dump this physical address from the chip to a file using the `read_flash` command:

```
esptool.py -p PORT -b 230400 read_flash 0x3fb000 0x1000 eeprom.bin
```

(Note I used 0x1000 as an arbitrary size to read, you'll have to check the ESP8266 Arduino library to know how large the EEPROM region actually is, the storage on the flash may be different to the storage size specified in EEPROM.begin(), to account for the flash needing to be erased each time it is changed.)

mimaamsa, 2018-08-01:
@projectgus 
Thanks a lot for the insight. Will try and update here.

projectgus, 2018-12-20:
Closing due to inactivity, and as this does not appear to be an esptool issue.

closed 2018-12-20 by projectgus