espressif/esptool #336: A fatal error occurred: Timed out waiting for packet header

defworx, 2018-07-29:
(Please delete any lines which don't apply.)

* Operating system: Win10
* Python version: (`python -V` to check this)
* ESP hardware in use: NodeMCU LUA Amica V2

# Full output from esptool.py (please copy and paste all lines of output)
esptool.py v2.3.1
Connecting....

A fatal error occurred: Timed out waiting for packet header
A fatal error occurred: Timed out waiting for packet header

# What is the expected behaviour?
I was hoping for a full upload of a simple blink example.

# Do you have any other information from investigating this?
I tried with pretty much every baud rate possible and also tried pressing the FLASH or FLASH + RST button while it is connecting.

projectgus, 2018-07-31:
Hi @defworx ,

Unfortunately it sounds like you may have faulty hardware. To be extra sure, please double-check the serial port and then run `esptool.py -p (port) --trace flash_id`. Try pressing and holding the FLASH button then pressing and releasing the RST button while it is trying to connect (keep the FLASH button held down the whole time). Then copy and paste the output into this issue.

mrfaptastic, 2018-09-28:
Hi,

This problem also occurs when using a LOLIN D32. Manually shorting D0 to GND resolves it, but I thought the boot mode could be changed via ESPtool?

Archiving built core (caching) in: C:\Users\Test\AppData\Local\Temp\arduino_cache_186141\core\core_esp32_esp32_d32_PartitionScheme_default,FlashFreq_80,UploadSpeed_115200,DebugLevel_debug_50838b78f9e89f0aaa3fdf6739a756ca.a
Sketch uses 177136 bytes (13%) of program storage space. Maximum is 1310720 bytes.
Global variables use 13884 bytes (4%) of dynamic memory, leaving 313796 bytes for local variables. Maximum is 327680 bytes.
C:\Users\Test\AppData\Local\Arduino15\packages\esp32\tools\esptool\2.3.1/esptool.exe --chip esp32 --port COM5 --baud 115200 --before default_reset --after hard_reset write_flash -z --flash_mode dio --flash_freq 80m --flash_size detect 0xe000 C:\Users\Test\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.0/tools/partitions/boot_app0.bin 0x1000 C:\Users\Test\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.0/tools/sdk/bin/bootloader_dio_80m.bin 0x10000 C:\Users\ANONYM~1\AppData\Local\Temp\arduino_build_769288/Blink.ino.bin 0x8000 C:\Users\Test\AppData\Local\Temp\arduino_build_769288/Blink.ino.partitions.bin 
esptool.py v2.3.1
Connecting........_____....._____....._____....._____....._____....._____....._____....._____....._____....._____

A fatal error occurred: Failed to connect to ESP32: Timed out waiting for packet header
A fatal error occurred: Failed to connect to ESP32: Timed out waiting for packet header

richard-scott, 2019-04-02:
@defworx - Have you connected D0 to RST to enable **ESP.deepSleep** to work?

I had the same issue when I erase_flash'd the unit. I had to disconnect my jumper cable between those pins before I could re-upload the next time.

Normal uploads are OK, it only went wrong after I erased the flash.

projectgus, 2019-05-28:
Hi @defworx ,

Did you make any progress with this issue? Does swapping to another NodeMCU board fix the problem?

Angus

defworx, 2019-05-28:
@projectgus sadly not, gave up on it after a week or so of trying the wildest things. It seemed like it was some kind of problem with the boards. Maybe even the whole batch of boards as I tried two similar boards ordered at the same time.

Had lots of fun with the Arduino MKR1010 tho. Incredible board but also quite expensive.

projectgus, 2019-05-29:
Hi @defworx ,

Sorry to hear that, but glad you found a good alternative board.

I'm going to close this issue as I think it's something specific to either your computer or these particular NodeMCU boards, and there's nothing else we can do from the esptool side.

closed 2019-05-29 by projectgus