espressif/esptool #340: Make sure all files are closed

borro0, 2018-08-01:
(Please delete any lines which don't apply.)

* Operating system: Windows 8
* Python version: (`python -V` to check this) 3.6.6
* ESP hardware in use: ESP8266

I used the esptool.py release 2.5.0

I've made very small modifications to the script so I could call it from another python script. (main takes args as arguments). I've done this so I can call it inside my python postbuild script.

When the esptool.main(args) call was finished, I could not delete the file which was used as input for the script. This was because the esptool did not close all file pointers.

I fixed this by replacing all f.open() with "with open() as f:"
Although this is already used in most places, it was not everywhere.

projectgus, 2018-08-01:
Hi @borro0 ,

This seems like quite a common use case for people using esptool (which is reasonable, as the other classes exposed in the esptool module don't offer the full functionality et.)

Would you be able to send a [GitHub pull request](https://help.github.com/articles/about-pull-requests/) with your modifications, please?

borro0, 2018-08-02:
Done: #341

commit b82df374 (2019-01-09): Properly closing all file pointers.

Closes #340

(From #341 with minor changes.)
closed 2019-01-09 by projectgus