espressif/esptool #342: Cannot read from flash using `read_flash` at any baudrate except 115200

sachin0x18, 2018-08-03:
* Operating system: MacOS
* Python version: 2.7.15
* ESP hardware in use: ESP-WROVER-KIT

Running the following command hangs in the middle with no errors 

`esptool --trace --baud 2000000 read_flash 0xd000 0x2000 tmp.bin`

However writing to flash with any baudrate works perfectly fine.

`esptool --trace --baud 2000000 write_flash 0xd000 tmp.bin`

### LOGS:

#### read_flash
```
esptool.py v2.5.0
Found 5 serial ports
Serial port /dev/cu.usbserial-143301
Connecting...TRACE +0.000 command op=0x08 data len=36 wait_response=1 timeout=0.100 data=
    0707122055555555 5555555555555555 | ... UUUUUUUUUUUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    55555555                          | UUUU
TRACE +0.001 Write 46 bytes:
    c000082400000000 0007071220555555 | ...$........ UUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    5555555555555555 5555555555c0     | UUUUUUUUUUUUU.
TRACE +0.106 Timed out waiting for packet header
.TRACE +0.052 command op=0x08 data len=36 wait_response=1 timeout=0.100 data=
    0707122055555555 5555555555555555 | ... UUUUUUUUUUUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    55555555                          | UUUU
TRACE +0.000 Write 46 bytes:
    c000082400000000 0007071220555555 | ...$........ UUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    5555555555555555 5555555555c0     | UUUUUUUUUUUUU.
TRACE +0.021 Read 1 bytes: c0
TRACE +0.000 Read 111 bytes:
    0108040007205555 00000000c0c00108 | ..... UU........
    0400072055550000 0000c0c001080400 | ... UU..........
    0720555500000000 c0c0010804000720 | . UU...........
    555500000000c0c0 0108040007205555 | UU........... UU
    00000000c0c00108 0400072055550000 | ........... UU..
    0000c0c001080400 0720555500000000 | ......... UU....
    c0c0010804000720 555500000000c0   | ....... UU.....
TRACE +0.001 Received full packet: 010804000720555500000000
TRACE +0.000 Received full packet: 010804000720555500000000
TRACE +0.000 Received full packet: 010804000720555500000000
TRACE +0.000 Received full packet: 010804000720555500000000
TRACE +0.000 Received full packet: 010804000720555500000000
TRACE +0.000 Received full packet: 010804000720555500000000
TRACE +0.000 Received full packet: 010804000720555500000000
TRACE +0.000 Received full packet: 010804000720555500000000

Detecting chip type...TRACE +0.000 command op=0x0a data len=4 wait_response=1 timeout=3.000 data=78000060
TRACE +0.000 Write 14 bytes: c0000a04000000000078000060c0
TRACE +0.013 Read 1 bytes: c0
TRACE +0.000 Read 13 bytes: 010a04000025121500000000c0
TRACE +0.000 Received full packet: 010a04000025121500000000
 ESP32
TRACE +0.000 command op=0x0a data len=4 wait_response=1 timeout=3.000 data=0ca00160
TRACE +0.000 Write 14 bytes: c0000a0400000000000ca00160c0
TRACE +0.015 Read 1 bytes: c0
TRACE +0.000 Read 13 bytes: 010a04000080000000000000c0
TRACE +0.000 Received full packet: 010a04000080000000000000
Chip is ESP32D0WDQ6 (revision 1)
TRACE +0.000 command op=0x0a data len=4 wait_response=1 timeout=3.000 data=0ca00160
TRACE +0.000 Write 14 bytes: c0000a0400000000000ca00160c0
TRACE +0.015 Read 1 bytes: c0
TRACE +0.000 Read 13 bytes: 010a04000080000000000000c0
[... 2174 lines cut ...]
TRACE +0.000 Read 8 bytes: 0113120000000000
TRACE +0.016 Read 1 bytes: 58
TRACE +0.000 Read 18 bytes:
    4350bcf317b3780b 3ad7c478a6066800 | CP....x.:..x..h.
    00c0                              | ..
TRACE +0.000 Received full packet:
    0113120000000000 584350bcf317b378 | ........XCP....x
    0b3ad7c478a60668 0000             | .:..x..h..
Hash of data verified.

Leaving...
TRACE +0.002 command op=0x02 data len=16 wait_response=1 timeout=3.000 data=00000000000000000040000000000000
TRACE +0.000 Write 26 bytes:
    c000021000000000 0000000000000000 | ................
    0000400000000000 00c0             | ..@.......
TRACE +0.013 Read 1 bytes: c0
TRACE +0.000 Read 11 bytes: 01020200000000000000c0
TRACE +0.000 Received full packet: 01020200000000000000
TRACE +0.001 command op=0x12 data len=4 wait_response=1 timeout=3.000 data=01000000
TRACE +0.000 Write 14 bytes: c0001204000000000001000000c0
TRACE +0.015 Read 1 bytes: c0
TRACE +0.000 Read 11 bytes: 01120200000000000000c0
TRACE +0.000 Received full packet: 01120200000000000000
Hard resetting via RTS pin...
```

projectgus, 2018-08-06:
Hi @sachin0x18 ,

I think this is a signal integrity issue (although it might be something with the FTDI chip or the driver). I'm able to reproduce this on Linux with a WROVER-KIT at 5Mbaud but not 2Mbaud - writing small amounts of data works at this speed but reading fails.

The symptom shown in the trace above is that the ESP32 sends a 0x1000 byte SLIP frame but esptool.py only reads 0xfc8 (0x38 bytes are lost somewhere in transit). As I said, I think this is most likely a signal integrity issue, some bytes are invalid on the wire and the FTDI chip is dropping them.

I've added an immediate error in the case where the SLIP frame length looks wrong when received, so it will error out if this happens. I don't think there's any viable way to work around this in esptool.py.

projectgus, 2018-09-27:
Closing as it seems like this a hardware-level issue, and the best thing esptool can do for now is error out.

closed 2018-09-27 by projectgus