espressif/esptool #346: ESP32 - Flash security and boot security

Stasikiii, 2018-08-17:
Hi there

I just try to use esptool for secure my app at esp32.
Can u please help's me with that?

Step 1 - Flash encryption:
1 - espsecure.py generate_flash_encryption_key key.bin
2 - espefuse.py --port /dev/ttyUSB0 burn_key flash_encryption key.bin
3 - espsecure encrypt_flash_data --keyfile key.bin --address 0x10000 -o my-app-encrypted.bin my-app.bin
4 - flash my-app-encrypted.bin to the esp32.

Is this right or i'm lose something?

projectgus, 2018-08-22:
Looks like you posted the same question on the ESP32 forums:
https://esp32.com/viewtopic.php?f=13&t=6809

I've replied to you there, so I'm going to close this issue (as it's more related to ESP-IDF & ESP32 than to esptool).

closed 2018-08-22 by projectgus

Stasikiii, 2018-09-13:
@projectgus thx you!