espressif/esptool #350: ImportError: No module named serial.tools.list_ports

eliyastein, 2018-08-24:
Any help would be much appreciated:

* Operating system:

```
$ uname -a
Darwin Chinaski.local 17.7.0 Darwin Kernel Version 17.7.0: Thu Jun 21 22:53:14 PDT 2018; root:xnu-4570.71.2~1/RELEASE_X86_64 x86_64
```

* Python version:  Python 2.7.10

* ESP hardware in use: ESP8266

```
$ ./esptool.py --port /dev/tty.SLAB_USBtoUART --baud 115200 write_flash --flash_freq 80m --flash_mode qio --flash_size 4m 0x0000 esp8266espruino/boot_v1.6.bin 0x1000 esp8266espruino/espruino_esp8266_user1.bin 0x3FC000 esp8266espruino/esp_init_data_default.bin 0x3FE000 esp8266espruino/blank.bin
Traceback (most recent call last):
  File "./esptool.py", line 35, in <module>
    import serial.tools.list_ports as list_ports
ImportError: No module named serial.tools.list_ports
```

nixmeer, 2018-09-03:
I got the same issue. I am running on macOS 10.13.6. Esptool installed with the esp32 core and works fine when flashing the esp32 through Arduino IDE. When trying to flash tasmota firmware to an esp8266, I get the same error. 

Python 2.7 and 3.7 installed.

projectgus, 2018-09-20:
Hi @nixmeer @eliyastein ,

Thanks for being patient while someone got back to you.

I think in both cases the version of `pyserial` which is installed is too old, and doesn't have `serial.tools.list_ports` module.

You should be able to fix this by running `pip install pyserial`, or re-installing esptool with `pip install esptool`.

I'll add a version check in the top of esptool.py, so the error is clearer.

jssilva25, 2018-09-25:
Hello,

I'm getting the exact same error as the OP, same OS and py27.

I already did `pip install pyserial` and `pip install esptool` and I don't think they are outdated:

`$ pip show pyserial
Name: pyserial
Version: 3.4
Summary: Python Serial Port Extension
Home-page: https://github.com/pyserial/pyserial
Author: Chris Liechti
Author-email: cliechti@gmx.net
License: BSD
Location: /opt/local/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/site-packages
Requires: 
Required-by: esptool`

`$ pip show esptool
Name: esptool
Version: 2.5.1.dev0
Summary: A serial utility to communicate & flash code to Espressif ESP8266 & ESP32 chips.
Home-page: https://github.com/espressif/esptool
Author: Fredrik Ahlberg (themadinventor) & Angus Gratton (projectgus)
Author-email: angus@espressif.com
License: GPLv2+
Location: /opt/local/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/site-packages/esptool-2.5.1.dev0-py2.7.egg
Requires: pyserial, pyaes, ecdsa
Required-by: `

Could you please help?
TIA
jss

siara-cc, 2018-10-13:
Facing same problem. Giving env details below:
```
Arun:hello_world arun$ uname -a
Darwin Arun 16.7.0 Darwin Kernel Version 16.7.0: Thu Jun 15 17:36:27 PDT 2017; root:xnu-3789.70.16~2/RELEASE_X86_64 x86_64
Arun:hello_world arun$ python --version
Python 2.7.10
```

closed 2018-12-21 by projectgus

vinayprydotinfotech, 2019-03-09:
Same error
-----------------------------------------------------------------------------------------------------------
Traceback (most recent call last):
  File "../tools/esptool.py", line 26, in <module>
    import serial
ImportError: No module named serial
../Makefile:202: recipe for target '.output/eagle/debug/bin/eagle.app.v6.bin' failed
make[1]: *** [.output/eagle/debug/bin/eagle.app.v6.bin] Error 1
make[1]: Leaving directory '/home/rydot/Downloads/nodemcu-firmware-2.2.1-master_20181207/app'
Makefile:292: recipe for target '.subdirs' failed
make: *** [.subdirs] Error 2

------------------------------------------------------------------------------
platform  pyhton 3.6 ,Ubuntu 16.04 LTS, 64bit 
installed 
--pip3 install pyserial --user
--pip3 install serial --user 
--pip3 install esptool --user
still not working  
Any more advice ?

projectgus, 2019-03-11:
Hi @vinayprydotinfotech,

(For the record: NodeMCU ships quite an old version of esptool ([v1.2-dev](https://github.com/nodemcu/nodemcu-firmware/blob/master/tools/esptool.py#L34)) plus some custom modifications.)

If you run `/usr/bin/env python` (which is the Python interpreter used by the esptool.py script in NodeMCU), is it version 3.6? (Use control-d to exit.)

As it's ubuntu, installing the native PySerial with `sudo apt-get install python3-serial` may help.

vinayprydotinfotech, 2019-03-11:
rydot@rydot-pc:~/Downloads/nodemcu-firmware-2.2.1-master_20181207$
/usr/bin/env python
Python 3.6.0 |Anaconda 4.3.0 (64-bit)| (default, Dec 23 2016, 12:22:00)
[GCC 4.4.7 20120313 (Red Hat 4.4.7-1)] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>>
i get this when '/usr/bin/env python'  and then followed sudo apt-get
install python3-serial but i get still the same error.

projectgus, 2019-03-13:
> Anaconda 4.3.

You seem to have Python installed via "Anaconda" in addition to (or instead of) the system Python. This is why 'pip' and system Python packages aren't working. You can try something like this to install pyserial under the "conda" package manager for Anaconda:
https://pyserial.readthedocs.io/en/latest/pyserial.html#from-conda

mshavandi, 2019-03-27:
I used
`pip install pyserial`
and this issue solved.

michael-angelozzi, 2019-04-12:
What solved it for me is not calling the script I was testing it with "serial.py", rename to something else.

NEEHITGOYAL, 2020-03-04:
> I used
> `pip install pyserial`
> and this issue solved.

working fine thanks

khataei, 2020-07-25:
I had the same problem. installing the pyserial didn't work. Then I realized that I don't have pip for Python2.7 and pyserial was installed on Python3.8. 
So make sure you have pip for Python2.7 by running
`pip --version`
And then go for 
`pip install pyserial`

henkiejan1, 2020-10-07:
After upgrade yesterday is esptool not working anymore. I thought that python-pip is not available anymore. Only python3-pip alias pip3

My system:
`Linux Inspiron-5370 5.8.12-050812-generic #202009261732 SMP Sat Sep 26 17:35:23 UTC 2020 x86_64 x86_64 x86_64 GNU/Linux` Linux Mint 20 with one of the latest kernels

Python3 version:
`Python 3.8.5`

When i try to install python-pip:
`Pakketlijsten worden ingelezen... Klaar
Boom van vereisten wordt opgebouwd       
De statusinformatie wordt gelezen... Klaar
E: Kan pakket python-pip niet vinden`
Dutch but the translation is: can´t find python-pip package

But after a short research you can get pip for python2 on this site of Python. https://pip.pypa.io/en/stable/installing/ after that you can install esptool and pyserial as berfore.

But there is a warning that python2 stops at januari 2021. So maybe a good idea that esptool runs on python3?

projectgus, 2020-10-08:
> But there is a warning that python2 stops at januari 2021. So maybe a good idea that esptool runs on python3?

@henkiejan1 esptool.py supports Python 3 since version 1.3 (released in January 2017).

In most Debian/Ubuntu based Linux distros (I think including Mint), the `pip` binary is always the Python 2 Pip but if you `apt install python3-pip` then you should be able to do `pip3 install esptool`.

danteruizv8, 2021-06-07:
From:

ESP8266_RTOS_SDK/components/esptool_py/esptool/esptool.py
""
esptool.py depends on pyserial, but there is a conflict with a currently installed package named 'serial'.

You may be able to work around this by 'pip uninstall serial; pip install pyserial' \
but this may break other installed Python software that depends on 'serial'.

There is no good fix for this right now, apart from configuring virtualenvs. \
See https://github.com/espressif/esptool/issues/269#issuecomment-385298196 for discussion of the underlying issue(s).""

jonshouse1, 2021-07-03:
Its 2021 and this same fucking crap is still going on.  Why is it ok for "apt update" to kill a frequently used utility. Can't somebody somewhere agree on what the library names for this shit show are going to be AND MAKE THEN UNIQUE ?

JeongJun-Lee, 2022-08-09:
If you are working on the VS, you can look into the version of Python environment you're using like below:
[image]

neddan, 2024-05-20:
> Hi @vinayprydotinfotech,
>
> (For the record: NodeMCU ships quite an old version of esptool ([v1.2-dev](https://github.com/nodemcu/nodemcu-firmware/blob/master/tools/esptool.py#L34)) plus some custom [...]

I was getting the same error on ubuntu 24.04, `pip3 install pyserial` did not help and serial was not not installed either when i tried to uninstall both. Eventually @projectgus comment of installing pyserial via apt did the trick.