espressif/esptool #352: Error code 01050000

vgonet, 2018-08-30:
(Please delete any lines which don't apply.)

* Operating system: Windows 10
* Python version: (`python -V` to check this) 3.6
* ESP hardware in use: WROOM-32

# Full esptool.py command line as run:
esptool.py --trace -p COM14 read_mac

# Full output from esptool.py (please copy and paste all lines of output)

esptool.py v2.5.0
Serial port COM14
Connecting...TRACE +0.000 command op=0x08 data len=36 wait_response=1 timeout=0.100 data=
    0707122055555555 5555555555555555 | ... UUUUUUUUUUUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    55555555                          | UUUU
TRACE +0.000 Write 46 bytes:
    c000082400000000 0007071220555555 | ...$........ UUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    5555555555555555 5555555555c0     | UUUUUUUUUUUUU.
TRACE +0.117 Timed out waiting for packet header
.TRACE +0.060 command op=0x08 data len=36 wait_response=1 timeout=0.100 data=
    0707122055555555 5555555555555555 | ... UUUUUUUUUUUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    55555555                          | UUUU
TRACE +0.000 Write 46 bytes:
    c000082400000000 0007071220555555 | ...$........ UUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    5555555555555555 5555555555c0     | UUUUUUUUUUUUU.
TRACE +0.010 Read 1 bytes: 1b
TRACE +0.010 Read invalid data: 1b
TRACE +0.000 Remaining data in serial buffer: 5b303b33326d4920
.TRACE +0.060 command op=0x08 data len=36 wait_response=1 timeout=0.100 data=
    0707122055555555 5555555555555555 | ... UUUUUUUUUUUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    55555555                          | UUUU
TRACE +0.000 Write 46 bytes:
    c000082400000000 0007071220555555 | ...$........ UUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    5555555555555555 5555555555c0     | UUUUUUUUUUUUU.
TRACE +0.000 Read 74 bytes:
    322043505520636f 7265732c20576946 | 2 CPU cores, WiF
    692f42542f424c45 2c2073696c69636f | i/BT/BLE, silico
    6e20726576697369 6f6e20312c20344d | n revision 1, 4M
    422065787465726e 616c20666c617368 | B external flash
    0d0a1b5b303b3332 6d49             | ...[0;32mI
TRACE +0.000 Read invalid data:
    322043505520636f 7265732c20576946 | 2 CPU cores, WiF
    692f42542f424c45 2c2073696c69636f | i/BT/BLE, silico
    6e20726576697369 6f6e20312c20344d | n revision 1, 4M
    422065787465726e 616c20666c617368 | B external flash
    0d0a1b5b303b3332 6d49             | ...[0;32mI
TRACE +0.010 Remaining data in serial buffer:
    2028313038292077 6966695f75706461 |  (108) wifi_upda
    74653a206e657477 6f726b496e69741b | te: networkInit.
    5b306d0d0a1b5b30 3b33326d49202831 | [0m...[0;32mI (1
    3039292077696669 5f75             | 09) wifi_u
.TRACE +0.070 command op=0x08 data len=36 wait_response=1 timeout=0.100 data=
    0707122055555555 5555555555555555 | ... UUUUUUUUUUUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    55555555                          | UUUU
TRACE +0.000 Write 46 bytes:
    c000082400000000 0007071220555555 | ...$........ UUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    5555555555555555 5555555555c0     | UUUUUUUUUUUUU.
TRACE +0.000 Read 20 bytes:
    2072782062756666 6572206e756d3a20 |  rx buffer num:
    33320d0a                          | 32..
TRACE +0.000 Read invalid data:
[... 538 lines cut ...]
TRACE +0.000 Read 1 bytes: 40
TRACE +0.000 Read 1 bytes: a4
TRACE +0.000 Read 1 bytes: 01
TRACE +0.000 Read 1 bytes: 05
TRACE +0.000 Read 1 bytes: 00
TRACE +0.000 Read 1 bytes: 00
TRACE +0.000 Read 1 bytes: c0
TRACE +0.000 Received full packet: 0107040000f940a401050000

A fatal error occurred: Failed to write to target RAM (result was 01050000)

# What is the expected behaviour?

Get the mac address

# Do you have any other information from investigating this?

# Is there any other information you can think of which will help us reproduce this problem?

We are trying to send this command through a STM32F0. We convert the USB frames to UART (like a FTDI device).

What is the error code 01050000 ?

Best regards,
Vincent

projectgus, 2018-09-19:
Hi @vgonet ,

Thanks for being patient while someone got back to you.

The error indicates that the ESP32 thought the memory write packet was malformed in some way. The fact that it reads back one byte at a time (different to all previous responses) seems to indicate some unusual timing with the USB/serial interface, is it possible the STM32F0 might be introducing some delays?

Angus

vgonet, 2018-09-20:
Hi Angus,

Thank you for your answer. You're right, it was a problem of delay. I've
solved the problem by sending the USB frames to the PC directly after
receiving an UART packet. I've used the DMA and the interrupts on the STM32
to be more efficient.

Best regards
Vincent

adamglowacki, 2018-10-12:
It's great that the problem of @vgonet was finally solved. But could somebody publish/point to a list of bootloader error codes? At times it could be really helpful to know what they mean.

projectgus, 2018-12-20:
@adamglowacki  https://github.com/espressif/esptool/wiki/Serial-Protocol#rom-loader-error-codes

closed 2018-12-20 by projectgus

hecnelovera, 2019-05-29:
> Hi Angus, Thank you for your answer. You're right, it was a problem of delay. I've solved the problem by sending the USB frames to the PC directly after receiving an UART packet. I've used the DMA [...]

Hello ! I'm having the same problem here, I have also a ST and I use DMA + Interruptions but I'm having exactly de same behavior. Can you please develop a bit more your solution ? 
Thanks.

vgonet, 2019-05-29:
Hello hecnelovera,

To solve my issue, I have replaced the UART DMA by UART IT (UART DMA was
not working but I don't know why).
The initialization of the UART is done inside the CDC_Init_FS(void)
function.

I give you my code (usbd_cdc_if.c). It is not optimized but I hope it will
help you.

Best regards,
Vincent

//_____________________________________________________________________________
//
// Copyright (C) 2018                   Mobsya                   CH-1020 Renens
//_____________________________________________________________________________
//
// PROJECT   Thymio-III
//_____________________________________________________________________________
//
//! \file    usbd_cdc_if.c
//! \brief   This module provides the useful functions to use the USB protocol
//
//! \author  Vincent Gonet
//
//! \version $Id: usbd_cdc_if.c 18076 2017-04-20 12:28:12Z v.gonet $
//_____________________________________________________________________________

//-----------------------------------------------------------------------------
// Include Section
//-----------------------------------------------------------------------------

#include "usbd_cdc_if.h"

#include "power.h"
#include "timer.h"
#include "uart.h"

//-----------------------------------------------------------------------------
// Constants/Macros Definitions
//-----------------------------------------------------------------------------

#define BUFFER_SIZE  2048u

#define when(cond) if(({static unsigned char prev; \
            unsigned char c = !!(cond); \
            unsigned char result = c && !prev; \
            prev = c; \
            result;}))

//-----------------------------------------------------------------------------
// Types Definitions
//-----------------------------------------------------------------------------

typedef struct
{
  uint8_t  Data[BUFFER_SIZE];  //!< USB Buffer
  uint16_t WrPos;      //!< Write pointer position in the ring data buffer.
  uint16_t RdPos;      //!< Read pointer position in the ring data buffer.
  uint16_t Count;      //!< Number of unread bytes in the buffer.
} T_Buffer;

//-----------------------------------------------------------------------------
// Exported Global Data
//-----------------------------------------------------------------------------

extern USBD_HandleTypeDef hUsbDeviceFS;

//-----------------------------------------------------------------------------
[... 311 lines cut ...]

void HAL_UART_TxCpltCallback(UART_HandleTypeDef* huart)
{
  // Initiate next USB packet transfer once UART completes transfer (transmitting data over TX line)
  USBD_CDC_ReceivePacket(&hUsbDeviceFS);
}

//_____________________________________________________________________________

void HAL_UART_RxCpltCallback(UART_HandleTypeDef* huart)
{
  //if (USBD_IsOpen())  // FIXME
  {
    TxBuffer.WrPos++;

    // To avoid buffer overflow
    if (TxBuffer.WrPos == BUFFER_SIZE)
    {
      TxBuffer.WrPos = 0;
    }
  }

  // Start another reception: provide the buffer pointer with offset and the buffer size
  HAL_UART_Receive_IT(huart, (uint8_t*)(TxBuffer.Data + TxBuffer.WrPos), 1);
}