espressif/esptool #353: "read_flash" not working on Windows10

vel2000, 2018-09-05:
(Please delete any lines which don't apply.)

* Operating system: Windows10/64
* Python version: 2.7
* ESP hardware in use: 8266

# Full esptool.py command line as run:
esptool.py --port COM5 -b 115200 read_flash 0 0x100000 firmware.bin
# Full output from esptool.py (please copy and paste all lines of output)
```
esptool.py v2.5.0
Serial port COM5
Connecting......
Detecting chip type... ESP8266
Chip is ESP8266EX
Features: WiFi
MAC: bc:dd:c2:ea:47:64
Uploading stub...
Running stub...
Stub `running...      <---- stucks here forever
```
# What is the expected behaviour?
read flash and save it to file "firmware.bin"
# Do you have any other information from investigating this?
same command line (except --Com 5) works on an Raspberry Pi with esptool 2.5 and flash is properly read and saved to firmware.bin
BTW: erase_flash and write_flash work on Win10 as expected.

andrethomas, 2018-09-07:
What type of usb serial connection are you using?

My best guess is the Windows driver for the usb serial device is malfunctioning.

projectgus, 2018-09-19:
Hi @vel2000 ,

Thanks for being patient while I got back to you. Can you please try adding the --trace argument, ie:

```
esptool.py --trace --port COM5 -b 115200 read_flash 0 0x100000 firmware.bin
```

lev2k, 2018-10-17:
This "issue" can be closed.
After some tests I can confirm, it's a hardware/driver (no official Win10 drivers), issue under Win 10.
Same device works reliable under Windows 7.

projectgus, 2018-10-26:
Thanks for the update, @lev2k.

closed 2018-10-26 by projectgus