espressif/esptool #356: Path .bin files on Mac OS X

GPL71, 2018-09-10:
* Operating system: Mac OSX Sierra
* Python version: 3.7
* ESP hardware in use: sonoff

Thank you for your hard work before. I'm new using this tool. I'm trying learn.
1 - Please what's the path on Mac OS X for .bin files to allow esptool upload them for flash?
2 - I need performe this command to flash firmware:

      esptool.py -p <ESPPort> --baud 115200 write_flash -fs 1M -fm dout -ff 40m \
      0x0 rboot.bin 0x1000 blank_config.bin 0x2000 otaboot.bin

     Can I run terminal on mac and paste this command and start (sonoff connected to FDTI)?

Thank you so much

projectgus, 2018-09-19:
HI @GPL71 ,

I'm not sure I understand the question.

If you place all the .bin files in the same directory and run esptool.py while in this directory, then you can flash.

projectgus, 2018-12-20:
Hi @GPL71 ,

Thanks for being patient while I got back to you.

If you install esptool.py via pip then it should already be on your PATH. So you can run it from any directory. You just have to use the terminal to change directory into the directory which contains your .bin files.

As this is a question rather than an issue with esptool, I'm going to close it. A good place to ask this kind of question is https://bbs.espressif.com or the https://esp8266.com community forum.

closed 2018-12-20 by projectgus