espressif/esptool #359: A fatal error occurred: Timed out waiting for packet content

Zeni241, 2018-09-12:
```
PS D:\PiArduino\esptools> esptool.py --port COM5 --baud 9600 erase_flash
esptool.py v2.5.0
Serial port COM5
Connecting....
Detecting chip type... ESP8266
Chip is ESP8266EX
Features: WiFi
MAC: 84:f3:xx:bx:xx:xx
Uploading stub...
Running stub...
Stub running...
Erasing flash (this may take a while)...

A fatal error occurred: Timed out waiting for packet content
```
I am getting this error when I tried to erase flash of my new Lolin NodeMCU V3. I want to flash new firmware once I am able to erase all the flash.
Any help please?

marcelstoer, 2018-09-13:
Try with `--no-stub`. I also suggest you increase the baud rate to at least 115'200.

ron4all, 2018-11-12:
Hi Zeni241 - Thank you for the update, my NodeMCU has the same issue, i can stop troubleshooting now!