espressif/esptool #361: [Enhancement] Add bootloader switchmode for Pycom Expansion Board 3.0

cyberman54, 2018-09-24:
Pycom provides an interesting series of ESP32 based boards, both delevopment and OEM modules. So it is a great platform to scale up from prototype to production. That does not mean to use Pycom software, just using their modules like ESP32 natives.

For this purpose it would be a useful enhancement to implement a bootloader switch in esptool, which 
makes the pycom expansion board 3.0 to switch a Pycom module to bootloader mode.

projectgus, 2018-10-01:
Hi @cyberman54 ,

Interesting suggestion. Do you have any technical information on how this works?

Angus

cyberman54, 2018-10-02:
@projectgus No, i don't, and unfortunately the pycom expansion boards are not well documented.
Since the pycom bootloader is able to switch the ESP32 in programming mode, they ultimately must use the same switching mechanism in their software as described [here](https://github.com/espressif/esptool/wiki/ESP32-Boot-Mode-Selection), what means that the hardware on the expansion board must support switching EN and GPIO0 pins. They write that the don't use a typical USB-UART chip like CP210x, but do it with a PIC processor with own firmware on the expansion board. So it should be possible to find out how it works by reverse engineering of the PIC firmware.

radimkarnis, 2020-12-15:
Hello everyone,
I am closing this issue because it seems like it is not relevant anymore. Please reopen if you think further discussion is needed.

closed 2020-12-15 by radimkarnis