espressif/esptool #364: idf.py: ImportError: No module named serial.tools.list_ports

magnomp, 2018-10-16:
* Operating system:
Windows 7 x64
* Python version: (`python -V` to check this)
Python 2.7.14
* ESP hardware in use:
ESP32

# Full idf.py command line as run:
idf.py menuconfig

# Full output from idf.py (please copy and paste all lines of output)
Traceback (most recent call last):
  File "C:\Users\Magno\esp\esp-idf\tools\idf.py", line 38, in <module>
    import serial.tools.list_ports
ImportError: No module named serial.tools.list_ports

# What is the expected behaviour?
Running menuconfig

# Do you have any other information from investigating this?
Already did pip install pyserial and pip install esptool as suggested on #350 

# Is there any other information you can think of which will help us reproduce this problem?
Output of pip show pyserial
Name: pyserial
Version: 3.4
Summary: Python Serial Port Extension
Home-page: https://github.com/pyserial/pyserial
Author: Chris Liechti
Author-email: cliechti@gmx.net
License: BSD
Location: c:\python27\lib\site-packages
Requires:

Output of pip show esptool
Name: esptool
Version: 2.5.1
Summary: A serial utility to communicate & flash code to Espressif ESP8266 & ESP
32 chips.
Home-page: https://github.com/espressif/esptool
Author: Fredrik Ahlberg (themadinventor) & Angus Gratton (projectgus)
Author-email: angus@espressif.com
License: GPLv2+
Location: c:\python27\lib\site-packages
Requires: pyserial, pyaes, ecdsa

magnomp, 2018-10-16:
Sorry, I think I should have posted it on esp-idf. Already reposted there

projectgus, 2018-10-25:
OK, I'm going to close this issue then.

closed 2018-10-25 by projectgus