espressif/esptool #365: Cannot flash with stub without getting a ESP_BAD_DATA_CHECKSUM (0xC1) error

Ralim, 2018-10-17:
* Operating system: Linux
* Python version: (`python -V` to check this)
* ESP hardware in use: ESP32-WROVER

# What is the expected behaviour?
ESP_OK 
# Do you have any other information from investigating this?

Currently trying to bootstrap the device using another microcontroller.
Been comparing the serial traces with those from the `--trace` option.
Sending the SPI setup params and loading the STUB program appears to go well.
Sending the 0x02 start command also appears to work (no error).
However, sending any 0x03 Message to send the flash data always results in a 0xC1 response.

I'm calculating the checksum using the same function pulled from the source code in this repository.
I'm calculating the checksum *before* SLIP encoding any of the data.
I only calculate the checksum over the data bytes (skipping first 16 bytes).

# Is there any other information you can think of which will help us reproduce this problem?
Non-stub also gives an error, but you *still* have not uploaded the ROM error codes so this is of little help.

Is there any more detailed resources on trying to speak this protocol? Struggling to figure out why the checksum of all things is failing.

projectgus, 2018-10-25:
Hi @Ralim,

To make sure I understand your post: you're not using esptool.py at all but you've written some microcontroller firmware to speak the flashing protocol?

The esptool.py source code (including the flasher stub source) and [this wiki page](https://github.com/espressif/esptool/wiki/Serial-Protocol) are the best resources to refer to. Have you read the [checksum](https://github.com/espressif/esptool/wiki/Serial-Protocol#checksum) section on that page?

Is the length field being set correctly?

Probably the best advice I can give is to find a way to have your code produce the same data packet as esptool.py (maybe flash the same data?), and then compare the protocol level output.

> Non-stub also gives an error, but you still have not uploaded the ROM error codes so this is of little help.

Honestly, I haven't done this because the ROM codes are not really very informative, I can't remember ever looking them up.

But I've just put them up here: https://github.com/espressif/esptool/wiki/Serial-Protocol#rom-loader-error-codes

Angus

Ralim, 2018-10-26:
Hi @projectgus, 

Yes indeed, I'm attempting to flash an ESP32 from another ESP32.

My code for the flashing is based on the code from the flasher stub program, and I'm using the exact same checksum routine etc. The length field is correct as far as I can decipher myself, and also shifting it by 1 either direction throws `ESP_BAD_DATA_LEN`.

I've been trying to force the two to upload the same data but manipulating the esptool to match hasn't gone overly well.

That's fair enough on the ROM codes, just found it odd that you note that they are different but then never document them.

I've also found that changing the baud rate is very sensitive (cant seem to go higher than 115200) without the ESP stopping responding to packets entirely, is there an interaction between using two ESP32's that I could be missing here ? (the flasher is set to fixed cpu frequency, without dynamic scaling).
Using uart1 with ref_tick so I thought that the baud rate should be fairly accurate?

projectgus, 2018-10-26:
> I've also found that changing the baud rate is very sensitive (cant seem to go higher than 115200) without the ESP stopping responding to packets entirely, is there an interaction between using two [...]
Using uart1 with ref_tick so I thought that the baud rate should be fairly accurate?

This implies maybe there's some signal integrity or baudrate mismatch problem, which could explain the checksum problem as well.

* Do you see framing errors in the logic analyzer?
* If you look on oscilloscope, do you see any noise in the UART signal?
* How are you synchronized baud rate between the two, does your code send the SYNC packets as esptool.py (which triggers auto-bauding behaviour in the stub loader)? Do you send the "Set baud rate" command to the stub, like esptool.py does? (this should set a new baud rate, based on the previously automatic synced rate as a calibration factor.)
* If you drop the baud rate to something very low like 9600bps, does anything start succeeding?

Ralim, 2018-10-26:
I send the same SYNC pattern as the boot loader until I see a response from the ESP that decodes correctly, (following both esptool and the wiki). I start the connection at 57600 initially and then once the stub boot loader has been uploaded and I have received the `OHAI` message I then send the spi parameters and then the baud rate change (I think this is the order, I've matched the same order as the esptool).

I dont see framing errors on the signal analyzer in its decoding of any of the commands that I send at the higher speeds, though it does measure the speed as about 1-2% fast if i let its "autobaud" work it out. (But does decode fine if i manually set the baud rate).

No discernable noise really on the oscilloscope, but I will confirm and double check this.

Running at 9600, 57600,115200 and 230400 all just result in bad checksum errors, anything higher than 230400 results in no response from the ESP at all.

projectgus, 2018-10-26:
That all sounds pretty reasonable to me... Have you checked the logic analyzer captures matches byte for byte what the firmware is sending?

projectgus, 2018-10-26:
> I have checked, and as far as I could spot it seemed to match (Hard to check all 16kb, but the first and last 512 definitely matched).

If you shrink the minimum data size, you can check the first flash data command at least?

Actually, that's something: Does the flashing fail on the first flash data command, or some later command? Does it always fail at the same place.

> Any idea how accurate the ESP32 UART is when in the bootloader ?

Short answer: pretty accurate.

Auto-bauding works pretty reliably at low speeds (less than 230400 or 460800). It's a little less reliable at very high speeds, which is why esptool.py autobauds at 115200 and then scales up.

The ESP32 ROM code estimates the crystal frequency using an internal RC oscillator, so it can sometimes set the wrong baud rate if the "set baud rate" command is sent to ROM code. This is not really a problem for low speeds either, but it can be flaky in the >1Mbaud range.

To get around this, the software loader assumes auto-baud is more reliable than crystal baud rate detection so it scales up the new baud rate based on whatever clock divider is currently in use for 115200 (based on auto-bauding). This is generally more reliable at higher baud rates.

But any of these approaches should be sufficient at the baud rates you're talking about.

Ralim, 2018-10-31:
Thanks for your information, comparing small writes to the esptool gave the necessary insight to find that my inital 0x02 params differed from the esptool. Seems to be working fairly stable now.

Also found that when flashing from the ESP32, had to have dynamic power control disabled at core at 160MHz for it to maintain the baud rate correctly.

closed 2018-10-31 by Ralim

projectgus, 2018-10-31:
@Ralim Glad you figured out the missing detail.

> found that when flashing from the ESP32, had to have dynamic power control disabled at core at 160MHz for it to maintain the baud rate correctly.

Do you mean dynamic frequency scaling? When this is enabled, the UART source clock is changed to REF_TICK which is a constant 1MHz clock, so baud rates will be divided from that master frequency rather than APB frequency ( which is 80MHz when DFS is not enabled, and variable when DFS is enabled).

Using REF_TICK can limit the full range of available baud rates. How fast was it running when you experienced problems?

Ralim, 2018-10-31:
Sorry, yes I meant dynamic frequency scaling.
Just 57600 baud and also 115200 baud, nothing too high.

igrr, 2018-10-31:
Hi @Ralim, could you please post the configuration you are passing to `esp_pm_configure`? Thanks.

Ralim, 2018-10-31:
I was only using the configuration in the make menu at the moment, and hadn't actually gone any further into the configuration of power management yet. 

I did have `uart_config_t use_ref_tick = true` in my uart setup however.