espressif/esptool #366: Error:Set ESP8266 Address timeout. & Detected flash size: Unknown & A fatal error occurred: Timed out waiting for packet content

xshocuspocusxd, 2018-10-18:
* Operating system: WINDOWS 10
* Python version: Python 3.6.4
* ESP hardware in use: NodeMCU V3 LoLin

# Full esptool.py command line as run: python esptool.py --port COM5 erase_flash

# Full output from esptool.py (please copy and paste all lines of output)
esptool.py v2.6-beta1
Serial port COM5
Connecting....
Detecting chip type... ESP8266
Chip is ESP8266EX
Features: WiFi
MAC: 84:0d:8e:86:db:ef
Uploading stub...
Running stub...
Stub running...
Erasing flash (this may take a while)...

A fatal error occurred: Timed out waiting for packet content

ALSO:

C:\Users\admin\Downloads\esptool-master>python esptool.py --port COM5 flash_id
esptool.py v2.6-beta1
Serial port COM5
Connecting....
Detecting chip type... ESP8266
Chip is ESP8266EX
Features: WiFi
MAC: 84:0d:8e:86:db:ef
Uploading stub...
Running stub...
Stub running...
Manufacturer: 00
Device: 0000
Detected flash size: Unknown
Hard resetting via RTS pin...

# Is there any other information you can think of which will help us reproduce this problem?

My 2 other boards have been flashed correctly. Only one have this issue. 

NODEMCU FIRMWARE PROGRAMMER:
Note:Serial port connected.
Note:Begin find ESP8266.
Note:ESP8266 ACK success.
Note:ESP8266 ACK success.
Error:Set ESP8266 Address timeout.

xshocuspocusxd, 2018-10-25:
SOLVED

Pins SC and S1 was connected together. Low quality of the soldering 
[image]

projectgus, 2018-10-26:
Good to hear you found the root cause.

closed 2018-10-26 by projectgus