espressif/esptool #367: Timeout error with esptool and ESP32

Andermutu, 2018-10-24:
(Please delete any lines which don't apply.)

* Operating system:
* Python version: (`python -V` to check this)
* ESP hardware in use:

# Full esptool.py command line as run:

# Full output from esptool.py (please copy and paste all lines of output)

# What is the expected behaviour?

# Do you have any other information from investigating this?

# Is there any other information you can think of which will help us reproduce this problem?

Andermutu, 2018-10-25:
I have been working a little more and I have noticed that my ESP32 may be corrupted. When I select flash mode I get this message from Minicom.

rst:0x1 (POWERON_RESET),boot:0x3 (DOWNLOAD_BOOT(UART0/UART1/SDIO_REI_REO_V2))
waiting for download

Then, after a few seconds, I get some trash characters and need to reboot my module:
�������������������������������������������������

This happens in flash mode or normal boot mode, after a few seconds I get this trash characters and my ESP32 hangs. 

I have run also this code to check out the state:

sudo espefuse.py --port /dev/ttyUSB0 summary

espefuse.py v2.5.0
Connecting........_____.....__
Efuse fuses:
WR_DIS                 Efuse write disable mask                          = 0 R/W (0x0)
RD_DIS                 Efuse read disablemask                            = 0 R/W (0x0)
CODING_SCHEME          Efuse variable block length scheme                = 0 R/W (0x0)
KEY_STATUS             Usage of efuse block 3 (reserved)                 = 0 R/W (0x0)

Security fuses:
FLASH_CRYPT_CNT        Flash encryption mode counter                     = 0 R/W (0x0)
FLASH_CRYPT_CONFIG     Flash encryption config (key tweak bits)          = 0 R/W (0x0)
CONSOLE_DEBUG_DISABLE  Disable ROM BASIC interpreter fallback            = 1 R/W (0x1)
ABS_DONE_0             secure boot enabled for bootloader                = 0 R/W (0x0)
ABS_DONE_1             secure boot abstract 1 locked                     = 0 R/W (0x0)
JTAG_DISABLE           Disable JTAG                                      = 0 R/W (0x0)
DISABLE_DL_ENCRYPT     Disable flash encryption in UART bootloader       = 0 R/W (0x0)
DISABLE_DL_DECRYPT     Disable flash decryption in UART bootloader       = 0 R/W (0x0)
DISABLE_DL_CACHE       Disable flash cache in UART bootloader            = 0 R/W (0x0)
BLK1                   Flash encryption key                              
  = 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 R/W 
BLK2                   Secure boot key                                   
  = 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 R/W 
BLK3                   Variable Block 3                                  
  = 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 R/W 

Calibration fuses:
BLK3_PART_RESERVE      BLOCK3 partially served for ADC calibration data  = 0 R/W (0x0)
ADC_VREF               Voltage reference calibration                     = 1121 R/W (0x3)

Config fuses:
XPD_SDIO_FORCE         Ignore MTDI pin (GPIO12) for VDD_SDIO on reset    = 0 R/W (0x0)
XPD_SDIO_REG           If XPD_SDIO_FORCE, enable VDD_SDIO reg on reset   = 0 R/W (0x0)
XPD_SDIO_TIEH          If XPD_SDIO_FORCE & XPD_SDIO_REG, 1=3.3V 0=1.8V   = 0 R/W (0x0)
SPI_PAD_CONFIG_CLK     Override SD_CLK pad (GPIO6/SPICLK)                = 0 R/W (0x0)
SPI_PAD_CONFIG_Q       Override SD_DATA_0 pad (GPIO7/SPIQ)               = 0 R/W (0x0)
SPI_PAD_CONFIG_D       Override SD_DATA_1 pad (GPIO8/SPID)               = 0 R/W (0x0)
SPI_PAD_CONFIG_HD      Override SD_DATA_2 pad (GPIO9/SPIHD)              = 0 R/W (0x0)
SPI_PAD_CONFIG_CS0     Override SD_CMD pad (GPIO11/SPICS0)               = 0 R/W (0x0)
DISABLE_SDIO_HOST      Disable SDIO host                                 = 0 R/W (0x0)

Identity fuses:
MAC                    MAC Address                                       
  = 24:0a:c4:8a:b2:d4 (CRC 4d OK) R/W 
CHIP_VER_REV1          Silicon Revision 1                                = 1 R/W (0x1)
CHIP_VERSION           Reserved for future chip versions                 = 0 R/W (0x0)
CHIP_PACKAGE           Chip package identifier                           = 0 R/W (0x0)

Flash voltage (VDD_SDIO) determined by GPIO12 on reset (High for 1.8V, Low/NC for 3.3V).

Please any help is higly appreciated.

Thanks.

Andermutu, 2018-10-25:
Hi, after removing all the GPIO and ADC connections everything works as expected. Can anyone please explain where is the problem with GPIOs? Maybe some of them can not been used?

Thanks again.

projectgus, 2018-10-26:
Hi @Andermutu,

GPIOs 6-11 are shared with the internal SPI flash of the module. Consult the ESP32-WROOM datasheet for more details.

There are also some strapping pins which can control the boot mode. Although from the UART log, it looks like the boot mode was correctly selected. For more details on strapping pins, see ESP32 Datasheet section "Strapping Pins" or [this wiki page](https://github.com/espressif/esptool/wiki/ESP32-Boot-Mode-Selection).

I can't explain why any of these would cause the module to boot into UART download mode and then produce noise output - maybe this is actually an electrical problem of some kind? Is the power supply stable and correct power supply capacitors are in place on the PCB?

It doesn't seem like the underlying issue is a problem with esptool.py, so I'm going to close this issue. If you have more questions about wiring up a WROOM module, the https://esp32.com forums are a great place to ask questions.

closed 2018-10-26 by projectgus