espressif/esptool #370: Flashing ESP8266 with Raspberry Pi 3 does not work

p-king, 2018-11-06:
Hi, I'm trying to flash a new firmware on an ESP8266 with my Raspberry Pi 3, following this [instructions](https://github.com/arendst/Sonoff-Tasmota/wiki/Flash-Sonoff-using-Raspberry-Pi). Trying to erase the flash with:
`sudo esptool.py --port /dev/ttyS0 erase_flash` 
seems to work until the line
`Erasing flash (this might take a while) ...`
is followed by:
`A fatal error occurred: Failed to erase flash. Only got 1 byte status response.`
This happens if I use the GPIO pins of the Raspi. If I use an USB to TTL adapter instead, all works well. 
I already tried to reduce the baud rate, double checked the wiring and tried to use another power supply for the ESP (I used the 3.3V and GND of the USB adapter, since this works when I use /dev/ttUSB0). But without success.

* Operating system: Raspian Stretch 2018-06-27
* Python version: (`python -V` to check this): 2.7.13 
* ESP hardware in use: integrated chip in [Wifi wall plug](https://github.com/arendst/Sonoff-Tasmota/wiki/Obi-Socket) 

# Full esptool.py command line as run:
`sudo esptool.py --port /dev/ttyS0 erase_flash` 

# Full output from esptool.py (please copy and paste all lines of output)
```
esptool.py v2.5.1
Serial port /dev/ttyS0
Connecting....
Detecting chip type... ESP8266
Chip is ESP8266EX
Features: WiFi
MAC: 2c:3a:e8:2f:5b:be
Uploading stub...
Running stub...
Stub running...
Erasing flash (this may take a while)...

A fatal error occurred: Failed to erase flash. Only got 1 byte status response.

```

p-king, 2018-11-07:
So sorry, seems that **I** made a mistake. All works well now. Thanks a lot and best regards, Peter

closed 2018-11-07 by p-king

p-king, 2019-01-11:
Unfortunately I can't recall exactly. I've forgotten some important step of the instructions. As I went back to zero, started over again and followed the instructions letter by letter, it worked at the end. If you use a Rasberry Pi for flashing, it's important that you start the ESP inbetween every flash operation once: start in flashing mode to erase, then start again in normal mode, then start again in flashing mode for flashing the new firmware. Hope that helps!

vjaunet, 2020-04-05:
In case someone ends up here, I just experienced the same knid of error. Powering the esp through a decent power supply instead of the RPi 3.3V rail solved it for me.
Best.