espressif/esptool #371: Add option to flash already compressed firmware

LaurentLouf, 2018-11-07:
It would be nice to be able to have an option to specify that the firmware that we want to flash has already been compressed. That way, flashing multiple units could be faster by building the firmware then compressing it, and using esptool to flash the units without having to choose between loosing time compressing (on Raspberry, it can take some time) or transmitting the firmware (gains in size obtained through compressing can be quite substantial).

As I see it, it would mean adding two options : 

1. Option to specify that the binary is already compressed
2. Option to specify the uncompressed size

Am I missing something ?

projectgus, 2018-11-26:
Hi @LaurentLouf ,

Sorry for the slow response to your suggestion.

I think there's no reason this couldn't work. The thing to establish is how useful such a feature would be. Every new option makes `esptool.py` a bit harder to use, and every new feature makes the codebase a little bit more complex to maintain. So before adding a feature it's important to establish how useful it will be, to how many people - and then decide if this tradeoff is beneficial.

I'm a little surprised that the time esptool needs to compress the binary image is a significant performance hit on Raspberry Pi. Do you have any benchmarks for the performance of zlib compression on a Pi?

Thanks,

Angus

projectgus, 2018-11-26:
Thinking about this some more, esptool currently uses the slowest (max compression) option of zlib (level 9) because of an assumption that serial transfer will always be more expensive than compression/decompression.

If you have a Pi handy, can you please run the following commands in a terminal and post the output:

```
time dd if=/dev/zero bs=4M count=1 | gzip -9 -c | dd of=/dev/null

time dd if=/dev/urandom bs=4M count=1 | gzip -9 -c | dd of=/dev/null

time dd if=/dev/zero bs=4M count=1 | gzip -6 -c | dd of=/dev/null

time dd if=/dev/urandom bs=4M count=1 | gzip -6 -c | dd of=/dev/null

time dd if=/dev/zero bs=4M count=1 | gzip -3 -c | dd of=/dev/null

time dd if=/dev/urandom bs=4M count=1 | gzip -3 -c | dd of=/dev/null
```

projectgus, 2018-12-20:
OK, I hope you figure out what the bottleneck is/was.

closed 2018-12-20 by projectgus