espressif/esptool #375: Is there any way to Update bootloader firmware over the air ?

jegajith, 2018-11-15:
(Please delete any lines which don't apply.)

* Operating system:
* Python version: (`python -V` to check this)
* ESP hardware in use:

# Full esptool.py command line as run:

# Full output from esptool.py (please copy and paste all lines of output)

# What is the expected behaviour?

# Do you have any other information from investigating this?

# Is there any other information you can think of which will help us reproduce this problem?

projectgus, 2018-11-26:
Hi @jegajith ,

esptool is a serial flashing tool, so it can't do any kind of OTA updating.

You haven't said what chip (ESP8266 or ESP32) you are using and which software framework, but in general this is not supported - there's only copy of the bootloader, and updating the bootloader requires erasing and then re-writing it. If a power failure or a crash occurs in the middle of this process, the device will be bricked until a physical reflash (with esptool.py or similar) can be done.

If you have more questions about this, good places to post are https://esp32.com (for ESP-IDF and ESP32 Arduino) and https://bbs.espressif.com (for ESP8266 official SDKs). If you're using a different SDK, they probably have a discussion forum as well.

Angus

closed 2018-11-26 by projectgus

jegajith, 2018-11-28:
@projectgus  Thank you for your answer