espressif/esptool #376: Problem with DSTIKE WiFi Packet MonitorV3

DinoMesina, 2018-11-17:
* Operating system: Linux 4.18.0 64bit
* Python version:  2.7.13
* ESP hardware in use: ESP32-WROVER ( https://www.tindie.com/products/lspoplove/dstike-wifi-packet-monitorv3/ )

_Command line:_ esptool.py -b 460800 read_mac
_Output:_
esptool.py v2.5.1
Found 1 serial ports
Serial port /dev/ttyUSB0
Connecting........_____....._____....._____....._____....._____....._____....._____
/dev/ttyUSB0 failed to connect: Failed to connect to Espressif device: Timed out waiting for packet header
A fatal error occurred: All of the 1 available serial ports could not connect to a Espressif device.

I am able to switch ESP32 to DOWNLOAD mode by shorting to GND the PIN25 (GPIO0) using a cable and then it work, but only for read the MAC ADDRESS, not for re-flash :(

When I try to flash (always shorting to GND PIN25) the response is:
esptool.py v2.5.1
Serial port /dev/ttyUSB0
Connecting........_
Chip is ESP32D0WDQ6 (revision 1)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse
MAC: b4:e6:2d:xx:yy:zz
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 460800
Changed.
Configuring flash size...
Warning: Could not auto-detect Flash size (FlashID=0x0, SizeID=0x0), defaulting to 4MB
Compressed 8192 bytes to 47...
Wrote 8192 bytes (47 compressed) at 0x0000e000 in 0.0 seconds (effective 16924.9 kbit/s)...

A fatal error occurred: Timed out waiting for packet header
A fatal error occurred: Timed out waiting for packet header

Where is the problem?

Thank you
Dino

EsdrasXavier, 2018-11-22:
Same issue

- Operation system:
    - NAME="Linux Mint"
    - VERSION="18.1 (Sonya)"
    - PRETTY_NAME="Linux Mint 18.2"
    - VERSION_ID="18.2"
    - VERSION_CODENAME=sonya
    - UBUNTU_CODENAME=xenial
- Python version: 2.7.12
- Board: https://ae01.alicdn.com/kf/HTB1hFoBQXXXXXbhXXXXq6xXFXXXC/Placa-de-Desenvolvimento-ESP32-WiFi-Bluetooth-Ultra-Baixo-Consumo-De-Energia-Dual-N-cleos-ESP-32.jpg_640x640.jpg

I'm trying to upload using arduino IDE and get the output:
```
esptool.py v2.5.0
Serial port /dev/ttyUSB0
Connecting........_____....._____....._____....._____....._____....._____....._____

A fatal error occurred: Failed to connect to ESP32: Timed out waiting for packet header
A fatal error occurred: Failed to connect to ESP32: Timed out waiting for packet header
```
I already read few issues and none of then workout for me.

projectgus, 2018-11-26:
Hi @DinoMesina ,

It looks like the ESP32 is failing part way through the flash for some reason. Looking at the photos of the PCB on Tindie, it looks like the creator hasn't followed the ESP32-WROVER guidelines (from the ESP32-WROVER datasheet) regarding placement of capacitors for the power supply. It's possible the ESP32 is browning out during the flash. Or it could be some other issue with the power supply (USB cable, USB port, etc.)

You could try flashing at a lower baud rate (maybe 115200bps) and see if things succeed then.

@EsdrasXavier Your log shows the same final error message, but the ESP32 is failing at a different stage of flashing. Suggest following each of the steps here - https://github.com/espressif/esptool/#bootloader-wont-respond - if you still get this error afterwards then please open a new issue.

Angus

DinoMesina, 2018-11-29:
Thank you, very much! 
The problem is the poor quality of the board, there is no capacitors on power (3.3V), I have added a capacitor of 100nF and one of 22uF on the pins GND and 3v3 and now I can re-flash the ESP32! 
Some time it not start to flash but if retry it go... 

Dino

projectgus, 2018-11-30:
Great, good to hear @DinoMesina ! Suggest forwarding this feedback to the board designer, as well.

closed 2018-11-30 by projectgus

EsdrasXavier, 2018-12-03:
@projectgus Thanks for your help, actually I change the board and worked fine, I think the board I was trying to use had some problem.