espressif/esptool #379: ImportError: No module named serial.tools.list_ports

ForceFedDSM, 2018-11-21:
(Please delete any lines which don't apply.)

* Operating system: Windows 10
* Python version: 3.7.1 (`python -V` to check this)
* ESP hardware in use: ESP8266

# Full esptool.py command line as run: esptool.py —ports COM5 erase_flash

# Full output from esptool.py (please copy and paste all lines of output)
 Traceback(most recent call last):
      File “C:\Users\Jared\Downloads\esptool-2.5.1\esptool-2.5.1\esptool.py”, line 35, in <module>
            import serial.tools.list_ports as list_ports
ImportError: No module named serial.tools.list_ports

# What is the expected behaviour?
For flash to be erased

# Do you have any other information from investigating this?
Uninstalled and reinstalled esptool and pyserial using pip. When running “python -m serial.tools.list_ports” , I get a response of COM5, 1 ports found

# Is there any other information you can think of which will help us reproduce this problem?

davydnorris, 2018-11-25:
I had the same problem with python3 - ended up reinstalling python2.7

projectgus, 2018-11-25:
Hi @ForceFedDSM ,

> ImportError: No module named serial.tools.list_ports

> When running “python -m serial.tools.list_ports” , I get a response of COM5, 1 ports found

Is it possible a different Python interpreter is running when you run esptool.py, compared to running python directly?

You can check this by trying `python esptool.py -p PORT erase_flash` which will should use the same interpreter as `python -m serial.tools.list_ports`.

The root case is a pyserial version earlier than v3.0, which shouldn't be there if esptool is installed via pip but might be the case if there is a second python installation hiding somewhere, which is running sometimes. See also #350, #348.

romgen, 2018-12-09:
I had the same problem with a fresh install of python 3 and pip install esptool.
It seems that on windows, you need to specify "python" before issueing the command. 
So instead if `esptool.py -p PORT erase_....`, you need to use `python esptool.py -p PORT erase`

ForceFedDSM, 2018-12-10:
Thanks for all the help guys. I used the suggestion from projectgus and it worked fine. I ran the command as python esptool.py -p COM5 erase_flash and that worked. I don’t know how to do the fancy formatting on my phone so sorry for the text only reply and thanks again for all your suggestions.

closed 2018-12-10 by ForceFedDSM

danteruizv8, 2021-06-07:
From:

ESP8266_RTOS_SDK/components/esptool_py/esptool/esptool.py
""
esptool.py depends on pyserial, but there is a conflict with a currently installed package named 'serial'.

You may be able to work around this by 'pip uninstall serial; pip install pyserial' \
but this may break other installed Python software that depends on 'serial'.

There is no good fix for this right now, apart from configuring virtualenvs. \
See https://github.com/espressif/esptool/issues/269#issuecomment-385298196 for discussion of the underlying issue(s).""