espressif/esptool #380: esptool dated 20/11/18 with NonOS SDK 3.0 bricking devices

davydnorris, 2018-11-23:
* Operating system: Win10
* Python version: 2.7.12
* ESP hardware in use: Lolin NodeMCU v3

I used the latest version to flash my NonOS SDK v3.0 code to the unit, and it appeared to work the first time. Upon reboot the program appeared to run OK too. I made changes and then tried to flash again and that's where the problems started.

The second time, esptool started up, connected to the unit and downloaded the stub, but then threw an error

```
d:/xtools/bin/esptool.py -p /dev/ttyS3 -b 230400 write_flash -ff 40m -fm qio -fs 4MB-c1 0x01000 firmware/upgrade/user1.4096.new.6.bin
esptool.py v2.6-beta1
Serial port /dev/ttyS3
Connecting....
Detecting chip type... ESP8266
Chip is ESP8266EX
Features: WiFi
MAC: 5c:cf:7f:02:9a:b7
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 230400
Changed.
Configuring flash size...
Compressed 235140 bytes to 173134...

A fatal error occurred: Timed out waiting for packet content
make: *** [Makefile:460: flash] Error 2
```
At this point the module refused to do anything more and spat garbage. Switching to the Espressif supplied flash tool also failed in a similar way, with the espressif tool reporting flash details:

```
flash vendor:
00h : N/A
flash devID:
0000h
N/A;N/A
crystal:
26 Mhz
```

Assuming the flash had problems, I have just spent several days looking at power supply, connections, and after the third NodeMCU unit was bricked in the same way I started to suspect esptool.

I then flashed a unit with the Espressif supplied tool using the same code, and repeated this about 4 times. I then used esptool and promptly bricked the unit.

Finally, I took the bricked unit and tried the NodeMCU flash tool, and IT WORKED. I then used the NodeMCU flash tool to unbrick all the other units and they are now working fine and reporting the right flash vendor etc.

I can repeat this behaviour reliably - flash with either the Espressif supplied tool or the NodeMCU tool and it's fine. Flash with esptool and it's broken in a big way. Only NodeMCU tool will fix it.

davydnorris, 2018-11-24:
Doing further investigation using 2.5.1 I have found that this version will flash the chip and not cause it to brick, but the code will not run.

I'm loading user1 but it looks like it's jumping straight to user2 and then failing (because there is no user2). There are no error messages during the flash and it appears to complete fine.

If I flash the exact same binary using the Espressif supplied tool it works correctly.

davydnorris, 2018-11-25:
Further update - I have found that I can unbrick a device simply by writing a blank.bin with the NodeMCU flasher, it doesn't even have to be the full binary.

I have also found that it seems to be NonOS v3.0 related - I have just used the 2.5.1 version of esptool to repeatedly flash a v2.2 binary and it worked fine, but have just flashed a v3.0 binary and it's now bricked. I was able to flash and reflash the v3.0 boot and init data, but the moment I flashed the user1 bin it died.

projectgus, 2018-11-25:
Hi @davydnorris ,

It looks like what happened in the first case is that flashing failed part way through. The "bricking"/garbage in this case is probably the partially flashed binary loading and then crashing in some unexpected way.

I noticed that you had baud rate 230400 set with esptool.py. If you flash with `-b 115200` or even `-b 9600`, do you have a higher success rate?

Angus

davydnorris, 2018-11-26:
Hi @projectgus 

I've been playing with this all weekend with repeatable results and lack of success. right now I am using the Espressif GUI based flash tool to flash my chips outside of my build process. Here's what I have so far:

 - NodeMCU flasher works every time, and will unbrick units that have been bricked by esptool. This works at baud rates up to 921600
 - the NodeMCU tool will unbrick a unit just by writing the 4k blank.bin at any position in the flash memory
 - the Espressif GUI based tool will work correctly but not after esptool has been used. It works up to baud rates of 460800 (not bothered to go any higher). If I use this tool from the start it flashes everything. If I use this tool after esptool it will either (esptool 2.5.1 and above) not flash at all and report an unknown flash vendor as shown above, or (esptool 2.5.0) show the chip getting an invalid magic and trying to boot user2. This is regardless of whether I flash just the user1 bin or the entire boot/user1/defaults. I have to use NodeMCU once to 'reset' the chip and then flash everything and then by binary will boot.
 - esptool behaves pretty much exactly the same at baud rates from 32k up to 460800. If I reset the chip using NodeMCU it will say it flashed correctly and will say the checksum is fine, but will not boot, and then it will either (2.5.1 and above) not flash again, or (2.5.0) appear to flash but not boot correctly.

All these tests were done with the same v3.0 user bin, and the chip plugged into the same USB port. I uninstalled and reinstalled python and pySerial as well with no change. It does tend to point to something weird going on with esptool and the latest v3.0 binaries. Perhaps the other tools are ignoring the v3.0 changes and that's why they are working? I don't know.

BTW the binary is prepared with gen_appbin.py supplied with the NonOS SDK, not with esptool.py

projectgus, 2018-11-30:
Thanks for the update. I must say I don't really have any idea what's going on with your board, this all seems weird.

One thing to note is that is really shouldn't be technically possible to "brick" an ESP8266 in a way which esptool can't recover from - esptool starts by talking to mask ROM code, nothing in the flash at reset-time can change this.

There is an uncommon but known bug (#193, #351) where if the RTC WDT period or deep sleep timeout is set to too short of a timeout, this isn't cleared by a normal reset and can require a power cycle to restore it. Maybe this is a case of this (if the NodeMCU firmware sets the RTC WDT).

To test this theory:

* Remove USB power entirely (ie unplug the cable) for a second or two
* Hold down the FLASH button on the NodeMCU board while power is disconnected.
* Reconnect USB power while still holding the FLASH button down.
* Run an esptool.py command. You can release the FLASH button after the "Connecting..." stage is finished.

davydnorris, 2018-11-30:
Yeah I know - it's really weird. And unfortunately I changed two things at the same time - I bounced esptool to the latest and I also moved to NonOS SDK v3.0 with the new partition table structure.

And I should clarify - the ESP modules are not utterly bricked. The esptool establishes comms and gets the chip details as you can see from my comment up the top, and it downloads the helper stub, but then it is unable to actually flash the content.

As I look at it, I think perhaps the error message being returned is misleading. I don't think it's actually timing out, I think perhaps the error is actually that the flash chip details have somehow been zapped and it's failing because it can't determine the correct flash info. This is what seems to be the case with the Espressif GUI tool as well. Perhaps the only thing the NodeMCU flash tool is doing is NOT checking and just blasting away at the flash!?

This is not limited to one unit - I have 5 Lolin NodeMCU v3 modules and can repeat the behaviour with all of them. I haven't had time to try on my other module from different vendors but I'll try them too - I have about 5 different types, including Espressif WROOM2 modules on my own board.

davydnorris, 2018-11-30:
BTW my code did deep sleep the module, but I have since commented that line out. Hard resets were the first thing I tried because I *was* indeed using deep sleep and it seemed to make no difference.

projectgus, 2018-11-30:
The flash chip info is read from the hardware (by sending SPI flash commands to the chip), so nothing in the flash will affect this.

When you say "hard reset", did you power cycle the chip while holding the FLASH button, as described? This is the only way to be sure it doesn't run any code from flash after power is re-established.

Is anything connected to the module which could be interfering with the SPI flash signals (GPIOs 6-11)?

davydnorris, 2018-11-30:
OK I think we are getting somewhere - I am now pretty sure it's related to the deep sleep issue, but that's raised a few questions...

I pulled the very latest esptool.py, and went through all the scenarios I described earlier and I found I could actually repeatedly use esptool to flash code that did not deep sleep.
I then uncommented the deep sleep lines and found it flashed the first time but the module was messed up after that.
I then went through the process and 'fixed' it with the NodeMCU flasher
I then did the process all over again and got same results
I then did the reset as you described and held the flash button while plugging in and kicking off esptool, this time holding it well into the flash process when I was sure it was trying to communicate with the module, and this time the esptool was able to flash the module correctly.
I then reset and checked the code was running, then tried to flash and got nowhere, then did the hard reset and flashed and it worked fine.

So I think you are correct - it's deep sleep related problems, but:
 - I have been flashing this same deep sleep enabled code for the last 6 months and had no problems until the last week or so when I moved to esptool latest and SDK3 latest
 - did something change in SDK v3.1 that has made deep sleep a bigger issue?
 - did something change in esptool that has made deep sleep a problem again?
 - why does the NodeMCU flash tool always work?
 - most importantly can we find out what they are doing and put it in esptool!!!???

I'm pleased the issue has been narrowed down, but it's still going to cause much annoyance with my development cycle if I have to reset like that every time I want to flash. I guess I can do some things with deep sleep disabled but it would be ideal if that didn't have to happen. I actually built up my test board because in the first revision of my custom PCB I connected the wake pin directly and it stopped me being able to autoflash (I've added a Schottky in rev2), so now I am back to square one.

projectgus, 2018-11-30:
If you use the `--no-stub` option to esptool.py does the problem go away? This was reported as a workaround for other deep sleep options.

projectgus, 2018-11-30:
The `--baud` option should still work for `--no-stub`, it just works in a different way (the initial sync happens at the target baud rate, whereas with the stub the initial sync happens at 115200bps and then the stub changes the baud rate after it's running. This is more reliable at higher baud rates.)

The ROM flasher is slower than the stub flasher in a number of ways, so this may explain the overall reduced performance.

davydnorris, 2018-11-30:
That doesn't seem to be what it's telling me. This is what I see:
```
d:/xtools/bin/esptool.py --no-stub -p /dev/ttyS3 -b 460800 write_flash -ff 40m -fm qio -fs 4MB-c1 0x01000 firmware/upgrade/user1.4096.new.6.bin
esptool.py v2.6-beta1
Serial port /dev/ttyS3
Connecting....
Detecting chip type... ESP8266
Chip is ESP8266EX
Features: WiFi
MAC: 5c:cf:7f:02:9a:b7
WARNING: ROM doesn't support changing baud rate. Keeping initial baud rate 115200
Enabling default SPI flash mode...
Configuring flash size...
Erasing flash...
Took 1.93s to erase flash block

Writing at 0x00001000... (0 %)
Writing at 0x00001400... (0 %)
Writing at 0x00001800... (1 %)
Writing at 0x00001c00... (1 %)
Writing at 0x00002000... (2 %)
.
.
.
Writing at 0x0003a000... (99 %)
Writing at 0x0003a400... (99 %)
Writing at 0x0003a800... (100 %)
Wrote 236544 bytes at 0x00001000 in 24.9 seconds (75.9 kbit/s)...

Leaving...
Hard resetting via RTS pin...
```

Whereas with the stub operating it's:
```
d:/xtools/bin/esptool.py -p /dev/ttyS3 -b 460800 write_flash -ff 40m -fm qio -fs 4MB-c1 0x01000 firmware/upgrade/user1.4096.new.6.bin
esptool.py v2.6-beta1
Serial port /dev/ttyS3
Connecting........_
Detecting chip type... ESP8266
Chip is ESP8266EX
Features: WiFi
MAC: 5c:cf:7f:02:9a:b7
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 460800
Changed.
Configuring flash size...
Compressed 235636 bytes to 173363...

Writing at 0x00001000... (9 %)
Writing at 0x00005000... (18 %)
Writing at 0x00009000... (27 %)
Writing at 0x0000d000... (36 %)
Writing at 0x00011000... (45 %)
Writing at 0x00015000... (54 %)
Writing at 0x00019000... (63 %)
Writing at 0x0001d000... (72 %)
Writing at 0x00021000... (81 %)
Writing at 0x00025000... (90 %)
Writing at 0x00029000... (100 %)
Wrote 235636 bytes (173363 compressed) at 0x00001000 in 4.5 seconds (effective 417.2 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
```

Very big diff

projectgus, 2018-12-21:
PS Yes sorry I was wrong about the baud rate behaviour with --no-stub - on ESP8266 it's always 115200 if --no-stub is set.