espressif/esptool #381: Failed to connect to ESP32: Timed out waiting for packet header on a custom board

lemonbuzz, 2018-11-24:
* Operating system:
mac OS
* Python version:
2.7.15

# Full esptool.py command line as run:
`esptool.py --port /dev/cu.usbserial-AI05M04C --before no_reset --after no_reset write_flash 0x1000 firmware.bin > log.txt`

# Full output from esptool.py (please copy and paste all lines of output)
Pretty big output, so here it is : [log.txt](https://github.com/espressif/esptool/files/2612377/log.txt)

# What is the expected behaviour?
Successful flashing 

# Do you have any other information from investigating this?
I'm not the only one who experienced this issue and they all seem to have resolved the issue in different ways because of varying problems. 

I know this is not an issue with esptool, i'm essentially seeking your help if possible. Otherwise please steer me in the right direction to get the support I need.

Additional details:

-ESP-WROOM-32 on a custom board
-GPIO0 held LOW when booting ( I  **DO** get the 'waiting for download prompt' )
```
rst:0x1 (POWERON_RESET),boot:0x3 (DOWNLOAD_BOOT(UART0/UART1/SDIO_REI_REO_V2))
waiting for download
```
-I have another ESP-WROOM-32 but it's installed on a dev board with EN/BOOT buttons that I can
flash successfully by pressing the BOOT button when the upload process starts
-I tried on a Windows machine with the latest FTDI CP12xx drivers and I'm facing the same upload issue.

Any help or suggestion on what I should try is greatly appreciated :)

gbdb, 2018-11-25:
Hi, 

I'm working on this with Lemonbuzz, here is the schematic that we use..

here is the main chip decoupling and EN pin https://i.imgur.com/T6Fp3Nm.png
IO12 is simply for external power button (even if called ResetSwitch) with a pull down.

here is our RX/TX header ; https://i.imgur.com/8foyFLr.png

I do have a onboard battery for RTC clock, (sharing just in case); https://i.imgur.com/Fd07B74.png
powering from 12volts; https://i.imgur.com/iq2hxK0.png

the rest of circuit is specific to our needs... 

the chip we use is a brand new module from digikey ; 1904-1024-1-ND 
https://www.digikey.ca/product-detail/en/espressif-systems/ESP32-WROOM-32D-(8MB)/1904-1024-1-ND/9381733

we try to flash using FT232R usb module from sunfounder.. 

please help us finding the way to flash this board :)

davydnorris, 2018-11-25:
What version of esptool are you using? It may be useful to try the Espressif supplied GUI version of the flash tool, as I have also been getting the same errors and corruptions with the latest versions of esptool on an ESP8266 unit. Perhaps they are related?

projectgus, 2018-11-26:
Hi @lemonbuzz @gbdb ,

Thanks for the comprehensive bug report and details.

The log shows that esptool is sending a lot of "sync" messages to the ESP32, but absolutely no response is received in the opposite direction.

Given you can see the "waiting for download" message, we know the correct download mode is being established and that communication from the ESP32 to the FT232R is working correctly.

The most likely explanation I can come up with is that communication in the opposite direction is not getting through, so the messages sent by esptool are never being received.

This may be because of a connection issue (short, cold solder joint) with the U0RXD pin of the module. If you put an oscilloscope probe on this pad of the WROOM, do you see the serial communications received correctly? (There will be a lot of square waves because of the 0x55 'U' being sent).

lemonbuzz, 2018-11-26:
Hey @projectgus ,

First of all, thank you for your timely response, you are very appreciated!

@gbdb took a closer look at the circuit this morning and it was indeed a case of cold solder joint on the RX pin!

Now it works like a charm! :)

Thank you.

closed 2018-11-26 by lemonbuzz

projectgus, 2018-11-27:
Great to hear :)