espressif/esptool #384: write timeout on macOS

RockfordWei, 2018-12-03:
(Please delete any lines which don't apply.)

* Operating system: macOS Mojave 10.14.1
* Python version: (`python -V` to check this) 3.6.5

# Full esptool.py command line as run:
`python3 /usr/local/lib/python3.6/site-packages/esptool.py --trace --port /dev/tty.usbmodem14101 write_flash --flash_mode dio --flash_size 1MB --flash_freq 26m  0x00000 QBIQ-ESP-FLASH-ALL-v1.1.51.bin`

# Full output from esptool.py (please copy and paste all lines of output)

```
esptool.py v2.5.1
Serial port /dev/tty.usbmodem14101
Connecting...TRACE +0.000 command op=0x08 data len=36 wait_response=1 timeout=0.100 data=
    0707122055555555 5555555555555555 | ... UUUUUUUUUUUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    55555555                          | UUUU
TRACE +0.000 Write 46 bytes: 
    c000082400000000 0007071220555555 | ...$........ UUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    5555555555555555 5555555555c0     | UUUUUUUUUUUUU.
TRACE +0.104 Timed out waiting for packet header
.TRACE +0.052 command op=0x08 data len=36 wait_response=1 timeout=0.100 data=
    0707122055555555 5555555555555555 | ... UUUUUUUUUUUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    55555555                          | UUUU
TRACE +0.000 Write 46 bytes: 
    c000082400000000 0007071220555555 | ...$........ UUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    5555555555555555 5555555555c0     | UUUUUUUUUUUUU.
TRACE +0.024 Read 1 bytes: c0
TRACE +0.000 Read 95 bytes: 
    0108020007071220 0000c0c001080200 | ....... ........
    070712200000c0c0 0108020007071220 | ... ........... 
    0000c0c001080200 070712200000c0c0 | ........... ....
    0108020007071220 0000c0c001080200 | ....... ........
    070712200000c0c0 0108020007071220 | ... ........... 
    0000c0c001080200 070712200000c0   | ........... ...
TRACE +0.000 Received full packet: 01080200070712200000
TRACE +0.000 Received full packet: 01080200070712200000
TRACE +0.000 Received full packet: 01080200070712200000
TRACE +0.000 Received full packet: 01080200070712200000
TRACE +0.000 Received full packet: 01080200070712200000
TRACE +0.000 Received full packet: 01080200070712200000
TRACE +0.000 Received full packet: 01080200070712200000
TRACE +0.000 Received full packet: 01080200070712200000

Detecting chip type...TRACE +0.000 command op=0x0a data len=4 wait_response=1 timeout=3.000 data=78000060
TRACE +0.000 Write 14 bytes: c0000a04000000000078000060c0
TRACE +0.018 Read 1 bytes: c0
TRACE +0.000 Read 11 bytes: 010a0200002006000000c0
TRACE +0.000 Received full packet: 010a0200002006000000
 ESP8266
TRACE +0.000 command op=0x0a data len=4 wait_response=1 timeout=3.000 data=5c00f03f
TRACE +0.000 Write 14 bytes: c0000a0400000000005c00f03fc0
TRACE +0.019 Read 1 bytes: c0
TRACE +0.000 Read 11 bytes: 010a02007fcf5c000000c0
TRACE +0.000 Received full packet: 010a02007fcf5c000000
TRACE +0.000 command op=0x0a data len=4 wait_response=1 timeout=3.000 data=5800f03f
TRACE +0.000 Write 14 bytes: c0000a0400000000005800f03fc0
TRACE +0.019 Read 1 bytes: c0
TRACE +0.000 Read 11 bytes: 010a020000b000640000c0
TRACE +0.000 Received full packet: 010a020000b000640000
TRACE +0.000 command op=0x0a data len=4 wait_response=1 timeout=3.000 data=5400f03f
TRACE +0.000 Write 14 bytes: c0000a0400000000005400f03fc0
TRACE +0.020 Read 1 bytes: c0
TRACE +0.000 Read 11 bytes: 010a0200cdc800020000c0
TRACE +0.000 Received full packet: 010a0200cdc800020000
TRACE +0.000 command op=0x0a data len=4 wait_response=1 timeout=3.000 data=5000f03f
TRACE +0.000 Write 14 bytes: c0000a0400000000005000f03fc0
[... 855 lines cut ...]
Detecting chip type... ESP8266
Chip is ESP8266EX
Features: WiFi
MAC: 5c:cf:7f:c8:cd:8d
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Flash params set to 0x0221
Compressed 1048576 bytes to 400654...
Wrote 1048576 bytes (400654 compressed) at 0x00000000 in 38.5 seconds (effective
 218.1 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
```

# Do you have any other information from investigating this?

Detail: The same command runs well on windows 7/64bit (a virtual machine) python 3.7.1 but failure on the macOS - so there should be nothing wrong with the command line, but must be an error either the esptool.py or pyserial library.

# Is there any other information you can think of which will help us reproduce this problem?

I just followed `sudo pip install esptool` and run the script then failed like this.

projectgus, 2018-12-03:
Hi @RockfordWei ,

What USB/serial hardware do you have?

The write timeout is set at 10 seconds to detect stuck or misbehaving OS-level drivers. (The command is only writing 6547 bytes which should take less than half a second at 115200bps, so a ten second timeout indicates something went badly wrong and it would have hung forever otherwise.)

Suggest checking if there is an update available for the macOS serial driver of whatever USB/serial chip is in use.

Angus

RockfordWei, 2018-12-03:
Thanks for asking @projectgus , I am not too sure how to get the hardware information or it is related to the hardware since I am actually using the macOS and found this issue, which windows 7 VM(virtual box) runs well on the same MacBook - if the hardware or USB driver could not run, how could the VM do the same job well? It seems that the only difference may be the implementation of pyserial lib, so generally speaking to me, it is very confusing.

```
USB 3.0 Bus:

  Host Controller Driver:	AppleUSBXHCILPTH
  PCI Device ID:	0x8c31 
  PCI Revision ID:	0x0005 
  PCI Vendor ID:	0x8086 

EFM32 Happy Gecko USB CDC serial port device:

  Product ID:	0x0003
  Vendor ID:	0x10c4  (Silicon Laboratories, Inc.)
  Version:	0.00
  Speed:	Up to 12 Mb/sec
  Manufacturer:	Silicon Laboratories Inc.
  Location ID:	0x14100000 / 5
  Current Available (mA):	500
  Current Required (mA):	100
  Extra Operating Current (mA):	0

```

RockfordWei, 2018-12-03:
Just confirmed that ubuntu 18.04 VM can also run esptool well on the same macOS by the same command line :-( why macOS itself has the issue?

projectgus, 2018-12-03:
@RockfordWei If you connect the USB device via the VM then it uses the VM guest's USB driver (Windows driver for Windows guest, Linux driver for Ubuntu guest.) So I would suggest the problem is the interaction between the macOS USB serial driver and the USB device.

An EFM32 microcontroller CDC device is not something I've come across before. As this is a microcontroller, there may be a bug in the EFM32 firmware which is triggered by the USB CDC driver in macOS (USB CDC is a common standard, so there's one driver for all CDC devices).

What hardware is this that uses the EFM32 as a serial interface?

RockfordWei, 2018-12-03:
Sounds reasonable. Currently my project is using an EFM32HG310F64 as the micro controller, which allows to boot the ESP to have a direct USB connection, so where can I find a proper driver? all windows and linux VM are using the same CDC driver for different OS actually

projectgus, 2018-12-03:
USB CDC is a common standard so there's no alternative driver for a given platform.

If the micro is running your own custom firmware, I'm afraid you'll need to debug this yourself - something the firmware is doing is probably causing the macOS CDC driver to think the port is unavailable, and stalling the write at the OS level. There's nothing that can be done about this from the esptool or pyserial side.

RockfordWei, 2018-12-04:
although I don't think it is a firmware case, thanks anyway. Before closing this issue, may I ask if there is any document about the write flash protocol between the host machine and ESP? If I had to debug the driver myself, at least I should know what data the host should send and what feedback it expects from ESP.

projectgus, 2018-12-04:
This is the flash protocol:
https://github.com/espressif/esptool/wiki/Serial-Protocol

You can use the "--trace" option with a working board to see what is sent/received as well.

Keep in mind the "write timeout" is not related to the flashing protocol at all, this is at the lower async serial layer sending/receiving raw bytes. The application is waiting for the OS to say "you can write some bytes to this serial port now", and for some reason the OS never says that the port is available for writing.
https://github.com/pyserial/pyserial/blob/master/serial/serialposix.py#L560

I'm going to close this issue as I don't think there's anything we can do about this in esptool itself.

closed 2018-12-04 by projectgus

RockfordWei, 2018-12-07:
@projectgus it looks like not as your assumption. After reading [this reference](https://stackoverflow.com/questions/39714730/reading-serial-output-from-pyserial-doesnt-work-reliably), I then had a try to modify esptool.py with the regarded solution `self._port.close(); self._port.open()` before the slip_reader() and the chip immediately came back with the `sync` frames:

```
_TRACE +0.055 command op=0x08 data len=36 wait_response=1 timeout=0.100 data=
    0707122055555555 5555555555555555 | ... UUUUUUUUUUUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    55555555                          | UUUU
TRACE +0.000 Write 46 bytes: 
    c000082400000000 0007071220555555 | ...$........ UUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    5555555555555555 5555555555c0     | UUUUUUUUUUUUU.
TRACE +0.031 Read 1 bytes: c0
TRACE +0.000 Read 73 bytes: 
    0108020007071220 0000c0c001080200 | ....... ........
    070712200000c0c0 0108020007071220 | ... ........... 
    0000c0c00108820c 8c20070712200000 | ......... ... ..
    c0c0010802000707 12200000c0c00108 | ......... ......
    0200070712200000 c0               | ..... ...
TRACE +0.000 Received full packet: 01080200070712200000
TRACE +0.000 Received full packet: 01080200070712200000
TRACE +0.000 Received full packet: 01080200070712200000
TRACE +0.000 Received full packet: 0108820c8c20070712200000
TRACE +0.000 Received full packet: 01080200070712200000
TRACE +0.000 Received full packet: 01080200070712200000
```

So I think this could be either a known `pyserial` issue or an improper usage between each write-read session. Could you please suggest the best line to insert the snippet? I am debugging now.

projectgus, 2018-12-17:
> I didn't find self._port.open() in the source code)

The port is already open when it's returned by `serial.serial_from_url()`.

> could be either a known pyserial issue or an improper usage between each write-read session.

If an additional close/open cycle fixes an issue for your particular USB/serial firmware, but this isn't required on any other hardware, then this still indicates that there's a problem with that particular USB/serial firmware. However now you've found a workaround for that problem.

> the chip immediately came back with the sync frames:

I don't really understand why responding to the sync indicates a change of behaviour - the original log you posted also has the chip responding to the sync but then failing part way through a flash. Does that failure no longer happen?

Can you post the full output when you run the same command that's given in the original issue report, with your patched esptool.py? Also please post the output of `git diff` to show the exact changes.

RockfordWei, 2018-12-17:
This is what I patched:

```
diff --git a/esptool.py b/esptool.py
index b35ab95..99816cd 100755
--- a/esptool.py
+++ b/esptool.py
@@ -1844,6 +1844,8 @@ def slip_reader(port, trace_function):
     """
     partial_packet = None
     in_escape = False
+    port.close()
+    port.open()
     while True:
         waiting = port.inWaiting()
         read_bytes = port.read(1 if waiting == 0 else waiting)
```

This is the command line:
```
esptool.py --trace --port /dev/tty.usbmodem142101 --baud 230400 write_flash --flash_freq 26m --flash_size 1MB --flash_mode dio 0x7E000 7e.bin
```

And this is the output (dead loop)

```
esptool.py v2.6-beta1
Serial port /dev/tty.usbmodem142101
Connecting...TRACE +0.000 command op=0x08 data len=36 wait_response=1 timeout=0.100 data=
    0707122055555555 5555555555555555 | ... UUUUUUUUUUUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    55555555                          | UUUU
TRACE +0.001 Write 46 bytes: 
    c000082400000000 0007071220555555 | ...$........ UUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    5555555555555555 5555555555c0     | UUUUUUUUUUUUU.
TRACE +0.116 Timed out waiting for packet header
.TRACE +0.055 command op=0x08 data len=36 wait_response=1 timeout=0.100 data=
    0707122055555555 5555555555555555 | ... UUUUUUUUUUUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    55555555                          | UUUU
TRACE +0.000 Write 46 bytes: 
    c000082400000000 0007071220555555 | ...$........ UUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    5555555555555555 5555555555c0     | UUUUUUUUUUUUU.
TRACE +0.028 Read 1 bytes: c0
TRACE +0.000 Read 78 bytes: 
    0108020007071220 0000c0c001080200 | ....... ........
    070712200000c0c0 0108020007071220 | ... ........... 
    0000c0c001080200 070712200000c0c0 | ........... ....
    0108020007008201 0802000707122000 | .............. .
    00c0c00108020007 0712200000c0     | .......... ...
TRACE +0.001 Received full packet: 01080200070712200000
TRACE +0.000 Received full packet: 01080200070712200000
TRACE +0.000 Received full packet: 01080200070712200000
TRACE +0.000 Received full packet: 01080200070712200000
TRACE +0.000 Received full packet: 
    0108020007008201 0802000707122000 | .............. .
    00                                | .
TRACE +0.000 Received full packet: 01080200070712200000
TRACE +3.000 Timed out waiting for packet header
.TRACE +0.056 command op=0x08 data len=36 wait_response=1 timeout=0.100 data=
    0707122055555555 5555555555555555 | ... UUUUUUUUUUUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    55555555                          | UUUU
TRACE +0.000 Write 46 bytes: 
    c000082400000000 0007071220555555 | ...$........ UUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    5555555555555555 5555555555c0     | UUUUUUUUUUUUU.
TRACE +0.028 Read 1 bytes: c0
TRACE +0.000 Read 80 bytes: 
    0108020007071220 0000c0c001080200 | ....... ........
    070712200000c0c0 0108020007078480 | ... ............
    c001080200070712 200000c0c0010802 | ........ .......
    00070712200000c0 c001080200070712 | .... ...........
    200000c0c0010802 00070712200000c0 |  ........... ...
TRACE +0.000 Received full packet: 01080200070712200000
TRACE +0.000 Received full packet: 01080200070712200000
TRACE +0.000 Received full packet: 0108020007078480
TRACE +0.000 Read invalid data: 
    0108020007071220 0000c0c001080200 | ....... ........
    070712200000c0c0 0108020007078480 | ... ............
    c001080200070712 200000c0c0010802 | ........ .......
    00070712200000c0 c001080200070712 | .... ...........
    200000c0c0010802 00070712200000c0 |  ........... ...
TRACE +0.000 Remaining data in serial buffer: 
.TRACE +0.051 command op=0x08 data len=36 wait_response=1 timeout=0.100 data=
    0707122055555555 5555555555555555 | ... UUUUUUUUUUUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    55555555                          | UUUU
TRACE +0.000 Write 46 bytes: 
    c000082400000000 0007071220555555 | ...$........ UUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    5555555555555555 5555555555c0     | UUUUUUUUUUUUU.
TRACE +0.028 Read 1 bytes: c0
TRACE +0.000 Read 81 bytes: 
    0108020007071220 0000c0c001080200 | ....... ........
    070712200000c0c0 0108020007071220 | ... ........... 
    0000c0c001060707 12200000c0c00108 | ......... ......
    0200070712200000 c0c0010802000707 | ..... ..........
    12200000c0c00108 0200070712200000 | . ........... ..
    c0                                | .
TRACE +0.000 Received full packet: 01080200070712200000
TRACE +0.000 Received full packet: 01080200070712200000
TRACE +0.000 Received full packet: 01080200070712200000
TRACE +0.000 Received full packet: 0106070712200000
TRACE +0.000 Received full packet: 01080200070712200000
TRACE +0.000 Received full packet: 01080200070712200000
TRACE +0.000 Received full packet: 01080200070712200000
```

projectgus, 2018-12-17:
> And this is the output (dead loop)

Do you mean it hangs after the last line of output?

If so, all that closing and reopening the port has done is reset the write timeout. So instead of hanging for 10 seconds and then failing due to the timeout, it's hanging forever.

RockfordWei, 2018-12-18:
No, dead loop means it is synchronizing back and forth forever, sending sync packets and receiving sync packets back

projectgus, 2018-12-18:
I'm not sure what your modification has done, then. I also don't see how this is progress towards solving your original problem, where sync succeeded but the flashing process failed randomly later on.

projectgus, 2018-12-20:
If the initial connect attempt fails, the connect attempt starts again which means it calls slip_reader() again. If your board uses RTS/DTR to reset then possibly closing and reopening the port is causing the RTS pin to toggle and resetting the board, probably into the wrong mode, setting it up for a subsequent failure and the whole process starts again.

I would be surprised if it's really in an infinite loop. After a certain number of tries esptool will fail the connection.