espressif/esptool #385: "Timed out waiting for packet header" with all of the commands

veiico, 2018-12-09:
* Operating system: Ubuntu 17.10
* Python version: 2.7.14
* ESP hardware in use: ESP-12S

# Full esptool.py command line as run:
sudo python ./esptool.py --port /dev/ttyACM0 -b 9600 -t chip_id

# Full output from esptool.py (please copy and paste all lines of output)
```
esptool.py v2.5.1
Serial port /dev/ttyACM0
Connecting...TRACE +0.000 command op=0x08 data len=36 wait_response=1 timeout=0.100 data=
    0707122055555555 5555555555555555 | ... UUUUUUUUUUUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    55555555                          | UUUU
TRACE +0.001 Write 46 bytes: 
    c000082400000000 0007071220555555 | ...$........ UUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    5555555555555555 5555555555c0     | UUUUUUUUUUUUU.
TRACE +0.101 Timed out waiting for packet header
.TRACE +0.050 command op=0x08 data len=36 wait_response=1 timeout=0.100 data=
    0707122055555555 5555555555555555 | ... UUUUUUUUUUUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    55555555                          | UUUU
TRACE +0.001 Write 46 bytes: 
    c000082400000000 0007071220555555 | ...$........ UUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    5555555555555555 5555555555c0     | UUUUUUUUUUUUU.
TRACE +0.101 Timed out waiting for packet header
.TRACE +0.051 command op=0x08 data len=36 wait_response=1 timeout=0.100 data=
    0707122055555555 5555555555555555 | ... UUUUUUUUUUUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    55555555                          | UUUU
TRACE +0.001 Write 46 bytes: 
    c000082400000000 0007071220555555 | ...$........ UUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    5555555555555555 5555555555c0     | UUUUUUUUUUUUU.
TRACE +0.101 Timed out waiting for packet header
.TRACE +0.051 command op=0x08 data len=36 wait_response=1 timeout=0.100 data=
    0707122055555555 5555555555555555 | ... UUUUUUUUUUUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    55555555                          | UUUU
TRACE +0.001 Write 46 bytes: 
    c000082400000000 0007071220555555 | ...$........ UUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    5555555555555555 5555555555c0     | UUUUUUUUUUUUU.
TRACE +0.101 Timed out waiting for packet header
.TRACE +0.050 command op=0x08 data len=36 wait_response=1 timeout=0.100 data=
    0707122055555555 5555555555555555 | ... UUUUUUUUUUUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    55555555                          | UUUU
TRACE +0.001 Write 46 bytes: 
    c000082400000000 0007071220555555 | ...$........ UUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    5555555555555555 5555555555c0     | UUUUUUUUUUUUU.
TRACE +0.101 Timed out waiting for packet header
```
# What is the expected behaviour?
Should show chip_id

# Do you have any other information from investigating this?
- Voltage is perfect through the whole circuit, so no energy drainage problems
- None of the commands (chip_id, flash_id, read_mac, write_flash) work, all of them spill out the same output.
- Tried with --before no_reset - says invalid head for packets every time for every cmd
- I'm quite new to ESP and microprocessors :/

# Is there any other information you can think of which will help us reproduce this problem?
I'm flashing throught arduino pro micro, using 3.3v regulator (getting 3.25-3.3v through out the circuit)

reithose, 2018-12-12:
Same error here after running: 
`esptool.py erase_flash`

```
esptool.py v2.5.1:
Found 3 serial ports
Serial port /dev/cu.wchusbserialfd120
Connecting....
Detecting chip type... ESP8266
Chip is ESP8266EX
etc..
Erasing flash (this may take a while)...

A fatal error occurred: Timed out waiting for packet content
```
I have a WEMOS D1 R2.

davydnorris, 2018-12-13:
@reithose said:
> Same error here after running:
> `esptool.py erase_flash`
>
> ```
> esptool.py v2.5.1:
> Found 3 serial ports
> Serial port /dev/cu.wchusbserialfd120
> Connecting....
> Detecting chip type... ESP8266
> Chip is [...]

I think your issue is slightly different, as you're getting an initial response from the unit (the chip type is returning values).

There are a series of problems being reported where esptool isn't flashing units that have gone into deep sleep or are stuck in a wdt reset loop (see #193, #351 and #380). In my case I found that the nodemcu flash tool worked all the time, but that I could use esptool by following the hard reset described by @projectgus in #380 

Obviously this is not an ideal approach and it would be great to work out how the nodemcu flash utility is managing to flash every time, but can you try the hard reset and see if that allows you to flash?

projectgus, 2018-12-17:
Hi @veiico ,

The trace you posted seems to indicate that nothing is being received from the ESP8266 over serial.

How are you resetting the ESP8266 into bootloader mode? This has to be done with either GPIOs or RTS/DTR signals, and I don't think using an Arduino as a usb/serial adapter allows for this. See https://github.com/espressif/esptool/wiki/ESP8266-Boot-Mode-Selection

Are you disabling the Arduino's AVR microcontroller as well, or using it as a serial relay using a sketch?

If you can, try to get hold of dedicated 3.3V USB/serial adapter board or cable to flash with.

Angus

projectgus, 2019-02-08:
Closing due to a lack of updates.

closed 2019-02-08 by projectgus

furqan4545, 2019-05-24:
I Got the solution, watch this video on youtube and u will get it, i suffered from the same problem too but this video resolved the problem, anyways thanks me later, the link is below:
https://youtu.be/ji71cHaGW8w