espressif/esptool #388: the file.bin sent to esp32 its different from the read_flash

nandohdc, 2018-12-24:
* Operating system: Ubuntu 18.04 - VirtualBox/ Host: MacOs Mojave
* Python version: Python 2.7.15rc1
* ESP hardware in use: ESP32 DEVKITV1
* hello-world.bin - generated with make in folder examples.
* flash-contents2.bin - download from esp32 with read_flash

Hi,

I've been trying to verify the integrity of the firmware that I have sent to the ESP32.
My goal its to use objdump to compare both files: 1 generated from make (hello-world.bin) and 1 downloaded with read_flash. Can you guys help me?

# Full esptool.py command line as run:
`/esptool.py --chip esp32 --port /dev/ttyUSB0 --baud 115200 --before default_reset --after hard_reset write_flash -z --flash_mode dio --flash_freq 40m --flash_size detect 0x1000 /root/esp/hello_world/build/bootloader/bootloader.bin 0x10000 /root/esp/hello_world/build/hello-world.bin 0x8000 /root/esp/hello_world/build/partitions_singleapp.bin` 
(working)

`esptool.py -p COM5 -b 460800 read_flash 0 0x23c30 flash_contents2.bin`
(working)

`xtensa-esp32-elf-objcopy -D hello-world.elf`
(working)

`xtensa-esp32-elf-objcopy -D -I flash_contents2.bin -O flash_contents2.elf`
(not working)

# Full output from esptool.py (please copy and paste all lines of output)

`xtensa-esp32-elf-objcopy: supported targets: elf32-xtensa-le elf32-xtensa-be elf32-little elf32-big plugin srec symbolsrec verilog tekhex binary ihex`

# Is there any other information you can think of which will help us reproduce this problem?
[flash_contents2.bin.zip](https://github.com/espressif/esptool/files/2707386/flash_contents2.bin.zip)

[hello-world.bin.zip](https://github.com/espressif/esptool/files/2707395/hello-world.bin.zip)

Somehow when I open the hello-world.bin and flash_contets2.bin in VIM and try to compare both, they're different from each other.

projectgus, 2019-01-01:
Hi @nandohdc ,

* The .bin format is an Espressif-only format so the toolchain objdump/objcopy tools don't know how to parse it, there's no way to go from a .bin back to an .elf

* In your read_flash command you're starting from address 0x0 but when writing flash you flash bootloader.bin at offset 0x1000, partitions.bin at 0x8000 and finally hello-world.bin at offset 0x10000. If you read_flash starting from 0x10000 instead, does the output match hello-world.bin?

nandohdc, 2019-01-02:
Hi @projectgus ,

Thank you so much for replying this thread.
I tried to use read_flash and the output didn't match hello-world.bin.

`esptool.py -p /dev/ttyUSB0 -b 115200 read_flash 0 0x10000 flash_contents.bin`

After this, I tried to compare both files

`ls -all hello-world.bin flash_contents.bin`

Result:
[image]

Both files are different in size and contents, you can see that in print right below.

`vbindiff hello-world.bin flash_contents.bin`

Result:
[image]

I don't know if the security configurations would make any difference but I'm post them here.

Result:
[image]

projectgus, 2019-01-02:
> I tried to use read_flash and the output didn't match hello-world.bin.
>
> esptool.py -p /dev/ttyUSB0 -b 115200 read_flash 0 0x10000 flash_contents.bin

The syntax of [read_flash is address size filename](https://github.com/espressif/esptool/#read-flash-contents-read_flash). So you dumped 64KB of data from offset 0x0.

To read back hello world, run `read_flash 0x10000 <length of hello-world.bin>`.

Angus

nandohdc, 2019-01-03:
Hi @projectgus ,

It worked just fine. 
Sorry for don't know how to read properly, hahaha xD
Thank you so much, Angus!

closed 2019-01-03 by nandohdc