espressif/esptool #391: Unable to write 1MB fw to 1MB Chip

lordneon, 2019-01-06:
* Operating system: Rasbian 9
* Python version: Python 2.7.13
* ESP hardware in use: ESP8266EX 062018 TUBOPH7R72
Flash Chip: PUYA P25D80H  (DS: http://www.lightchip.net.cn/inc/lib/product/download.php?ProId=98 )
# Full esptool.py command line as run:
` esptool.py --port /dev/ttyUSB0  write_flash -fm dout --flash_size=detect 0 esp8266-20180511-v1.9.4.bin`

# Full output from esptool.py (please copy and paste all lines of output)
```

esptool.py v2.5.1
Serial port /dev/ttyUSB0
Connecting....
Detecting chip type... ESP8266
Chip is ESP8266EX
Features: WiFi
MAC: 84:f3:eb:61:6d:f4
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Auto-detected Flash size: 1MB
Flash params set to 0x0320
Compressed 604872 bytes to 394893...
Wrote 604872 bytes (394893 compressed) at 0x00000000 in 43.4 seconds (effective 111.6 kbit/s)...
File  md5: ba846b10e9de1bd349fc0fbd7f96e8aa
Flash md5: 527f29528c13e1ed846673614909870b
MD5 of 0xFF is 70bd056da289c10920f62ea94d3589b7
```

# What is the expected behaviour?
Successfully writing the firmware. 

# Do you have any other information from investigating this?
Writing 512k firmware works fine: 
```
$ esptool.py --port /dev/ttyUSB0  write_flash -fm dout --flash_size=detect 0 esp8266-512k-20180511-v1.9.4.bin
esptool.py v2.5.1
Serial port /dev/ttyUSB0
Connecting....
Detecting chip type... ESP8266
Chip is ESP8266EX
Features: WiFi
MAC: 84:f3:eb:61:6d:f4
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Auto-detected Flash size: 1MB
Flash params set to 0x0320
Compressed 496432 bytes to 330700...
Wrote 496432 bytes (330700 compressed) at 0x00000000 in 34.6 seconds (effective 114.7 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
```
The detected flash chip is different to what is physically on the board:
```
$ esptool.py --port /dev/ttyUSB0 flash_id
esptool.py v2.5.1
Serial port /dev/ttyUSB0
Connecting......
Detecting chip type... ESP8266
Chip is ESP8266EX
Features: WiFi
MAC: 84:f3:eb:61:6d:f4
Uploading stub...
Running stub...
Stub running...
Manufacturer: 85
Device: 6014
Detected flash size: 1MB
Hard resetting via RTS pin...
```

Does anyone have any suggestions on why I am not able to write a 1MB image to a 1MB chip?

projectgus, 2019-01-06:
Hi @lordneon,

It looks like PUYA flash chips have some unusual behaviours around erase sizes. The Arduino ESP8266 project has already added a "quirk" to deal with them in particular:
https://github.com/esp8266/Arduino/issues/5493

If you try either one of these commands, do you have any further success:

* `esptool.py ... write_flash --no-compress ...`
* `esptool.py .. --no-stub write_flash ..`

(For the no-stub option, you'll need to run the verify_flash command afterwards to check the flash succeeded as only the stub reads back the contents and checks the MD5Sum).

I'll try to get hold of an ESP-01 module with a PUYA flash chip myself, for further testing.

lordneon, 2019-01-07:
Hi @projectgus thanks for your response.

So with --no-compress flag I get the same checksum error.

With the --no-stub option I get the fatal error: "A fatal error occurred: Failed to write to target Flash after seq 18 (result was 0108)"

This occurs when I set the baudrate to 9400 as well.

Anything else I can try?

```
$ esptool.py --port /dev/ttyUSB0 -b 9400 --no-stub write_flash -fm dout --flash_size=detect 0 esp8266-20180511-v1.9.4.bin
esptool.py v2.5.1
Serial port /dev/ttyUSB0
Connecting........_____....._____.....
Detecting chip type... ESP8266
Chip is ESP8266EX
Features: WiFi
MAC: 84:f3:eb:61:6d:f4
Enabling default SPI flash mode...
Configuring flash size...
Auto-detected Flash size: 1MB
Erasing flash...
Flash params set to 0x0320
Took 0.37s to erase flash block
Writing at 0x00004800... (3 %)
A fatal error occurred: Failed to write to target Flash after seq 18 (result was 0108)

```

MateusRodCosta, 2019-09-08:
Hello, has any progress been made on this issue?

My friend has a ESP-01 with the puya chip and we have been trying to flash it for some weeks now and the flashing always fails midway.

We have been searching the Web for a solution and it seems that we need either a build of the firmware that supports puya or a flashing tool that knows how to flash the chip correctly.

I don't have the chip on me right now but I would provide any info requested.

radimkarnis, 2020-12-14:
Hello everyone,
it seems like there is a hardware problem with the PUYA chips (as reporter [here](https://www.addicore.com/ESP8266-ESP-01-p/130.htm) and many other places: "_These modules have XTX, Winbond, or GigaDevice flash parts and *not* PUYA flash parts. These modules are not affected by the problems currently seen with PUYA parts._ ").
I would therefore recommend to refrain from using PUYA chips, sorry for your troubles.
I am closing this issue, feel free to reopen if you feel further discussion is needed.

closed 2020-12-14 by radimkarnis