espressif/esptool #392: Issue with flashing code in Pico D4

vijayprakash1, 2019-01-09:
Hi Everyone,
I made a new board with ESP 32 Pico D4 using schematic suggested by ESPressif
My board is ready schematic is attached  and I am trying to flash the code using Arduino but not able to write, I am getting the following error 
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/pico32.html
PLATFORM: Espressif 32 > ESP32 Pico Kit
HARDWARE: ESP32 240MHz 320KB RAM (4MB Flash)
Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF MODES: FINDER(chain) COMPATIBILITY(soft)
Collected 24 compatible libraries
Scanning dependencies...
No dependencies
Retrieving maximum program size .pioenvs\pico32\firmware.elf
Checking size .pioenvs\pico32\firmware.elf
Memory Usage -> http://bit.ly/pio-memory-usage
DATA:    [          ]   4.1% (used 13404 bytes from 327680 bytes)
PROGRAM: [=         ]  13.5% (used 176908 bytes from 1310720 bytes)
Configuring upload protocol...
AVAILABLE: esptool
CURRENT: upload_protocol = esptool
Looking for upload port...
Auto-detected: COM8
Uploading .pioenvs\pico32\firmware.bin
Serial port COM8
Connecting........_____.....__
Chip is ESP32-PICO-D4 (revision 1)
Features: WiFi, BT, Dual Core, Embedded Flash, VRef calibration in efuse
MAC: d8:a0:1d:65:54:fc
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 921600
Changed.
Configuring flash size...
[image]

Warning: Could not auto-detect Flash size (FlashID=0xffffff, SizeID=0xff), defaulting to 4MB
Compressed 15088 bytes to 9755...

A fatal error occurred: Timed out waiting for packet content

projectgus, 2019-02-08:
Hi @vijayprakash1 ,

Thanks for being patient while someone got back to you.

It looks like the flash chip in the PICO-D4 is not responding at all. Usually this indicates a hardware design issue or a board assembly problem.

Suggest checking the QFN is reflowed properly with no cold joints.

The schematic snippet you posted is unfortunately too small to read, so I wasn't able to check it over. However from what I can see, it looks like the bypass capacitors may have an inductor placed between them and the VCC pins. This is not correct - any added inductance for filtering should be on the incoming side of VCC, not between the bypass capacitors and the pins. This may cause power stability problems with explain the failure to flash.

vijayprakash1, 2019-02-08:
Hi @projectgus ,
Thanks for the response. I have solved the problem now my device is working well.
Problem was from my hardware design side. I have connected GPIO16 and GPIO17 (UART2) to SIM module. When I read datasheet in details its saying these two pins GPIO16 and GPIO17 are connected to SPI Flash so I can't use these two pins as peripheral.
When I disconnected these two pins its started working as expected.
Regarding the inductor I will correct in my design.

Note: Earlier I was using ESP32 Wroom for my application thought to replace this by PICO D4 and ESPressif reverse the few connections like (GPIO9 and GPIO10) are connected to SPI Flash in ESP32 Wroom) but In PICO D4 they are using (GPIO16 and GPIO17) or SPI Flash.

projectgus, 2019-02-11:
I see. Thanks for updating and I'm glad you found the solution.

closed 2019-02-11 by projectgus