espressif/esptool #393: ESP-01 with Flash which only supports DOUT

renegat56, 2019-01-09:
(Please delete any lines which don't apply.)

* Operating system: windows 7
* Python version: (`python -V` to check this) 3.6.5
* ESP hardware in use: ESP-01 with SPI flash, which only supports DOUT mode

# What is the expected behaviour?
I try to use the esptool (version 2.6 and 2.7_dev) with the command -fm dout in read_flash or write_flash mode and always get the answer "invalid choice : dout"

How is it possible to write or read such flash with esptool.py?

projectgus, 2019-01-10:
esptool.py uses simple single SPI for all of its operations. The flash mode argument for write_flash is to configure the binary boot mode for when the ESP8266 runs the firmware.

For this reason the `read_flash` command has no `-fm` argument, and doesn't need one. It should work without it.

The `write_flash` command does use this option. It must be placed after `write_flash` on the command line. If it's not working, can you please post the full esptool.py command line that was run?

renegat56, 2019-01-10:
Hi Angus,

thx for the answer. I tried it with the -fm parameter before the write(read)_flash command like -b or -p parameter.
And yes, the reading works without that parameter. Unfortunately if have two different ESP-01 with different flash. One gave the error on the command "esptool.py -p COM4 -b 115200 read_flash 0 0x100000 test.bin" :

A fatal error occurred: Corrupt data, expected 0x1000 bytes but received 0xfff bytes.

So I thought this is related with the flash_mode, but it is not. Strange is, that the ESP-01 with the 'only dout' flash can be read, but the other one with the QIO flash gave the above error.
Nevertheless the Arduino sketch on it is working, but I can not make a .bin backup with esptool.

Do you or anybody else have an explanation for the error message?

Thx!

projectgus, 2019-01-11:
Usually receiving less bytes than expected (in this case 1 less) means there is some problem with the serial link. You could try lowering the baud rate, and check all of the items in the [Troubleshooting](https://github.com/espressif/esptool/#troubleshooting) section.

renegat56, 2019-01-13:
Hi Angus,

here is a result of reading the esp-01 flash with lower baudrate (below 38200 it does not connect at all) 

D:\Apps\ProgrammingTools\ESP8266 Tools\esptool_for_windows>esptool.py -p COM4 -b 38400 read_flash **0x50 0x10** test.bin
esptool.py v2.6
Serial port COM4
Connecting....
Detecting chip type... ESP8266
Chip is ESP8266EX
Features: WiFi
MAC: cc:50:e3:07:d7:7c
Uploading stub...
Running stub...
Stub running...

A fatal error occurred: Corrupt data, expected 0x1000 bytes but received 0xf bytes

Can I read starting from anywhere (0x50) and can I read less than 4096bytes? The error looks strange to me.

Any other recommendation?

Rainer

projectgus, 2019-01-14:
I agree that looks strange. Most ESP8266 development boards should sync down to 9600 if not lower baud rates.

Is it possible there's an electrical problem with the serial connection or power connections, or some issue with the USB/serial interface? Check all wires, especially grounds are tightly connected.

How are you powering the ESP-01?

renegat56, 2019-01-14:
Again thx for the reply. Everything else works fine, writing flash with espressif tool, running applications in micropython, lua or arduino. I'm programmimg esp8266 and esp32 since more than a year and have about 20 running as server or clients for my wlan net.
So I'm sure that everything is electrically ok. It is powered externally, not by usb. I have another ESP-01 which I can read with esptool.py.

So maybe something wrong especially with one board.

But now I found another tool, which read the flash correctly (crosschecked by writing back to another esp-01).
It's called esp_tool.exe and running under windows 7. Do you know that?
[esp_tool_ReadFlash.zip](https://github.com/espressif/esptool/files/2754971/esp_tool_ReadFlash.zip)

projectgus, 2019-02-19:
Hi @renegat56 ,

Thanks for updating. Sorry for the long delay in replying.

It seems like probably something about the serial link hardware or the ESP-01 module is a little unreliable when sending a lot of data. `esptool.py read_flash` sends the data at a high rate, for maximum throughput. Maybe this other tool you found performs the read in a slower fashion.

I'm going to close this because I think it's hardware specific in some way, and I don't think there is anything we can do in esptool.py to fix it.

closed 2019-02-19 by projectgus

snikul, 2019-09-10:
I have the same problem with LOLIN D32 Pro. I cannot back up flash with esptool.py v2.8 on Ubuntu

TheChatty, 2020-11-28:
I can reproduce @projectgus's behaviour with an ESP8285 (within a Gosund SP1). TX/RX are only loosely connected (too small to solder for me) and esptool.py always fails to read flash with the above mentioned Corrupted data error message although it successfully executes "flash_id". The other flash tool always starts to read and sometimes succeeds to read the whole flash. It also uses 115200 baud (my ESP8285 does not even offer to change the flash baud rate).