espressif/esptool #394: A fatal error occurred: Timed out waiting for packet content

AlwinCrysta, 2019-01-10:
Hello Guys,

I am facing the same issue which is reported by many, but I could not find a solution by trying the advised steps under the other threads. Can you please help me out in this...…….

alwin@DESKTOP-T229N9N MINGW32 ~/flash
$ python ~/esp/esp-idf/components/esptool_py/esptool/esptool.py --chip esp32 --port COM4 --baud 115200 --before default_reset --after hard_reset write_flash -z --flash_mode dio --flash_freq 40m --flash_size detect 0x1000 bootloader.bin  0x10000  camera.bin 0x8000 partitions_singleapp.bin
esptool.py v2.6-beta1
Serial port COM4
Connecting....
Chip is ESP32D0WDQ6 (revision 1)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
MAC: 30:ae:a4:f3:10:c0
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Warning: Could not auto-detect Flash size (FlashID=0xffffff, SizeID=0xff), defaulting to 4MB
Compressed 20928 bytes to 12364...

A fatal error occurred: Timed out waiting for packet content

projectgus, 2019-01-11:
Hi @AlwinCrysta,

> Warning: Could not auto-detect Flash size (FlashID=0xffffff, SizeID=0xff), defaulting to 4MB

This message normally means that the ESP32 can't talk to its attached flash chip (the flash chip may be in the module). Check the GPIO pins 6,7,8,9,10 & 11 (which are shared with the flash chip) are not connected to anything external or shorted together, etc.

Ritwik123, 2019-04-13:
i am also facing the same problem. its showing: 
esptool.py v2.6-beta1
Serial port COM3
Connecting........_
Chip is ESP32D0WDQ6 (revision 1)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
MAC: 24:0a:c4:30:fb:84
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Warning: Could not auto-detect Flash size (FlashID=0xffffff, SizeID=0xff), defaulting to 4MB
Compressed 8192 bytes to 47...

A fatal error occurred: Timed out waiting for packet content
A fatal error occurred: Timed out waiting for packet content

projectgus, 2019-04-15:
@Ritwik123 See comment here: https://github.com/espressif/esptool/issues/394#issuecomment-453613886

Ritwik123, 2019-04-15:
> @Ritwik123 See comment here: [#394 (comment)](https://github.com/espressif/esptool/issues/394#issuecomment-453613886)

YES I HAVE GONE THROUGH ALL THIS.. BT NO RESPONDS...

Ritwik123, 2019-04-15:
[image]

projectgus, 2019-04-16:
@Ritwik123 Please stop posting in all caps and copying the same content onto multiple issues.

It looks like the flash chip connected to your ESP32 is not responding at all. This is a hardware issue.

The most likely root cause is that the board is faulty. Has this board ever worked before?

The other (less likely) possibility is that the chip's internal efuses have become corrupted. If you'd like us to check this, please post the output of "espefuse.py summary".

In either case, I'm afraid the solution is probably new hardware.

Ritwik123, 2019-04-17:
no its not working from the beginning till now i have purchased

Ritwik123, 2019-04-17:
how can i be able to get  "espefuse.py summary". ?

projectgus, 2019-04-17:
> how can i be able to get "espefuse.py summary". ?

From the same command prompt where you run esptool.py, run `espefuse.py --port PORT summary` (replace PORT with the name of the serial port you are using.)

Ritwik123, 2019-04-18:
i am unable to run espefuse.py --port COM3 summary.. its showingbad command.. i am using windows10.. and i have also downloaded esptool,espefuse and all  the other required files..
[image]

Ritwik123, 2019-04-18:
[image]

projectgus, 2019-04-23:
Try `cd $IDF_PATH/components/esptool_py/esptool` to change directory into the esptool directory in IDF, and then run `espefuse.py --port PORT summary`

projectgus, 2019-05-28:
Hi @AlwinCrysta ,

Did you end up making any progress with this issue?

Angus

freezyoff, 2019-07-15:
Please help, my esp32 just arrive. 
i try to upload bin but got the same error:
```
Warning: Could not auto-detect Flash size (FlashID=0xffffff, SizeID=0xff), defaulting to 4MB
A fatal error occurred: Timed out waiting for packet content
A fatal error occurred: Timed out waiting for packet content
```

- ESP32 Doit Rev 1
- Windows 10
- Python: 3.7.3
- Phyton serial: 3.4.x

Below my espefuse.py summary:
```
D:\CLOUD DRIVE & REPOSITORIES\Git Repositories\IOT\esptool-2.6>py espefuse.py --port com6 summary
espefuse.py v2.6
Connecting........__
EFUSE_NAME             Description = [Meaningful Value] [Readable/Writeable] (Hex Value)
----------------------------------------------------------------------------------------
Efuse fuses:
WR_DIS                 Efuse write disable mask                          = 0 R/W (0x0)
RD_DIS                 Efuse read disablemask                            = 0 R/W (0x0)
CODING_SCHEME          Efuse variable block length scheme                = 0 R/W (0x0)
KEY_STATUS             Usage of efuse block 3 (reserved)                 = 0 R/W (0x0)

Calibration fuses:
BLK3_PART_RESERVE      BLOCK3 partially served for ADC calibration data  = 0 R/W (0x0)
ADC_VREF               Voltage reference calibration                     = 1100 R/W (0x10)

Security fuses:
FLASH_CRYPT_CNT        Flash encryption mode counter                     = 0 R/W (0x0)
FLASH_CRYPT_CONFIG     Flash encryption config (key tweak bits)          = 0 R/W (0x0)
CONSOLE_DEBUG_DISABLE  Disable ROM BASIC interpreter fallback            = 1 R/W (0x1)
ABS_DONE_0             secure boot enabled for bootloader                = 0 R/W (0x0)
ABS_DONE_1             secure boot abstract 1 locked                     = 0 R/W (0x0)
JTAG_DISABLE           Disable JTAG                                      = 0 R/W (0x0)
DISABLE_DL_ENCRYPT     Disable flash encryption in UART bootloader       = 0 R/W (0x0)
DISABLE_DL_DECRYPT     Disable flash decryption in UART bootloader       = 0 R/W (0x0)
DISABLE_DL_CACHE       Disable flash cache in UART bootloader            = 0 R/W (0x0)
BLK1                   Flash encryption key
  = 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 R/W
BLK2                   Secure boot key
  = 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 R/W
BLK3                   Variable Block 3
  = 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 R/W

Identity fuses:
MAC                    Factory MAC Address
  = 30:ae:a4:ff:02:a0 (CRC 37 OK) R/W
CHIP_VER_REV1          Silicon Revision 1                                = 1 R/W (0x1)
CHIP_VERSION           Reserved for future chip versions                 = 2 R/W (0x2)
CHIP_PACKAGE           Chip package identifier                           = 0 R/W (0x0)

Config fuses:
XPD_SDIO_FORCE         Ignore MTDI pin (GPIO12) for VDD_SDIO on reset    = 0 R/W (0x0)
XPD_SDIO_REG           If XPD_SDIO_FORCE, enable VDD_SDIO reg on reset   = 0 R/W (0x0)
XPD_SDIO_TIEH          If XPD_SDIO_FORCE & XPD_SDIO_REG, 1=3.3V 0=1.8V   = 0 R/W (0x0)
SPI_PAD_CONFIG_CLK     Override SD_CLK pad (GPIO6/SPICLK)                = 0 R/W (0x0)
SPI_PAD_CONFIG_Q       Override SD_DATA_0 pad (GPIO7/SPIQ)               = 0 R/W (0x0)
SPI_PAD_CONFIG_D       Override SD_DATA_1 pad (GPIO8/SPID)               = 0 R/W (0x0)
SPI_PAD_CONFIG_HD      Override SD_DATA_2 pad (GPIO9/SPIHD)              = 0 R/W (0x0)
SPI_PAD_CONFIG_CS0     Override SD_CMD pad (GPIO11/SPICS0)               = 0 R/W (0x0)
DISABLE_SDIO_HOST      Disable SDIO host                                 = 0 R/W (0x0)

Flash voltage (VDD_SDIO) determined by GPIO12 on reset (High for 1.8V, Low/NC for 3.3V).
```

Below my espefuse.py dump:
```
D:\CLOUD DRIVE & REPOSITORIES\Git Repositories\IOT\esptool-2.6>py espefuse.py --port com6 dump
espefuse.py v2.6
Connecting........_____.
EFUSE block 0:
00000000 a4ff02a0 003730ae 0000a000 00001033 00000000 00000004
EFUSE block 1:
00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
EFUSE block 2:
00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
EFUSE block 3:
00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
```

projectgus, 2019-07-15:
Hi @freezyoff ,

Your chip efuses look fine. This warning:
```

Warning: Could not auto-detect Flash size (FlashID=0x0, SizeID=0x0), defaulting to 4MB
```
Suggests that esptool can't talk to the flash chip that's attached to the ESP32.

If there is nothing attached to the development board pins, it's not sitting on a metallic surface or something else which might short the pins out, and the power from USB is stable then it's probably a faulty board, I'm afraid.

freezyoff, 2019-07-15:
thx for fast reply @projectgus , 
please tell (i would very appreciate it if in details), how you could tell efuse is ok?

projectgus, 2019-07-15:
@freezyoff 

> how you could tell efuse is ok?

They look the same as efuses from a brand new ESP32-D0 chip (mostly zeroes, the values which are set look correct).

The "flash read err, 1000" also points to a faulty flash chip, or bad connections between the ESP32 and the flash chip on the board. You almost certainly have a faulty board.

freezyoff, 2019-07-15:
thx @projectgus 

Exactly Faulty Board. :(
I take my board to seller store and bark like angry dog. 
Btw they replace it with new one and give me fault board as souvenir :D

Err-Rahul, 2019-08-05:
EFUSE_NAME             Description = [Meaningful Value] [Readable/Writeable] (Hex Value)
----------------------------------------------------------------------------------------
Security fuses:
FLASH_CRYPT_CNT        Flash encryption mode counter                     = 0 R/W (0x0)
FLASH_CRYPT_CONFIG     Flash encryption config (key tweak bits)          = 0 R/W (0x0)
CONSOLE_DEBUG_DISABLE  Disable ROM BASIC interpreter fallback            = 1 R/W (0x1)
ABS_DONE_0             secure boot enabled for bootloader                = 0 R/W (0x0)
ABS_DONE_1             secure boot abstract 1 locked                     = 0 R/W (0x0)
JTAG_DISABLE           Disable JTAG                                      = 0 R/W (0x0)
DISABLE_DL_ENCRYPT     Disable flash encryption in UART bootloader       = 0 R/W (0x0)
DISABLE_DL_DECRYPT     Disable flash decryption in UART bootloader       = 0 R/W (0x0)
DISABLE_DL_CACHE       Disable flash cache in UART bootloader            = 0 R/W (0x0)
BLK1                   Flash encryption key                              
  = 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 R/W 
BLK2                   Secure boot key                                   
  = 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 R/W 
BLK3                   Variable Block 3                                  
  = 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 R/W 

Efuse fuses:
WR_DIS                 Efuse write disable mask                          = 0 R/W (0x0)
RD_DIS                 Efuse read disablemask                            = 0 R/W (0x0)
CODING_SCHEME          Efuse variable block length scheme                = 0 R/W (0x0)
KEY_STATUS             Usage of efuse block 3 (reserved)                 = 0 R/W (0x0)

Efuse fuses:
WR_DIS                 Efuse write disable mask                          = 0 R/W (0x0)
RD_DIS                 Efuse read disablemask                            = 0 R/W (0x0)
CODING_SCHEME          Efuse variable block length scheme                = 0 R/W (0x0)
KEY_STATUS             Usage of efuse block 3 (reserved)                 = 0 R/W (0x0)

Config fuses:
XPD_SDIO_FORCE         Ignore MTDI pin (GPIO12) for VDD_SDIO on reset    = 0 R/W (0x0)
XPD_SDIO_REG           If XPD_SDIO_FORCE, enable VDD_SDIO reg on reset   = 0 R/W (0x0)
XPD_SDIO_TIEH          If XPD_SDIO_FORCE & XPD_SDIO_REG, 1=3.3V 0=1.8V   = 0 R/W (0x0)
SPI_PAD_CONFIG_CLK     Override SD_CLK pad (GPIO6/SPICLK)                = 0 R/W (0x0)
SPI_PAD_CONFIG_Q       Override SD_DATA_0 pad (GPIO7/SPIQ)               = 0 R/W (0x0)
SPI_PAD_CONFIG_D       Override SD_DATA_1 pad (GPIO8/SPID)               = 0 R/W (0x0)
SPI_PAD_CONFIG_HD      Override SD_DATA_2 pad (GPIO9/SPIHD)              = 0 R/W (0x0)
SPI_PAD_CONFIG_CS0     Override SD_CMD pad (GPIO11/SPICS0)               = 0 R/W (0x0)
DISABLE_SDIO_HOST      Disable SDIO host                                 = 0 R/W (0x0)

Identity fuses:
MAC                    Factory MAC Address                               
  = cc:50:e3:94:b8:7c (CRC 2f OK) R/W 
CHIP_VER_REV1          Silicon Revision 1                                = 1 R/W (0x1)
CHIP_VERSION           Reserved for future chip versions                 = 2 R/W (0x2)
CHIP_PACKAGE           Chip package identifier                           = 0 R/W (0x0)

Calibration fuses:
BLK3_PART_RESERVE      BLOCK3 partially served for ADC calibration data  = 0 R/W (0x0)
ADC_VREF               Voltage reference calibration                     = 1135 R/W (0x5)

Flash voltage (VDD_SDIO) determined by GPIO12 on reset (High for 1.8V, Low/NC for 3.3V).
This is the summary of my device. Please reply with required procedure so it can be reconfigured

projectgus, 2019-08-06:
@Err-Rahul Your ESP32 appears to be configured normally in efuse.

If you are getting the same `Could not auto-detect Flash size (FlashID=0xffffff, SizeID=0xff)` then either something is connected to a GPIO pin that shouldn't be (see earlier comment) or you have faulty hardware.

Mbonea-Mjema, 2019-08-12:
I have the same problem, at the end of the summary it says flash voltage vdd_sdio internal regulator disabled by efuse

projectgus, 2019-08-13:
@Mbonea-Mjema Can you post the full output of "espefuse.py summary" for your board, please? Have you run any espefuse.py commands to change the voltage settings?

bangonkali, 2019-08-24:
I have the same issue @projectgus this is the espefuse.py summary

```
espefuse.py v2.6
Connecting........_
EFUSE_NAME             Description = [Meaningful Value] [Readable/Writeable] (Hex Value)
----------------------------------------------------------------------------------------
Security fuses:
FLASH_CRYPT_CNT        Flash encryption mode counter                     = 0 R/W (0x0)
FLASH_CRYPT_CONFIG     Flash encryption config (key tweak bits)          = 0 R/W (0x0)
CONSOLE_DEBUG_DISABLE  Disable ROM BASIC interpreter fallback            = 1 R/W (0x1)
ABS_DONE_0             secure boot enabled for bootloader                = 0 R/W (0x0)
ABS_DONE_1             secure boot abstract 1 locked                     = 0 R/W (0x0)
JTAG_DISABLE           Disable JTAG                                      = 0 R/W (0x0)
DISABLE_DL_ENCRYPT     Disable flash encryption in UART bootloader       = 0 R/W (0x0)
DISABLE_DL_DECRYPT     Disable flash decryption in UART bootloader       = 0 R/W (0x0)
DISABLE_DL_CACHE       Disable flash cache in UART bootloader            = 0 R/W (0x0)
BLK1                   Flash encryption key
  = 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 R/W
BLK2                   Secure boot key
  = 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 R/W
BLK3                   Variable Block 3
  = 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 R/W

Efuse fuses:
WR_DIS                 Efuse write disable mask                          = 0 R/W (0x0)
RD_DIS                 Efuse read disablemask                            = 0 R/W (0x0)
CODING_SCHEME          Efuse variable block length scheme                = 0 R/W (0x0)
KEY_STATUS             Usage of efuse block 3 (reserved)                 = 0 R/W (0x0)

Config fuses:
XPD_SDIO_FORCE         Ignore MTDI pin (GPIO12) for VDD_SDIO on reset    = 0 R/W (0x0)
XPD_SDIO_REG           If XPD_SDIO_FORCE, enable VDD_SDIO reg on reset   = 0 R/W (0x0)
XPD_SDIO_TIEH          If XPD_SDIO_FORCE & XPD_SDIO_REG, 1=3.3V 0=1.8V   = 0 R/W (0x0)
SPI_PAD_CONFIG_CLK     Override SD_CLK pad (GPIO6/SPICLK)                = 0 R/W (0x0)
SPI_PAD_CONFIG_Q       Override SD_DATA_0 pad (GPIO7/SPIQ)               = 0 R/W (0x0)
SPI_PAD_CONFIG_D       Override SD_DATA_1 pad (GPIO8/SPID)               = 0 R/W (0x0)
SPI_PAD_CONFIG_HD      Override SD_DATA_2 pad (GPIO9/SPIHD)              = 0 R/W (0x0)
SPI_PAD_CONFIG_CS0     Override SD_CMD pad (GPIO11/SPICS0)               = 0 R/W (0x0)
DISABLE_SDIO_HOST      Disable SDIO host                                 = 0 R/W (0x0)

Identity fuses:
MAC                    Factory MAC Address
  = 84:0d:8e:05:d1:6c (CRC 79 OK) R/W
CHIP_VER_REV1          Silicon Revision 1                                = 1 R/W (0x1)
CHIP_VERSION           Reserved for future chip versions                 = 2 R/W (0x2)
CHIP_PACKAGE           Chip package identifier                           = 0 R/W (0x0)

Calibration fuses:
BLK3_PART_RESERVE      BLOCK3 partially served for ADC calibration data  = 0 R/W (0x0)
ADC_VREF               Voltage reference calibration                     = 1107 R/W (0x1)

Flash voltage (VDD_SDIO) determined by GPIO12 on reset (High for 1.8V, Low/NC for 3.3V).
```

Any advise?

bangonkali, 2019-08-25:
Ok, just an update on my end. It seems that I've been using `HC-SR04` and for some reason, upon startup, since it's shared with VIN & Ground of the `NodeMCU-ESP32s` the board has low power. I figured out that if I energized VCC of `HC-SR04` after `NodeMCU-ESP32s` has connected to WiFi then I do not get the error. But this does mean that I had to use a relay mechanism. 😒

yujiy0713, 2019-11-24:
I had the same issue with my own design.
The issue was IO12 I was using for HSPI and I placed 10K pull up.
This pin is a strapping pin for Internal LDO for flash.
By pulling it up, the LDO went to 1.8V mode and the flash did not function. 
In my case, removing 10K pull up from IO12 was the solution.
Hope you can fix yours.

Yuji

projectgus, 2019-11-28:
Closing as the original poster never responded. Please open a new issue if you're still having problems with your board.

closed 2019-11-28 by projectgus

shaswat-dharaiya, 2020-04-17:
Hi I'm having a similar problem.
I'm using ESP32

This is my output from Arduino IDE when I try to upload a simple blink code..

esptool.py v2.6
Serial port /dev/ttyUSB0
Connecting........_
Chip is ESP32D0WDQ6 (revision 1)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
MAC: 3c:71:bf:9d:e6:64
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 921600
Changed.
Configuring flash size...
Warning: Could not auto-detect Flash size (FlashID=0xffffff, SizeID=0xff), defaulting to 4MB
Compressed 8192 bytes to 47...

A fatal error occurred: Timed out waiting for packet content
A fatal error occurred: Timed out waiting for packet content

shaswat-dharaiya, 2020-04-17:
And this is my out from running the command "python espefuse.py --port /dev/ttyUSB0  summary":
espefuse.py v2.8
Connecting........_
EFUSE_NAME             Description = [Meaningful Value] [Readable/Writeable] (Hex Value)
----------------------------------------------------------------------------------------
Security fuses:
FLASH_CRYPT_CNT        Flash encryption mode counter                     = 0 R/W (0x0)
FLASH_CRYPT_CONFIG     Flash encryption config (key tweak bits)          = 0 R/W (0x0)
CONSOLE_DEBUG_DISABLE  Disable ROM BASIC interpreter fallback            = 1 R/W (0x1)
ABS_DONE_0             secure boot enabled for bootloader                = 0 R/W (0x0)
ABS_DONE_1             secure boot abstract 1 locked                     = 0 R/W (0x0)
JTAG_DISABLE           Disable JTAG                                      = 0 R/W (0x0)
DISABLE_DL_ENCRYPT     Disable flash encryption in UART bootloader       = 0 R/W (0x0)
DISABLE_DL_DECRYPT     Disable flash decryption in UART bootloader       = 0 R/W (0x0)
DISABLE_DL_CACHE       Disable flash cache in UART bootloader            = 0 R/W (0x0)
BLK1                   Flash encryption key                              
  = 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 R/W 
BLK2                   Secure boot key                                   
  = 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 R/W 
BLK3                   Variable Block 3                                  
  = 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 R/W 

Efuse fuses:
WR_DIS                 Efuse write disable mask                          = 0 R/W (0x0)
RD_DIS                 Efuse read disablemask                            = 0 R/W (0x0)
CODING_SCHEME          Efuse variable block length scheme                = 0 R/W (0x0)
KEY_STATUS             Usage of efuse block 3 (reserved)                 = 0 R/W (0x0)

Config fuses:
XPD_SDIO_FORCE         Ignore MTDI pin (GPIO12) for VDD_SDIO on reset    = 0 R/W (0x0)
XPD_SDIO_REG           If XPD_SDIO_FORCE, enable VDD_SDIO reg on reset   = 0 R/W (0x0)
XPD_SDIO_TIEH          If XPD_SDIO_FORCE & XPD_SDIO_REG, 1=3.3V 0=1.8V   = 0 R/W (0x0)
CLK8M_FREQ             8MHz clock freq override                          = 51 R/W (0x33)
SPI_PAD_CONFIG_CLK     Override SD_CLK pad (GPIO6/SPICLK)                = 0 R/W (0x0)
SPI_PAD_CONFIG_Q       Override SD_DATA_0 pad (GPIO7/SPIQ)               = 0 R/W (0x0)
SPI_PAD_CONFIG_D       Override SD_DATA_1 pad (GPIO8/SPID)               = 0 R/W (0x0)
SPI_PAD_CONFIG_HD      Override SD_DATA_2 pad (GPIO9/SPIHD)              = 0 R/W (0x0)
SPI_PAD_CONFIG_CS0     Override SD_CMD pad (GPIO11/SPICS0)               = 0 R/W (0x0)
DISABLE_SDIO_HOST      Disable SDIO host                                 = 0 R/W (0x0)

Identity fuses:
MAC                    Factory MAC Address                               
  = 3c:71:bf:9d:e6:64 (CRC 0xd8 OK) R/W 
CHIP_VER_REV1          Silicon Revision 1                                = 1 R/W (0x1)
CHIP_VER_REV2          Silicon Revision 2                                = 0 R/W (0x0)
CHIP_VERSION           Reserved for future chip versions                 = 2 R/W (0x2)
CHIP_PACKAGE           Chip package identifier                           = 0 R/W (0x0)

Calibration fuses:
BLK3_PART_RESERVE      BLOCK3 partially served for ADC calibration data  = 0 R/W (0x0)
ADC_VREF               Voltage reference calibration                     = 1086 R/W (0x12)

shaswat-dharaiya, 2020-04-17:
espefuse.py v2.8
Connecting........_
EFUSE_NAME             Description = [Meaningful Value] [Readable/Writeable] (Hex Value)
----------------------------------------------------------------------------------------
Security fuses:
FLASH_CRYPT_CNT        Flash encryption mode counter                     = 0 R/W (0x0)
FLASH_CRYPT_CONFIG     Flash encryption config (key tweak bits)          = 0 R/W (0x0)
CONSOLE_DEBUG_DISABLE  Disable ROM BASIC interpreter fallback            = 1 R/W (0x1)
ABS_DONE_0             secure boot enabled for bootloader                = 0 R/W (0x0)
ABS_DONE_1             secure boot abstract 1 locked                     = 0 R/W (0x0)
JTAG_DISABLE           Disable JTAG                                      = 0 R/W (0x0)
DISABLE_DL_ENCRYPT     Disable flash encryption in UART bootloader       = 0 R/W (0x0)
DISABLE_DL_DECRYPT     Disable flash decryption in UART bootloader       = 0 R/W (0x0)
DISABLE_DL_CACHE       Disable flash cache in UART bootloader            = 0 R/W (0x0)
BLK1                   Flash encryption key                              
  = 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 R/W 
BLK2                   Secure boot key                                   
  = 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 R/W 
BLK3                   Variable Block 3                                  
  = 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 R/W 

Efuse fuses:
WR_DIS                 Efuse write disable mask                          = 0 R/W (0x0)
RD_DIS                 Efuse read disablemask                            = 0 R/W (0x0)
CODING_SCHEME          Efuse variable block length scheme                = 0 R/W (0x0)
KEY_STATUS             Usage of efuse block 3 (reserved)                 = 0 R/W (0x0)

Config fuses:
XPD_SDIO_FORCE         Ignore MTDI pin (GPIO12) for VDD_SDIO on reset    = 0 R/W (0x0)
XPD_SDIO_REG           If XPD_SDIO_FORCE, enable VDD_SDIO reg on reset   = 0 R/W (0x0)
XPD_SDIO_TIEH          If XPD_SDIO_FORCE & XPD_SDIO_REG, 1=3.3V 0=1.8V   = 0 R/W (0x0)
CLK8M_FREQ             8MHz clock freq override                          = 51 R/W (0x33)
SPI_PAD_CONFIG_CLK     Override SD_CLK pad (GPIO6/SPICLK)                = 0 R/W (0x0)
SPI_PAD_CONFIG_Q       Override SD_DATA_0 pad (GPIO7/SPIQ)               = 0 R/W (0x0)
SPI_PAD_CONFIG_D       Override SD_DATA_1 pad (GPIO8/SPID)               = 0 R/W (0x0)
SPI_PAD_CONFIG_HD      Override SD_DATA_2 pad (GPIO9/SPIHD)              = 0 R/W (0x0)
SPI_PAD_CONFIG_CS0     Override SD_CMD pad (GPIO11/SPICS0)               = 0 R/W (0x0)
DISABLE_SDIO_HOST      Disable SDIO host                                 = 0 R/W (0x0)

Identity fuses:
MAC                    Factory MAC Address                               
  = 3c:71:bf:9d:e6:64 (CRC 0xd8 OK) R/W 
CHIP_VER_REV1          Silicon Revision 1                                = 1 R/W (0x1)
CHIP_VER_REV2          Silicon Revision 2                                = 0 R/W (0x0)
CHIP_VERSION           Reserved for future chip versions                 = 2 R/W (0x2)
CHIP_PACKAGE           Chip package identifier                           = 0 R/W (0x0)

Calibration fuses:
BLK3_PART_RESERVE      BLOCK3 partially served for ADC calibration data  = 0 R/W (0x0)
ADC_VREF               Voltage reference calibration                     = 1086 R/W (0x12)

markuspla, 2020-12-06:
I have this same problem.

###################################################################################
Building in release mode
Retrieving maximum program size .pio\build\esp32doit-devkit-v1\firmware.elf
Checking size .pio\build\esp32doit-devkit-v1\firmware.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [=         ]  12.0% (used 39224 bytes from 327680 bytes)
Flash: [=====     ]  49.3% (used 645958 bytes from 1310720 bytes)
Configuring upload protocol...
AVAILABLE: esp-prog, espota, esptool, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa
CURRENT: upload_protocol = esptool
Looking for upload port...
Auto-detected: COM3
Uploading .pio\build\esp32doit-devkit-v1\firmware.bin
esptool.py v2.6
Serial port COM3
Connecting........_
Chip is ESP32D0WDQ6 (revision 1)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
MAC: b4:e6:2d:97:ee:0d
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 460800
Changed.
Configuring flash size...
Warning: Could not auto-detect Flash size (FlashID=0xffffff, SizeID=0xff), defaulting to 4MB
Compressed 15872 bytes to 10319...
*** [upload] Error 2

A fatal error occurred: Timed out waiting for packet content

############################################################# 

My ESP32 works from begining with oled display SSD1351 (with using wifi). Today I reloaded my ESP with succesfuly (several hundred times at all) until to this error. On board the AMS117 (module supply) is overtemp (burns) with main module (the metal biggest box). I think it is broke.  
I don't know the reason. I did not make a short circuit!

alvarini, 2021-05-17:
Same problem
I have made a custom esp32 pico PCB and I’m not able to upload first code...
my espfuse summary

```

C:\Users\alvar\Desktop\esp-idf>espefuse.py --port COM3 summary
Connecting........_____....._____....._____....._____.....___
Detecting chip type... ESP32
espefuse.py v3.0
EFUSE_NAME (Block)                       Description  = [Meaningful Value] [Readable/Writeable] (Hex Value)
----------------------------------------------------------------------------------------
Calibration fuses:
BLK3_PART_RESERVE (BLOCK0):              BLOCK3 partially served for ADC calibration data   = False R/W (0b0)
ADC_VREF (BLOCK0):                       Voltage reference calibration                      = 1086 R/W (0b10010)

Config fuses:
XPD_SDIO_FORCE (BLOCK0):                 Ignore MTDI pin (GPIO12) for VDD_SDIO on reset     = False R/W (0b0)
XPD_SDIO_REG (BLOCK0):                   If XPD_SDIO_FORCE, enable VDD_SDIO reg on reset    = False R/W (0b0)
XPD_SDIO_TIEH (BLOCK0):                  If XPD_SDIO_FORCE & XPD_SDIO_REG                   = 1.8V R/W (0b0)
CLK8M_FREQ (BLOCK0):                     8MHz clock freq override                           = 60 R/W (0x3c)
SPI_PAD_CONFIG_CLK (BLOCK0):             Override SD_CLK pad (GPIO6/SPICLK)                 = 6 R/W (0b00110)
SPI_PAD_CONFIG_Q (BLOCK0):               Override SD_DATA_0 pad (GPIO7/SPIQ)                = 17 R/W (0b10001)
SPI_PAD_CONFIG_D (BLOCK0):               Override SD_DATA_1 pad (GPIO8/SPID)                = 8 R/W (0b01000)
SPI_PAD_CONFIG_HD (BLOCK0):              Override SD_DATA_2 pad (GPIO9/SPIHD)               = 11 R/W (0b01011)
SPI_PAD_CONFIG_CS0 (BLOCK0):             Override SD_CMD pad (GPIO11/SPICS0)                = 16 R/W (0b10000)
DISABLE_SDIO_HOST (BLOCK0):              Disable SDIO host                                  = False R/W (0b0)

Efuse fuses:
WR_DIS (BLOCK0):                         Efuse write disable mask                           = 0 R/W (0x0000)
RD_DIS (BLOCK0):                         Efuse read disable mask                            = 0 R/W (0x0)
CODING_SCHEME (BLOCK0):                  Efuse variable block length scheme
   = NONE (BLK1-3 len=256 bits) R/W (0b00)
KEY_STATUS (BLOCK0):                     Usage of efuse block 3 (reserved)                  = False R/W (0b0)

Identity fuses:
MAC (BLOCK0):                            Factory MAC Address
   = d8:a0:1d:63:c2:4c (CRC 0x72 OK) R/W
MAC_CRC (BLOCK0):                        CRC8 for factory MAC address                       = 114 R/W (0x72)
CHIP_VER_REV1 (BLOCK0):                  Silicon Revision 1                                 = True R/W (0b1)
CHIP_VER_REV2 (BLOCK0):                  Silicon Revision 2                                 = False R/W (0b0)
CHIP_VERSION (BLOCK0):                   Reserved for future chip versions                  = 0 R/W (0b00)
CHIP_PACKAGE (BLOCK0):                   Chip package identifier                            = 5 R/W (0b101)
MAC_VERSION (BLOCK3):                    Version of the MAC field                           = 0 R/W (0x00)

Security fuses:
FLASH_CRYPT_CNT (BLOCK0):                Flash encryption mode counter                      = 0 R/W (0b0000000)
UART_DOWNLOAD_DIS (BLOCK0):              Disable UART download mode (ESP32 rev3 only)       = False R/W (0b0)
FLASH_CRYPT_CONFIG (BLOCK0):             Flash encryption config (key tweak bits)           = 0 R/W (0x0)
CONSOLE_DEBUG_DISABLE (BLOCK0):          Disable ROM BASIC interpreter fallback             = True R/W (0b1)
ABS_DONE_0 (BLOCK0):                     Secure boot V1 is enabled for bootloader image     = False R/W (0b0)
ABS_DONE_1 (BLOCK0):                     Secure boot V2 is enabled for bootloader image     = False R/W (0b0)
JTAG_DISABLE (BLOCK0):                   Disable JTAG                                       = False R/W (0b0)
DISABLE_DL_ENCRYPT (BLOCK0):             Disable flash encryption in UART bootloader        = False R/W (0b0)
DISABLE_DL_DECRYPT (BLOCK0):             Disable flash decryption in UART bootloader        = False R/W (0b0)
DISABLE_DL_CACHE (BLOCK0):               Disable flash cache in UART bootloader             = False R/W (0b0)
BLOCK1 (BLOCK1):                         Flash encryption key
   = 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 R/W
BLOCK2 (BLOCK2):                         Secure boot key
   = 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 R/W
BLOCK3 (BLOCK3):                         Variable Block 3
   = 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 R/W

Flash voltage (VDD_SDIO) determined by GPIO12 on reset (High for 1.8V, Low/NC for 3.3V).

```

I have followed every step here with no luck,

any thoughts?

robomaniac, 2021-06-03:
> I had the same issue with my own design.
> The issue was IO12 I was using for HSPI and I placed 10K pull up.
> This pin is a strapping pin for Internal LDO for flash.
> By pulling it up, the LDO went to [...]

Thank you! On my PCB, I randomly assign IO12 with a pull up for switch. I remove the pull up and got it working!

juanbomfim22, 2021-07-13:
Had the same error with ESP32-CAM, I couldn't upload into the flash.

My solution ADD upload_speed: `platform.ini`

```
[env:esp32cam]
platform = espressif32
board = esp32cam
framework = arduino
monitor_speed = 115200
upload_speed = 115200     <-- ADD this!
```

With this, I had to make sure both RX/TX leds were blinking (in my case I was using an Arduino to transmit data) and worked!

stuartblair, 2022-02-01:
I had the same issue, but it turned out that it was because I'd forgotten to close a window I'd left open to a monitor process from earlier when I was attempting to flash a new program to the esp32. I killed the monitor window and hey presto, no more "Serial data stream stopped: Possible serial noise or corruption" errors.

limaarduino, 2022-02-06:
> Ola pessoal,
> 
> Estou enfrentando o mesmo problema relatado por muitos, mas não consegui encontrar uma solução tentando as etapas recomendadas nos outros tópicos. Você pode me ajudar nessa……….
> 
> alwin@DESKTOP-T229N9N MINGW32 ~/flash $ python ~/esp/esp-idf/components/esptool_py/esptool/esptool.py --chip esp32 --port COM4 --baud 115200 --before default_reset --after hard_reset write_flash - z --flash_mode dio --flash_freq 40m --flash_size detect 0x1000 bootloader.bin 0x10000 camera.bin 0x8000 partitions_singleapp.bin esptool.py v2.6-beta1 Porta serial COM4 Conectando.... Chip é ESP32D0WDQ6 (revisão 1) Características: WiFi, BT, Dual Core, 240MHz, calibração VRef em efuse, Esquema de codificação Nenhum MAC: 30:ae:a4:f3:10:c0 Carregando stub... Executando stub... Stub em execução... Configurando o tamanho do flash.. . Aviso: não foi possível detectar automaticamente o tamanho do Flash (FlashID=0xffffff, SizeID=0xff), com o padrão de 4 MB Comprimido 20928 bytes para 12364...
> 
> Ocorreu um erro fatal: Tempo esgotado aguardando o conteúdo do pacote

I had this problem and in my case it was the internal flash chip, pin 8, it was cold soldered. I resoldered and flash was detected again.

[image]

ahmedriazku, 2022-02-15:
I had the same issue with an ESP32 builtinto a module. I changed the usb cable and it worked.

dion2619, 2022-03-31:
[image]
 I had the same issue on a custom pcb.

The problem was I/O pin 2 being pulled up to 3.3v.

prawinrai1993, 2022-04-09:
same issue
Compilation complete.
esptool.py v3.0-dev
Serial port COM3
Connecting........_____....._____
Chip is ESP32-D0WD (revision 1)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: 8c:ce:4e:86:a5:58
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 921600
Changed.
Configuring flash size...
Warning: Could not auto-detect Flash size (FlashID=0x0, SizeID=0x0), defaulting to 4MB
Compressed 8192 bytes to 47...
Writing at 0x0000e000... (100 %)
Wrote 8192 bytes (47 compressed) at 0x0000e000 in 0.0 seconds (effective 4096.1 kbit/s)...

A fatal error occurred: Timed out waiting for packet header
Upload error: Failed uploading: uploading error: exit status 2

anamariarincon03, 2023-01-05:
[SOLVED]: ESP32-CAM A fatal error occurred: Timed out waiting for packet content 

Hello, I was working with an ESP32 and this type of error also happened to me. It happened when I was trying to flashing some arduino code, even an empty one. The ESP32 didn't respond and didn't enter in programming mode and also the message "Could not auto-detect flash size" appeared in console.

I managed to fix this problem as follows:

Connect the ESP32 to serial port but disconnect the power of the ESP32.
Upload arduino sketch.
While arduino is loading sketch and before "Conecting..." appears in the console, press the reset button and keep it press.
Connect ESP32 to power while pressing reset button.
Press the programming button to enter ESP32 in programming mode. Then, release buttons.
Automatically, It will start to writing the code into the ESP32.

I hope this would help you too.

dion2619, 2023-01-05:
I solved this problem. You leave the boot mode and En pin N/C then pull
reset to ground for 200ms or so then release.

This enters run mode and the program will run.

Pull boot mode to ground then pull reset to ground and release.

This enters program mode and will upload code through rx tx pins.

I built a sub circuit that does this by controlling two optical isolators
from boot and en pin to ground. Using a tv remote to have a program mode or
run mode button

Or simply a spst switch on boot pin and a push button on en pin works. Or
the push buttons on the esp32 modules if you're not using a custom pcb..

ManojBhakarPCM, 2023-06-10:
i had the same issue(no falsh detected and cant upload).
but then i realised that the pin near the VIN is not misprinted to CMD, it is actually CMD pin. its not "misprinted GND".

ahmedriazku, 2023-06-13:
Hi, may I ask did you get to upload the code?

amirizat, 2023-12-20:
Hello guys. 
After a month of continuous operation, my custom ESP32 board is experiencing persistent reboots, accompanied by the following error.

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
flash read err, 1000
ets_main.c 371 
ets Jun  8 2016 00:22:57

After run efuse summary command "espefuse.py -p COM5 summary" below the result

FLASH_CRYPT_CNT (BLOCK0):                          Flash encryption mode counter                      = 0 R/W (0b0000000)

This recurring issue prompts the question of its root cause and potential solutions. Each time the problem occurs, I find myself having to re-flash the .BIN file to restore normal operation. How to solve it?