espressif/esptool #395: A fatal error occurred: Failed to connect to ESP32: Timed out waiting for packet header

Twistx77, 2019-01-16:
Hi,
First of all, I now this issue is not new and there are many threads talking about this issue. I have read all I could find on the subject and I haven't found a solution and since this is part of a device we plan to sell we need to figure this out.

We are using a custom board and we are implementing automatic reset and boot with two npn transistors as done in most dev boards like this: https://dl.espressif.com/dl/schematics/esp32_devkitc_v4-sch-20180607a.pdf

We get this error when trying to program form the Arduino IDE: 
esptool.py v2.3.1
Connecting........_____....._____....._____....._____....._____....._____....._____....._____....._____.....____An error occurred while uploading the sketch.

The thing is that first prototype of the board worked perfectly with the same conections in pins GPIO0, EN, GPIO2, GPIO15 and GPIO12.

Also in this new board, it works the fist time I connect it ot the PC but then it doesn't work anymore until I unplug and plug it again.(Linux PC). I have also tried in Windows PCs where it doesn't even work the first time I plug it. Also I have tried with different cables.

esptool.py v2.3.1
Connecting....
Chip is ESP32D0WDQ6 (revision (unknown 0xa))
Features: WiFi, BT, Dual Core, VRef calibration in efuse
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 921600
Changed.
Configuring flash size...
Auto-detected Flash size: 4MB
Compressed 8192 bytes to 47...

Writing at 0x0000e000... (100 %)
Wrote 8192 bytes (47 compressed) at 0x0000e000 in 0.0 seconds (effective 22512.5 kbit/s)...
Hash of data verified.
Compressed 15072 bytes to 9733...

Writing at 0x00001000... (100 %)
Wrote 15072 bytes (9733 compressed) at 0x00001000 in 0.1 seconds (effective 1047.3 kbit/s)...
Hash of data verified.
Compressed 174992 bytes to 88364...

Writing at 0x00010000... (16 %)
Writing at 0x00014000... (33 %)
Writing at 0x00018000... (50 %)
Writing at 0x0001c000... (66 %)
Writing at 0x00020000... (83 %)
Writing at 0x00024000... (100 %)
Wrote 174992 bytes (88364 compressed) at 0x00010000 in 1.6 seconds (effective 863.1 kbit/s)...
Hash of data verified.
Compressed 3072 bytes to 144...

Writing at 0x00008000... (100 %)
Wrote 3072 bytes (144 compressed) at 0x00008000 in 0.0 seconds (effective 5754.1 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...

We have checked the power which is perfect and also we have connected a logic analyzer to pins TX, RX RESET, GPIO0, GPIO15, GPIO12 and GPIO2 and haven't found anything relevant apart the fact that the ESP32 doesn't "respond" and that is what it seems to prevent the downloading of the code.

Working sample
[image]
[image]

Not working sample
[image]

We don't see any important differences in the GPIO and Reset signals.... So we can't figure out why the ESP32 is not responding.

You can examine the captures with Saleae software just load the files contained in the following zip.
[Captures.zip](https://github.com/espressif/esptool/files/2766203/Captures.zip)

The only thing that we find different between the boards that we have (Made by us and dev boards bought) Is that the ones that work are revision 0 and the ones that doesn't (Which are three boards with the new design) have ESP32's with revision 0xa. But I have read in some threads that, revision shouldn't be the issue.

Please we need to throw some light into this issue so we can make sure we find a solution for the first small batch of boards. Thank you.

projectgus, 2019-02-08:
What USB to serial adapter chip do you have? If it's CP2102N, then you may be running into this issue: https://github.com/espressif/esptool/issues/399

If a different chip, are you able to share the relevant parts of the schematic?

Twistx77, 2019-02-08:
Sorry, I forgot to respond to this post. Finally we found the solution... it was a mistake in the PCB assembly a 1K pull-up resistor was soldered to the EN pin instead of 10K. We had the correct value in the schematic but there was an error in the assembly process. 

We are using a CP2102N. I hope this can be useful for someone else.

closed 2019-02-08 by Twistx77