espressif/esptool #397: ESP8266 Chip ID: 0x00000000

CurryGuy, 2019-01-18:
Hi, I'm experiencing a problem with an ESP8266 module and I'm not sure if it's a production defect or some kind of bug. The module came not programmed so I flashed it with the AT firmware, and all the procedure went well. However, the firmware seems to be unable to boot. I have investigated a bit, and I found that the reported Chip ID is 0.

* Operating system: Ubuntu 18.04 LTS
* Python version: Python 2.7.15+
* ESP hardware in use: ESP8266EX

# Full esptool.py command line as run:
esptool.py --before no_reset --after no_reset -p  /dev/ttyACM0 chip_id

# Full output from esptool.py (please copy and paste all lines of output)
esptool.py v2.7-dev
Serial port /dev/ttyACM0
Connecting...
Detecting chip type... ESP8266
Chip is ESP8266EX
Features: WiFi
MAC: 18:fe:34:00:00:00
Uploading stub...
Running stub...
Stub running...
Chip ID: 0x00000000

# What is the expected behavior?
I have launched the same command on other modules, and all of them returned a Chip ID different from 0. I don't know if the two things are related, but all the modules with an ID different from 0 are working as expected (I programmed them using the same procedure).

# Do you have any other information from investigating this?
I analyzed the bootloader startup log (the one at 74880 baud) and I found that it's restarting as it's unable to recognize the board as an ESP8266 and I really believe that it has to do with the null ChipId.

```
' 'ets' 'Jan' '' '8' '2013,rst' 'cause:2,' 'boot' 'mode:(3,6)

load' '0x40100000,' 'len' '2592,' 'room' '16' '
ntail' '0
nchksum' '0xf3
load' '0x3ffe8000,' 'len' '764,' 'room' '8' '
tail' '4\r\nchksum' '0x92
load' '0x3ffe82fc,' 'len' '676,' 'room' '4' '
tail' '0\r\nchksum' '0x22
csum' '0x22
2nd' 'boot' 'version' ':' '1.7(5d6f877)
SPI' 'Speed' ':' '40MHz\r\nSPI' 'Mode' ':' 'QIO
SPI' 'Flash' 'Size' '&' 'Map:' '8Mbit(512KB+512KB)
jump' 'to' 'run' 'user1' '@' '1000
Firmware' 'ONLY' 'supports' 'ESP8266!!!
```

projectgus, 2019-02-04:
Hi @CurryGuy ,

Thanks for being patient while someone got back to you.

This is indeed very strange. It does seem like the the ESP8266 got damaged somehow and has lost its internal configuration including the MAC address bits. (The [chip id is the last 3 bytes of the MAC](https://github.com/espressif/esptool/wiki/Advanced-Commands#chip_id).) 

Can you give any details about where you bought this module from? Do you have a photo of the module, ideally showing the ESP8266EX chip?

CurryGuy, 2019-03-12:
Hi @projectgus,

Thanks for your reply, and sorry for the late response.
I don't know the origin of the module, our PCB manufacturer got it from a chinese supplier.
Here's a photo of the module:
[image]

Thank you for your time.

projectgus, 2019-03-13:
Thanks @CurryGuy . So this is the exact chip? The date code suggests it was manufactured in late 2016.

One more favour, could you please run this command `esptool.py --no-stub --trace read_mac` and post the full output? (Will be a few dozen lines.)

projectgus, 2019-03-13:
My colleagues have asked for a few more details:

* Do you have a photo of the whole module, please?
* Are you able to get the name of the distributor?
* Is this faulty module one from a larger lot/package you're using?

projectgus, 2019-10-18:
Closing due to a lack of more information. @CurryGuy please let me know if you have any more information we can use to chase this up.

closed 2019-10-18 by projectgus

l0ud, 2024-07-26:
I happen to have identical unfortunate chip in ESP-12F module.
Bought from there: https://www.aliexpress.com/item/32339917567.html?spm=a2g0o.order_list.order_list_main.45.1d951802gwu8rd

```
esptool.py --no-stub --trace --port /dev/ttyUSB0 read_mac                                                                2 ✘  8s  
esptool.py v4.7.0
Serial port /dev/ttyUSB0
Connecting...TRACE +0.000 command op=0x08 data len=36 wait_response=1 timeout=0.100 data=
    0707122055555555 5555555555555555 | ... UUUUUUUUUUUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    55555555                          | UUUU
TRACE +0.000 Write 46 bytes: 
    c000082400000000 0007071220555555 | ...$........ UUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    5555555555555555 5555555555c0     | UUUUUUUUUUUUU.
TRACE +0.100 No serial data received.
.TRACE +0.050 command op=0x08 data len=36 wait_response=1 timeout=0.100 data=
    0707122055555555 5555555555555555 | ... UUUUUUUUUUUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    55555555                          | UUUU
TRACE +0.000 Write 46 bytes: 
    c000082400000000 0007071220555555 | ...$........ UUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    5555555555555555 5555555555c0     | UUUUUUUUUUUUU.
TRACE +0.008 Read 1 bytes: c0
TRACE +0.000 Read 31 bytes: 
    0108020007071220 0000c0c001080200 | ....... ........
    070712200000c0c0 01080200070712   | ... ...........
TRACE +0.000 Received full packet: 01080200070712200000
TRACE +0.000 Received full packet: 01080200070712200000
TRACE +0.002 Read 1 bytes: 20
TRACE +0.000 Read 31 bytes: 
    0000c0c001080200 070712200000c0c0 | ........... ....
    0108020007071220 0000c0c0010802   | ....... .......
TRACE +0.000 Received full packet: 01080200070712200000
TRACE +0.000 Received full packet: 01080200070712200000
TRACE +0.000 Received full packet: 01080200070712200000
TRACE +0.002 Read 1 bytes: 00
TRACE +0.000 Read 31 bytes: 
    070712200000c0c0 0108020007071220 | ... ........... 
    0000c0c001080200 070712200000c0   | ........... ...
TRACE +0.000 Received full packet: 01080200070712200000
TRACE +0.000 Received full packet: 01080200070712200000
TRACE +0.000 Received full packet: 01080200070712200000

Detecting chip type...TRACE +0.000 command op=0x14 data len=0 wait_response=1 timeout=3.000 data=
TRACE +0.000 Write 10 bytes: c00014000000000000c0
TRACE +0.004 Read 1 bytes: c0
TRACE +0.000 Read 31 bytes: 
    0108020007071220 0105c0c001080200 | ....... ........
    070712200105c0c0 01080200070712   | ... ...........
TRACE +0.000 Received full packet: 01080200070712200105
 Unsupported detection protocol, switching and trying again...
Connecting...TRACE +0.152 command op=0x08 data len=36 wait_response=1 timeout=0.100 data=
    0707122055555555 5555555555555555 | ... UUUUUUUUUUUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    55555555                          | UUUU
TRACE +0.000 Write 46 bytes: 
    c000082400000000 0007071220555555 | ...$........ UUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    5555555555555555 5555555555c0     | UUUUUUUUUUUUU.
TRACE +0.008 Read 1 bytes: c0
TRACE +0.000 Read 31 bytes: 
    0108020007071220 0000c0c001080200 | ....... ........
    070712200000c0c0 01080200070712   | ... ...........
TRACE +0.000 Received full packet: 01080200070712200000
TRACE +0.000 Received full packet: 01080200070712200000
TRACE +0.002 Read 1 bytes: 20
TRACE +0.000 Read 31 bytes: 
    0000c0c001080200 070712200000c0c0 | ........... ....
[... 227 lines cut ...]
TRACE +0.000 Received full packet: 01090200530000000000
TRACE +0.000 command op=0x09 data len=16 wait_response=1 timeout=3.000 data=2402006000000070ffffffff00000000
TRACE +0.000 Write 26 bytes: 
    c000091000000000 0024020060000000 | .........$..`...
    70ffffffff000000 00c0             | p.........
TRACE +0.006 Read 1 bytes: c0
TRACE +0.000 Read 11 bytes: 01090200530000000000c0
TRACE +0.000 Received full packet: 01090200530000000000
TRACE +0.000 command op=0x0a data len=4 wait_response=1 timeout=3.000 data=5000f03f
TRACE +0.000 Write 14 bytes: c0000a0400000000005000f03fc0
TRACE +0.005 Read 1 bytes: c0
TRACE +0.000 Read 11 bytes: 010a0200000000000000c0
TRACE +0.000 Received full packet: 010a0200000000000000
TRACE +0.000 command op=0x0a data len=4 wait_response=1 timeout=3.000 data=5400f03f
TRACE +0.000 Write 14 bytes: c0000a0400000000005400f03fc0
TRACE +0.004 Read 1 bytes: c0
TRACE +0.000 Read 11 bytes: 010a0200000000000000c0
TRACE +0.000 Received full packet: 010a0200000000000000
TRACE +0.000 command op=0x0a data len=4 wait_response=1 timeout=3.000 data=5c00f03f
TRACE +0.000 Write 14 bytes: c0000a0400000000005c00f03fc0
TRACE +0.005 Read 1 bytes: c0
TRACE +0.000 Read 11 bytes: 010a0200000000000000c0
TRACE +0.000 Received full packet: 010a0200000000000000
MAC: 18:fe:34:00:00:00
Hard resetting via RTS pin...