espressif/esptool #399: Upload crashes SILICON LABS CP2102N usb to serial converter on linux

yonie1, 2019-01-24:
After the tool finishes uploading, my usb to serial converter(CP2102N) does not receive any serial data anymore. If I try to upload something again, it fails. This problem can only be solved by unplugging and plugging my esp board back in. I think this happens because the converter gets disattached to quick. I could solve this problem by adding  a time.sleep(2) after hard_reset() like this:
```python
    def hard_reset(self):
        self._setRTS(True)  # EN->LOW
        time.sleep(0.1)
        self._setRTS(False)
        time.sleep(2)
```

projectgus, 2019-02-04:
Unfortunately there seems to be an issue with CP2102N if the port is closed while serial data is still being received.

When the ESP32 comes out of reset (`self._setRTS(False)`) it immediately starts outputting some data (the ROM boot code messages, etc.) It looks like adding the 2 second sleep gets past this point and starts running your just-flashed firmware, but this workaround will only work as long as the chip doesn't output anything on the serial at that exact moment (two seconds after reset).

I reported this to Silicon Labs last year, there's a thread here:
https://www.silabs.com/community/interface/forum.topic.html/cp2102n_stops_receiv-GeH2

They recently updated saying that their VCP drivers were updated. It looks like the fix is in the Windows driver (marked MCUSW-134 in the release notes). I'm not sure if there is any equivalent fix for MacOS or Linux users, yet.

projectgus, 2019-05-28:
Closing as this is a hardware issue with CP2102N, we can work around it in esptool.py depending on the firmware behaviour but we can't resolve it here. :|

closed 2019-05-28 by projectgus

jeremyherbert, 2019-08-15:
FYI this is now fixed in CP2102N rev A02 as far as I can tell. Just made a new board with one and no issues so far.