espressif/esptool #404: erase-all or erase_all ?

sunbearc22, 2019-02-04:
I notice **erase-all** is used in this [line](https://github.com/espressif/esptool/blob/master/esptool.py#L2501) and **erase_all** in this [line](https://github.com/espressif/esptool/blob/master/esptool.py#L2128). Should esptool.py#L2501 be  "erase_all" instead?

projectgus, 2019-02-04:
`erase-all` isn't a valid Python token (Python would parse it as "erase minus all"). The argparse module converts symbols like `-` in command line arguments to `_`, internally when building the Python representation.

(I realise this is confusing right now because esptool.py arguments use a mixture of hyphens and underscores in different commands, an inconsistency dating back to pre-v1.0. I'm trying to standardize on hyphens so eventually we can remove the underscores from command line arguments, hopefully in a backwards compatible way.)

sunbearc22, 2019-02-04:
Can standardize to using underscore instead?

projectgus, 2019-02-04:
Do you have a reason to pick underscores over hyphens?

sunbearc22, 2019-02-04:
While reading esptools.py, I noticed most of `args` attributes requiring spacing btwn words used underscore. E.g. `args.override_vddsdio`, `args.spi_connection`, `args.no_stub`, `args.flash_size`, `args.addr_filename`, ... Also the same convention is used on the attributes of `esp`. the only exception seem to be `ersae-all` ? I think it can make the code more consistent and easier to read.

projectgus, 2019-02-04:
Look again:
https://github.com/espressif/esptool/blob/master/esptool.py#L2453
https://github.com/espressif/esptool/blob/master/esptool.py#L2443
https://github.com/espressif/esptool/blob/master/esptool.py#L2433

OK, this one is an odd one out:
https://github.com/espressif/esptool/blob/master/esptool.py#L2492

closed 2019-02-04 by projectgus

projectgus, 2019-02-04:
Maybe I misunderstood what you're asking for and you'd like all these options to be changed to underscores, the same as `flash_size` currently is.  This isn't a likely change, because hyphens in command line options are a more common convention and most esptool.py users will be use the command line rather than Python.

Sorry for the inconvenience of having most options in two forms (at least soon they'll probably all be consistently in two forms!)

sunbearc22, 2019-02-04:
No issue. I understand from you that for argparse, the options in commandline will appear with hypen. These are then converted to underscore at the backend, hence code-wise attributes of args with more than one word will still be denoted by underscore, e.g. `args.override_vddsdio`, `args.spi_connection`, `args.no_stub`, `args.flash_size`, `args.addr_filename`, `args.erase_all`,...  

Look forward to finally having a consistent convention.  ;)