espressif/esptool #405: error connecting to board

nobody102, 2019-02-10:
(Please delete any lines which don't apply.)

* Operating system: WIndows 10
* Python version: (`python -V` to check this) 3.6.5
* ESP hardware in use: nodemcu esp32-s wroom

# Full esptool.py command line as run: esptool.py --port COM5 erase_flash

# Full output from esptool.py (please copy and paste all lines of output)
C:\Users\amikam>esptool.py --port COM5 erase_flash
esptool.py v2.4.1
Serial port COM5
Connecting...
Traceback (most recent call last):
  File "c:\users\amikam\appdata\local\programs\python\python36-32\lib\site-packages\esptool.py", line 313, in command
    p = self.read()
  File "c:\users\amikam\appdata\local\programs\python\python36-32\lib\site-packages\esptool.py", line 258, in read
    return next(self._slip_reader)
  File "c:\users\amikam\appdata\local\programs\python\python36-32\lib\site-packages\esptool.py", line 1771, in slip_reader
    raise FatalError("Timed out waiting for packet %s" % waiting_for)
esptool.FatalError: Timed out waiting for packet header

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "C:\Users\amikam\AppData\Local\Programs\Python\Python36-32\Scripts\esptool.py-script.py", line 11, in <module>
    load_entry_point('esptool==2.4.1', 'console_scripts', 'esptool.py')()
  File "c:\users\amikam\appdata\local\programs\python\python36-32\lib\site-packages\esptool.py", line 2814, in _main
    main()
  File "c:\users\amikam\appdata\local\programs\python\python36-32\lib\site-packages\esptool.py", line 2515, in main
    esp = ESPLoader.detect_chip(each_port, initial_baud, args.before, args.trace)
  File "c:\users\amikam\appdata\local\programs\python\python36-32\lib\site-packages\esptool.py", line 242, in detect_chip
    detect_port.connect(connect_mode)
  File "c:\users\amikam\appdata\local\programs\python\python36-32\lib\site-packages\esptool.py", line 441, in connect
    last_error = self._connect_attempt(mode=mode, esp32r0_delay=False)
  File "c:\users\amikam\appdata\local\programs\python\python36-32\lib\site-packages\esptool.py", line 421, in _connect_attempt
    self.sync()
  File "c:\users\amikam\appdata\local\programs\python\python36-32\lib\site-packages\esptool.py", line 360, in sync
    timeout=SYNC_TIMEOUT)
  File "c:\users\amikam\appdata\local\programs\python\python36-32\lib\site-packages\esptool.py", line 324, in command
    self._port.timeout = saved_timeout
  File "c:\users\amikam\appdata\local\programs\python\python36-32\lib\site-packages\serial\serialutil.py", line 368, in timeout
    self._reconfigure_port()
  File "c:\users\amikam\appdata\local\programs\python\python36-32\lib\site-packages\serial\serialwin32.py", line 222, in _reconfigure_port
    'Original message: {!r}'.format(ctypes.WinError()))
serial.serialutil.SerialException: Cannot configure port, something went wrong. Original message: OSError(22, 'The I/O operation has been aborted because of either a thread exit or an application request.', None, 995)
# What is the expected behaviour?

# Do you have any other information from investigating this?
I can connect to the board with putty
# Is there any other information you can think of which will help us reproduce this problem?

projectgus, 2019-02-11:
Hi @nobody102 ,

What USB/serial interface chip does this nodemcu board have?

If the board has a BOOT button and you hold it down while running esptool, does anything different happen?

nobody102, 2019-02-12:
I got it working now.  Had to hold the button down for a while.  Thansks!

projectgus, 2019-02-12:
No problems, glad you found a solution. You should be able to release BOOT once the flashing process is underway.

(Normally development boards will automatically go into bootloader mode, but on some combinations of hardware and host computer the automatic reset circuit doesn't work so this kind of workaround is required.)

closed 2019-02-12 by projectgus