espressif/esptool #407: esptool.FatalError: Timed out waiting for packet header

sunbearc22, 2019-02-11:
I encountered this issue `esptool.FatalError: Timed out waiting for packet header` while using the `get_chip_description()` method from the class `esptool.ESP32ROM`. 

```
Traceback (most recent call last):
  File "~/project/test.py", line 217, in _detect_esp_progress_update_v1
    chip_type = self.esp.get_chip_description()
  File "~/.local/share/virtualenvs/Esp32-zoIgntnu/bin/esptool.py", line 1095, in get_chip_description
    word3 = self.read_efuse(3)
  File "~/.local/share/virtualenvs/Esp32-zoIgntnu/bin/esptool.py", line 1159, in read_efuse
    return self.read_reg(self.EFUSE_REG_BASE + (4 * n))
  File "~/.local/share/virtualenvs/Esp32-zoIgntnu/bin/esptool.py", line 477, in read_reg
    val, data = self.command(self.ESP_READ_REG, struct.pack('<I', addr))
  File "~/.local/share/virtualenvs/Esp32-zoIgntnu/bin/esptool.py", line 334, in command
    p = self.read()
  File "~/.local/share/virtualenvs/Esp32-zoIgntnu/bin/esptool.py", line 279, in read
    return next(self._slip_reader)
  File "~/.local/share/virtualenvs/Esp32-zoIgntnu/bin/esptool.py", line 1888, in slip_reader
    raise FatalError("Timed out waiting for packet %s" % waiting_for)
esptool.FatalError: Timed out waiting for packet header
```

The strange part of this issue is that the same device is detectable via terminal command (see below). I have tested my code with the `get_chip_description()` method. It is able to get the description for another similar device. But it constantly fails to work on the same device. **How can I overcome this issue?** 

```
$ esptool.py --chip esp32 --port /dev/ttyUSB2 flash_id
esptool.py v2.6
Serial port /dev/ttyUSB2
Connecting.....
Chip is ESP32D0WDQ6 (revision 1)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
MAC: xx:xx:xx:xx:xx:xx
Uploading stub...
Running stub...
Stub running...
Manufacturer: ef
Device: 4016
Detected flash size: 4MB
Hard resetting via RTS pin...
```

sunbearc22, 2019-02-11:
Below is a section of my code. After introducing the line `self.esp.connect()` there, my code worked. **Why is this the case? Why is .connect() needed before running .get_chip_description()?**

```
        try:
            self.esp = esptool.ESP32ROM( self.port.get(), self.baud.get(),  )
            self.esp.connect()  #I added this line
        except (esptool.FatalError, OSError) as err:
            self.esp = None
            self.detecting = False
            raise
            print( "{} failed to connect to a Espressif device: {}".format( port, err ) )
        else:
            print( 'self.esp = ', self.esp.__dict__ )
            self.detecting = False
            self.status.set( 'Detected' )

```

sunbearc22, 2019-02-11:
I also encountered the same Timeout fatal error during the .slip_reader() function when the .write_flash() function is executed. **May I know what is causing the Timeout error and how to overcome it?** Below is a printout from the .write_flash() function prior to the error.  

```
args.compress =  True

Verify file sizes fit in flash

flash_end = <class 'int'> 4194304
address = <class 'int'> 4096
argfile.tell() = <class 'int'> 1087456

write_flash main loop
Erasing flash...

def pad_to(data, alignment, pad_character=b'ÿ'):
len(data) =  1087456
pad_mod =  0
##Completed pad_to

def _update_image_flash_params(esp, address, args, image):
magic=233, flash_mode=2, flash_size_freq=32
esp.BOOTLOADER_FLASH_OFFSET=4096, address=4096
esp.ESP_IMAGE_MAGIC =  233
flash_mode =  0
flash_freq =  0
flash_size =  32
flash_size =  32
Flash params set to 0x0020
uncsize =  1087456

###Download file to flash Get ratio & block
parity='N', stopbits=1, timeout=None, xonxoff=False, rtscts=False, dsrdtr=False)
compress is True
esp.FLASH_WRITE_SIZE =  1024
esp.IS_STUB =  False
ratio =  1.5819658776156227
num_blocks =  672
erase_blocks =  1062
write_size =  1087488
timeout =  32.62464
Compressed 1087456 bytes to 687408...
Exception in Tkinter callback
 (most recent call last):
  File "~/.pyenv/versions/3.6.7/lib/python3.6/tkinter/__init__.py", line 1705, in __call__
    return self.func(*args)
  File "~/project/test.py", line 451, in _write_flash
    blocks = esp.flash_defl_begin(uncsize, len(image), address)
  File "~/.local/share/virtualenvs/Esp32-zoIgntnu/bin/esptool.py", line 104, in inner
    return func(*args, **kwargs)
  File "~/.local/share/virtualenvs/Esp32-zoIgntnu/bin/esptool.py", line 630, in flash_defl_begin
    timeout=timeout)
  File "~/.local/share/virtualenvs/Esp32-zoIgntnu/bin/esptool.py", line 356, in check_command
    val, data = self.command(op, data, chk, timeout=timeout)
  File "~/.local/share/virtualenvs/Esp32-zoIgntnu/bin/esptool.py", line 334, in command
    p = self.read()
  File "~/.local/share/virtualenvs/Esp32-zoIgntnu/bin/esptool.py", line 279, in read
    return next(self._slip_reader)
  File "~/.local/share/virtualenvs/Esp32-zoIgntnu/bin/esptool.py", line 1890, in slip_reader
    raise FatalError("Timed out waiting for packet %s" % waiting_for)
esptool.FatalError: Timed out waiting for packet header
```

projectgus, 2019-02-11:
>  **Why is this the case? Why is .connect() needed before running .get_chip_description()?**

`connect()` is required because this is the function which actually resets the chip into the bootloader mode and establishes bootloader communications.

A shortcoming of the Python API (to add to notes in in #208) is that the class itself doesn't track the connection state, so there's no software-side indication that it won't respond to the caller at all.

(Although I suppose in some cases it may be desirable to do this from code, for example if you know for a fact that some other piece of code has established bootloader mode already, and you just need to pass control a new ESPLoader new object instance.)

> I also encountered the same Timeout fatal error during the .slip_reader() function when the .write_flash() function is executed. **May I know what is causing the Timeout error and how to overcome it?**

Without seeing that code that you're calling to make this printout I really can't tell anything.

sunbearc22, 2019-02-12:
@projectgus I have just uploaded my code [here](https://github.com/sunbearc22/ESP32FlashWriter). Appreciate your advice on what is causing the timeout fatal error.

Summarising my issuer, I was running [esptool.py write_flash() method] (https://github.com/sunbearc22/ESP32FlashWriter/blob/master/esp32flashwriter_v2.py#L398) and encountered the fatal Timeout error. To understand the matter, I replicated the `write_flash` and a few other relevant esptooly.py functions to my script (see lines 403 to 566) to understand the issue. I traced the issue to originate from [line 455](https://github.com/sunbearc22/ESP32FlashWriter/blob/master/esp32flashwriter_v2.py#L455). According to python's Exception msgs, the matter is traced to slip_reader() finally.

sunbearc22, 2019-02-12:
@projectgus Can someone explain the meaning of the following byte value: `b'\xc0'`, `b'\xdb'`, `b'\xdc'`, `b'\xdd'` used in slip_read() function?? Where is documentation describing these byte values? 

```
>>> int.from_bytes(b'\xc0', byteorder='little')
192
>>> int.from_bytes(b'\xdb', byteorder='little')
219
>>> int.from_bytes(b'\xdc', byteorder='little')
220
>>> int.from_bytes(b'\xdd', byteorder='little')
221
```

projectgus, 2019-02-12:
@sunbearc22 SLIP is https://en.wikipedia.org/wiki/Serial_Line_Internet_Protocol

* \xc0 is the packet delimiter.
* \xdb\xdc is how you represent a literal \xc0 in the stream
* \xdb\xdd is how you represent a literal \xdb in the stream
* all other bytes are represented as themselves in the stream

The `slip_reader()` function is a Python generator that tries to read a full SLIP packet from the input serial stream, decode it, and then yield it to the caller.

As a next step, I suggest you try to enable the equivalent of the `--trace` option in your code and look for differences in the serial traffic between `esptool.py --trace --no-stub write_flash ...` and what your code is doing.

sunbearc22, 2019-02-13:
@projectgus After activating the equivalent of `--no-stub` in my code, i.e. by setting `self.args.no_stub = False` in `WriteFlash()` object, and running `esptool.ESPLoader.run_stub()` before calling the esptool.write_flash(), the fatal timeout error disappeared and my code was able to write to flash. 

However, when access the device via `$ picocom /dev/ttyUSB0 -b 115200`, I end up seeing an endless trial of the same data (see below). In short, I am able to write to flash now but the writing seem erroneous. **Can you tell me what is causing these `1162 mmu set 00010000, pos 00010000`?** Thanks. I have uploaded my [revised code] for your consideration.(https://github.com/sunbearc22/ESP32FlashWriter/blob/master/esp32flashwriter_v2.py) . 

```
$ picocom /dev/ttyUSB0 -b 115200
picocom v1.7

port is        : /dev/ttyUSB0
flowcontrol    : none
baudrate is    : 115200
parity is      : none
databits are   : 8
escape is      : C-a
local echo is  : no
noinit is      : no
noreset is     : no
nolock is      : no
send_cmd is    : sz -vv
receive_cmd is : rz -vv
imap is        : 
omap is        : 
emap is        : crcrlf,delbs,

Terminal ready
�0x00,hd_drv:0x00,wp_drv:0x00
mode:QIO, clock div:2
load:0x3fff0018,len:4
load:0xcdc72100,len:687535457
1162 mmu set 00010000, pos 00010000
1162 mmu set 00020000, pos 00020000
1162 mmu set 00030000, pos 00030000
1162 mmu set 00040000, pos 00040000
1162 mmu set 00050000, pos 00050000
1162 mmu set 00060000, pos 00060000
1162 mmu set 00070000, pos 00070000
1162 mmu set 00080000, pos 00080000
1162 mmu set 00090000, pos 00090000
1162 mmu set 000a0000, pos 000a0000
1162 mmu set 000b0000, pos 000b0000
1162 mmu set 000c0000, pos 000c0000
1162 mmu set 000d0000, pos 000d0000
1162 mmu set 000e0000, pos 000e0000
1162 mmu set 000f0000, pos 000f0000
1162 mmu set 00100000, pos 00100000
1162 mmu set 00110000, pos 00110000
1162 mmu set 00120000, pos 00120000
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:QIO, clock div:2
load:0x3fff0018,len:4
load:0xcdc72100,len:687535457
1162 mmu set 00010000, pos 00010000
1162 mmu set 00020000, pos 00020000
1162 mmu set 00030000, pos 00030000
1162 mmu set 00040000, pos 00040000
1162 mmu set 00050000, pos 00050000
1162 mmu set 00060000, pos 00060000
1162 mmu set 00070000, pos 00070000
1162 mmu set 00080000, pos 00080000
1162 mmu set 00090000, pos 00090000
1162 mmu set 000a0000, pos 000a0000
1162 mmu set 000b0000, pos 000b0000
1162 mmu set 000c0000, pos 000c0000
1162 mmu set 000d0000, pos 000d0000
1162 mmu set 000e0000, pos 000e0000
1162 mmu set 000f0000, pos 000f0000
1162 mmu set 00100000, pos 00100000
1162 mmu set 00110000, pos 00110000
1162 mmu set 00120000, pos 00120000
ets Jun  8 2016 00:22:57
.
.
.

```

projectgus, 2019-02-13:
Hi @sunbearc22 ,

Glad you sorted out the code.

Flashing the bootloader image with the QIO flash mode set doesn't work, they need to be flashed in DIO mode.

The clue is here:

```
mode:QIO, clock div:2
load:0x3fff0018,len:4
load:0xcdc72100,len:687535457
1162 mmu set 00010000, pos 00010000
```

The second load address and length are garbage, because of QIO mode.

A general guide to the messages printed by the bootloader during a (normal) boot can be found here: https://github.com/espressif/esptool/wiki/ESP32-Boot-Mode-Selection#later-boot-messages

sunbearc22, 2019-02-21:
@projectgus Thank you. Flashing in DIO mode worked.

Questions: 
1. According to wiki on [Serial Protocol](https://github.com/espressif/esptool/wiki/Serial-Protocol#serial-protocol):

   > By default, esptool.py uploads a stub "software loader" to the IRAM of the chip. The software loader then replaces the ROM loader for all future interactions. This standardizes much of the behaviour.
  
   This occurs because `args.no_stub=False` by default.
   When writing to flash, can `args.no_stub=True` be used instead? When should one use `args.no_stub=True`?

2. What does `b'\x07\x07\x12\x20' + 32 * b'\x55'` do in the [.syn() method](https://github.com/espressif/esptool/blob/master/esptool.py#L378)? 

3. In Wiki's explanation on [Initial Synchronization](https://github.com/espressif/esptool/wiki/Serial-Protocol#initial-synchronisation), it writes:
   > The ESP chip is reset into UART bootloader mode. The host starts by sending SYNC commands. These commands have a large data payload which is also used by the ESP chip to detect the configured baud rate. The ESP8266 will initialise at 74800bps with a 26MHz crystal and 115200bps with a 40MHz crystal, and ESP32 always initialises at 115200bps. However the sync packets can be sent at any baud rate, and the UART peripheral will detect this.

    - Am I correct to say that "The ESP chip is reset into UART bootloader mode." is achieved during `._connect_attempt()` method via Line [418](https://github.com/espressif/esptool/blob/master/esptool.py#L418) to [434](https://github.com/espressif/esptool/blob/master/esptool.py#L434)? Is the UART bootloader the same as the ROM bootloader mention in my question 1? If not, then can you explain what is "UART bootloader mode"?
   - Am I correct to say that the rest of the above quote describes what the [.syn() method](https://github.com/espressif/esptool/blob/master/esptool.py#L378) do?

   > esptool.py then (by default) uses the "RAM Download" sequence to upload software stub loader code to IRAM of the chip. The MEM_END command contains the entry-point address to run the software loader. The software loader then sends a custom SLIP packet of the sequence OHAI (0xC0 0x4F 0x48 0x41 0x49 0xC0), indicating that it is now running. This is the only unsolicited packet ever sent by the ESP. If the --no-stub argument is supplied to esptool.py, this entire step is skipped.
   - Am I correct to say that the above description occurs when [.run_stub() method](https://github.com/espressif/esptool/blob/master/esptool.py#L576) is activated and not during the [.connect() method](https://github.com/espressif/esptool/blob/master/esptool.py#L452)?

projectgus, 2019-02-21:
> 1. This occurs because `args.no_stub=False` by default.
>       When writing to flash, can `args.no_stub=True` be used instead? When should one use > `args.no_stub=True`?

There are some operations which are only supported by the stub loader. But you can write flash with or without using the stub loader. The stub loader is faster, though.

>  * What does `b'\x07\x07\x12\x20' + 32 * b'\x55'` do in the [.syn() method](https://github.com/espressif/esptool/blob/master/esptool.py#L378)?

I'm actually not sure what the first 4 bytes signify, I think the chip will echo them back (although I think esptool doesn't actually check what gets echoed back).

The 0x55s are useful for auto-bauding (the sync packets are dual purpose: the ROM in the chip will use the chip UART auto-baud feature to sync with whatever the incoming baud rate is, and then after this happens it will try to receive a sync packet). 0x55 is close to a square wave in terms of alternating bits.

>  * Am I correct to say that "The ESP chip is reset into UART bootloader mode." is achieved during `._connect_attempt()` method via Line [418](https://github.com/espressif/esptool/blob/master/esptool.py#L418) to [434](https://github.com/espressif/esptool/blob/master/esptool.py#L434)?

Yes.

> Is the UART bootloader the same as the ROM bootloader mention in the quote in question 1? If not, then can you explain what is "UART bootloader mode"?

Yes, this is sloppy terminology. The bootloader mode is implemented in mask ROM and communicates using the UART, so it gets called both things.

> * Am I correct to say that the rest of the above quote describes what the [.syn() method](https://github.com/espressif/esptool/blob/master/esptool.py#L378) do?

Yes.

>     * Am I correct to say that the above description occurs when [.run_stub() method](https://github.com/espressif/esptool/blob/master/esptool.py#L576) is activated and not during the [.connect() method](https://github.com/espressif/esptool/blob/master/esptool.py#L452)?

Yes.

> * during the [.connect() method](https://github.com/espressif/esptool/blob/master/esptool.py#L452)?

* I am confuse with which device does the input and output in [lines 438 and 439](https://github.com/espressif/esptool/blob/master/esptool.py#L438#L439) refer to. Can you help clarify? Does input refers to the ESP32 device UART buffer and output refers to the computer(i.e. host) UART buffer? Or is it the other way round?

These are from the point of view of the computer's serial interface.

Input = bytes received on serial interface (sent from the ESP32, we hope).

Output = bytes sent on the serial interface (received by the ESP32, we hope).

Hope this helps clarify some things. :)

closed 2019-02-21 by projectgus

sunbearc22, 2019-02-28:
@projectgus Thank you for your advices. Here is the [link to the GUI application](https://github.com/sunbearc22/ESP32FlashWriter) that I have developed. I have mentioned you in my acknowledgement. Can you give it a try and share with me your comments? Cheers.

juan98mg, 2019-05-22:
@sunbearc22   Hello, hey friend I figured out the solution. http://arduinoamuete.blogspot.com/2015/12/reiniciar-nodemcu-en-windows.html on this link you can see the solve problem it worked for me, just remember while is started flash you need to press down flash button and keep it down until the program finish. and then your board is gonna work

juan98mg, 2019-05-22:
Hello, hey friend I figured out the solution. http://arduinoamuete.blogspot.com/2015/12/reiniciar-nodemcu-en-windows.html on this link you can see the solve problem it worked for me, just remember while is started flash you need to press down flash button and keep it down until the program finish. and then your board is gonna work

juan98mg, 2019-05-22:
@sunbearc22  read my answer above

Shahrizanijan7, 2019-08-14:
D:\Arduino Apps\Arduino\arduino-builder -dump-prefs -logger=machine -hardware D:\Arduino Apps\Arduino\hardware -hardware C:\Users\Asus\AppData\Local\Arduino15\packages -tools D:\Arduino Apps\Arduino\tools-builder -tools D:\Arduino Apps\Arduino\hardware\tools\avr -tools C:\Users\Asus\AppData\Local\Arduino15\packages -built-in-libraries D:\Arduino Apps\Arduino\libraries -libraries C:\Users\Asus\Documents\Arduino\libraries -fqbn=esp8266:esp8266:espresso_lite_v2:xtal=80,vt=flash,exception=disabled,ssl=all,eesz=4M,ResetMethod=ck,ip=lm2f,dbg=Disabled,lvl=None____,wipe=none,baud=115200 -ide-version=10805 -build-path C:\Users\Asus\AppData\Local\Temp\arduino_build_624871 -warnings=none -build-cache C:\Users\Asus\AppData\Local\Temp\arduino_cache_332387 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.xtensa-lx106-elf-gcc.path=C:\Users\Asus\AppData\Local\Arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\2.5.0-3-20ed2b9 -prefs=runtime.tools.python.path=C:\Users\Asus\AppData\Local\Arduino15\packages\esp8266\tools\python\3.7.2-post1 -prefs=runtime.tools.mkspiffs.path=C:\Users\Asus\AppData\Local\Arduino15\packages\esp8266\tools\mkspiffs\2.5.0-3-20ed2b9 -verbose D:\Iot_Workshop\Iot_Workshop.ino
D:\Arduino Apps\Arduino\arduino-builder -compile -logger=machine -hardware D:\Arduino Apps\Arduino\hardware -hardware C:\Users\Asus\AppData\Local\Arduino15\packages -tools D:\Arduino Apps\Arduino\tools-builder -tools D:\Arduino Apps\Arduino\hardware\tools\avr -tools C:\Users\Asus\AppData\Local\Arduino15\packages -built-in-libraries D:\Arduino Apps\Arduino\libraries -libraries C:\Users\Asus\Documents\Arduino\libraries -fqbn=esp8266:esp8266:espresso_lite_v2:xtal=80,vt=flash,exception=disabled,ssl=all,eesz=4M,ResetMethod=ck,ip=lm2f,dbg=Disabled,lvl=None____,wipe=none,baud=115200 -ide-version=10805 -build-path C:\Users\Asus\AppData\Local\Temp\arduino_build_624871 -warnings=none -build-cache C:\Users\Asus\AppData\Local\Temp\arduino_cache_332387 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.xtensa-lx106-elf-gcc.path=C:\Users\Asus\AppData\Local\Arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\2.5.0-3-20ed2b9 -prefs=runtime.tools.python.path=C:\Users\Asus\AppData\Local\Arduino15\packages\esp8266\tools\python\3.7.2-post1 -prefs=runtime.tools.mkspiffs.path=C:\Users\Asus\AppData\Local\Arduino15\packages\esp8266\tools\mkspiffs\2.5.0-3-20ed2b9 -verbose D:\Iot_Workshop\Iot_Workshop.ino
Using board 'espresso_lite_v2' from platform in folder: C:\Users\Asus\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2
Using core 'esp8266' from platform in folder: C:\Users\Asus\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2
Detecting libraries used...
"C:\Users\Asus\AppData\Local\Arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\2.5.0-3-20ed2b9/bin/xtensa-lx106-elf-g++" -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ "-IC:\Users\Asus\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2/tools/sdk/include" "-IC:\Users\Asus\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2/tools/sdk/lwip2/include" "-IC:\Users\Asus\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2/tools/sdk/libc/xtensa-lx106-elf/include" "-IC:\Users\Asus\AppData\Local\Temp\arduino_build_624871/core" -c -w -Os -g -mlongcalls -mtext-section-literals -fno-rtti -falign-functions=4 -std=c++11  -ffunction-sections -fdata-sections -fno-exceptions  -w -x c++ -E -CC -DNONOSDK221=1 -DF_CPU=80000000L -DLWIP_OPEN_SRC -DTCP_MSS=536 -DLWIP_FEATURES=1 -DLWIP_IPV6=0   -DARDUINO=10805 -DARDUINO_ESP8266_ESPRESSO_LITE_V2 -DARDUINO_ARCH_ESP8266 -DARDUINO_BOARD="ESP8266_ESPRESSO_LITE_V2"  -DFLASHMODE_DIO  -DESP8266 "-IC:\Users\Asus\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2\cores\esp8266" "-IC:\Users\Asus\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2\variants\espresso_lite_v2" "C:\Users\Asus\AppData\Local\Temp\arduino_build_624871\sketch\Iot_Workshop.ino.cpp" -o "nul"
"C:\Users\Asus\AppData\Local\Arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\2.5.0-3-20ed2b9/bin/xtensa-lx106-elf-g++" -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ "-IC:\Users\Asus\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2/tools/sdk/include" "-IC:\Users\Asus\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2/tools/sdk/lwip2/include" "-IC:\Users\Asus\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2/tools/sdk/libc/xtensa-lx106-elf/include" "-IC:\Users\Asus\AppData\Local\Temp\arduino_build_624871/core" -c -w -Os -g -mlongcalls -mtext-section-literals -fno-rtti -falign-functions=4 -std=c++11  -ffunction-sections -fdata-sections -fno-exceptions  -w -x c++ -E -CC -DNONOSDK221=1 -DF_CPU=80000000L -DLWIP_OPEN_SRC -DTCP_MSS=536 -DLWIP_FEATURES=1 -DLWIP_IPV6=0   -DARDUINO=10805 -DARDUINO_ESP8266_ESPRESSO_LITE_V2 -DARDUINO_ARCH_ESP8266 -DARDUINO_BOARD="ESP8266_ESPRESSO_LITE_V2"  -DFLASHMODE_DIO  -DESP8266 "-IC:\Users\Asus\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2\cores\esp8266" "-IC:\Users\Asus\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2\variants\espresso_lite_v2" "-IC:\Users\Asus\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2\libraries\ESP8266WiFi\src" "C:\Users\Asus\AppData\Local\Temp\arduino_build_624871\sketch\Iot_Workshop.ino.cpp" -o "nul"
[... cut ...]
  File "C:/Users/Asus/AppData/Local/Arduino15/packages/esp8266/hardware/esp8266/2.5.2/tools/esptool\esptool.py", line 2653, in main
    esp.connect(args.before)
  File "C:/Users/Asus/AppData/Local/Arduino15/packages/esp8266/hardware/esp8266/2.5.2/tools/esptool\esptool.py", line 468, in connect
    raise FatalError('Failed to connect to %s: %s' % (self.CHIP_NAME, last_error))
esptool.FatalError: Failed to connect to ESP8266: Timed out waiting for packet header
esptool.FatalError: Failed to connect to ESP8266: Timed out waiting for packet header

mirospacek, 2019-10-29:
At the "esptool.FatalError: Timed out waiting for packet header" error.
Using a J-DEAL® NodeMCU LUA WiFi Module Internet Based on ESP8266 Development Board New Version from amazon.
Hold down the FLASH button while uploading. Release it when done.
Then unplug USB cable. Then plug USB cable back in.
The uploaded program should start running.

sudarkce, 2019-11-18:
https://github.com/espressif/esptool/issues/441#issuecomment-554841909. esptool fatal error solved through this method.

Damiortiz, 2020-01-07:
I tried all the things I saw on the Internet. The only solution that worked was this.

    Press the flash and RST button at the same time for a few seconds.
    Release only the RST button
    Without releasing the FLASH button upload the code with NodeMCU version 1.0

I hope someone will use it!

MNRPATIL, 2020-02-25:
esptool.py v2.8
Serial port COM2
Connecting........_____....._____....._____....._____....._____....._____.....____Traceback (most recent call last):
  File "C:\Users\kec\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.6.3/tools/upload.py", line 65, in <module>
    esptool.main(cmdline)
  File "C:/Users/kec/AppData/Local/Arduino15/packages/esp8266/hardware/esp8266/2.6.3/tools/esptool\esptool.py", line 2890, in main
    esp.connect(args.before)
  File "C:/Users/kec/AppData/Local/Arduino15/packages/esp8266/hardware/esp8266/2.6.3/tools/esptool\esptool.py", line 483, in connect
    raise FatalError('Failed to connect to %s: %s' % (self.CHIP_NAME, last_error))
esptool.FatalError: Failed to connect to ESP8266: Timed out waiting for packet header
esptool.FatalError: Failed to connect to ESP8266: Timed out waiting for packet header

mmors01, 2020-02-27:
Hi! how to solve this problem? thank you :)
I'm using Arduino to program ESP8266 - ESP-01

__An error occurred while uploading the sketch
esptool.py v2.8
Serial port COM8
Connecting........_____....._____....._____....._____....._____....._____....._____
Traceback (most recent call last):
  File "C:\Users\acer\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.6.3/tools/upload.py", line 65, in <module>
    esptool.main(cmdline)
  File "C:/Users/acer/AppData/Local/Arduino15/packages/esp8266/hardware/esp8266/2.6.3/tools/esptool\esptool.py", line 2890, in main
    esp.connect(args.before)
  File "C:/Users/acer/AppData/Local/Arduino15/packages/esp8266/hardware/esp8266/2.6.3/tools/esptool\esptool.py", line 483, in connect
    raise FatalError('Failed to connect to %s: %s' % (self.CHIP_NAME, last_error))
esptool.FatalError: Failed to connect to ESP8266: Timed out waiting for packet header
esptool.FatalError: Failed to connect to ESP8266: Timed out waiting for packet header

nazanin97, 2020-04-14:
Hi, I have this error, how can I fix it?
I'm using Arduino Uno with esp8266-01

Arduino: 1.8.12 (Mac OS X), Board: "Generic ESP8266 Module, 80 MHz, Flash, Legacy (new can return nullptr), All SSL ciphers (most compatible), dtr (aka nodemcu), 26 MHz, 40MHz, DOUT (compatible), 1MB (FS:64KB OTA:~470KB), 2, nonos-sdk 2.2.1+100 (190703), v2 Lower Memory, Disabled, None, Only Sketch, 115200"

/Applications/Arduino.app/Contents/Java/arduino-builder -dump-prefs -logger=machine -hardware /Applications/Arduino.app/Contents/Java/hardware -hardware /Users/nazanin/Library/Arduino15/packages -tools /Applications/Arduino.app/Contents/Java/tools-builder -tools /Applications/Arduino.app/Contents/Java/hardware/tools/avr -tools /Users/nazanin/Library/Arduino15/packages -built-in-libraries /Applications/Arduino.app/Contents/Java/libraries -libraries /Users/nazanin/Documents/Arduino/libraries -fqbn=esp8266:esp8266:generic:xtal=80,vt=flash,exception=legacy,ssl=all,ResetMethod=nodemcu,CrystalFreq=26,FlashFreq=40,FlashMode=dout,eesz=1M64,led=2,sdk=nonosdk_190703,ip=lm2f,dbg=Disabled,lvl=None____,wipe=none,baud=115200 -vid-pid=0X2341_0X0043 -ide-version=10812 -build-path /var/folders/11/fjrxwd413yl0rhm04vsfnj7c0000gn/T/arduino_build_295774 -warnings=none -build-cache /var/folders/11/fjrxwd413yl0rhm04vsfnj7c0000gn/T/arduino_cache_186496 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.mkspiffs.path=/Users/nazanin/Library/Arduino15/packages/esp8266/tools/mkspiffs/2.5.0-4-b40a506 -prefs=runtime.tools.mkspiffs-2.5.0-4-b40a506.path=/Users/nazanin/Library/Arduino15/packages/esp8266/tools/mkspiffs/2.5.0-4-b40a506 -prefs=runtime.tools.mklittlefs.path=/Users/nazanin/Library/Arduino15/packages/esp8266/tools/mklittlefs/2.5.0-4-69bd9e6 -prefs=runtime.tools.mklittlefs-2.5.0-4-69bd9e6.path=/Users/nazanin/Library/Arduino15/packages/esp8266/tools/mklittlefs/2.5.0-4-69bd9e6 -prefs=runtime.tools.xtensa-lx106-elf-gcc.path=/Users/nazanin/Library/Arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/2.5.0-4-b40a506 -prefs=runtime.tools.xtensa-lx106-elf-gcc-2.5.0-4-b40a506.path=/Users/nazanin/Library/Arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/2.5.0-4-b40a506 -prefs=runtime.tools.python3.path=/Users/nazanin/Library/Arduino15/packages/esp8266/tools/python3/3.7.2-post1 -prefs=runtime.tools.python3-3.7.2-post1.path=/Users/nazanin/Library/Arduino15/packages/esp8266/tools/python3/3.7.2-post1 -verbose /Users/nazanin/Library/Arduino15/packages/esp8266/hardware/esp8266/2.6.3/libraries/ESP8266WiFi/examples/WiFiScan/WiFiScan.ino
/Applications/Arduino.app/Contents/Java/arduino-builder -compile -logger=machine -hardware /Applications/Arduino.app/Contents/Java/hardware -hardware /Users/nazanin/Library/Arduino15/packages -tools /Applications/Arduino.app/Contents/Java/tools-builder -tools /Applications/Arduino.app/Contents/Java/hardware/tools/avr -tools /Users/nazanin/Library/Arduino15/packages -built-in-libraries /Applications/Arduino.app/Contents/Java/libraries -libraries /Users/nazanin/Documents/Arduino/libraries -fqbn=esp8266:esp8266:generic:xtal=80,vt=flash,exception=legacy,ssl=all,ResetMethod=nodemcu,CrystalFreq=26,FlashFreq=40,FlashMode=dout,eesz=1M64,led=2,sdk=nonosdk_190703,ip=lm2f,dbg=Disabled,lvl=None____,wipe=none,baud=115200 -vid-pid=0X2341_0X0043 -ide-version=10812 -build-path /var/folders/11/fjrxwd413yl0rhm04vsfnj7c0000gn/T/arduino_build_295774 -warnings=none -build-cache /var/folders/11/fjrxwd413yl0rhm04vsfnj7c0000gn/T/arduino_cache_186496 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.mkspiffs.path=/Users/nazanin/Library/Arduino15/packages/esp8266/tools/mkspiffs/2.5.0-4-b40a506 -prefs=runtime.tools.mkspiffs-2.5.0-4-b40a506.path=/Users/nazanin/Library/Arduino15/packages/esp8266/tools/mkspiffs/2.5.0-4-b40a506 -prefs=runtime.tools.mklittlefs.path=/Users/nazanin/Library/Arduino15/packages/esp8266/tools/mklittlefs/2.5.0-4-69bd9e6 -prefs=runtime.tools.mklittlefs-2.5.0-4-69bd9e6.path=/Users/nazanin/Library/Arduino15/packages/esp8266/tools/mklittlefs/2.5.0-4-69bd9e6 -prefs=runtime.tools.xtensa-lx106-elf-gcc.path=/Users/nazanin/Library/Arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/2.5.0-4-b40a506 -prefs=runtime.tools.xtensa-lx106-elf-gcc-2.5.0-4-b40a506.path=/Users/nazanin/Library/Arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/2.5.0-4-b40a506 -prefs=runtime.tools.python3.path=/Users/nazanin/Library/Arduino15/packages/esp8266/tools/python3/3.7.2-post1 -prefs=runtime.tools.python3-3.7.2-post1.path=/Users/nazanin/Library/Arduino15/packages/esp8266/tools/python3/3.7.2-post1 -verbose /Users/nazanin/Library/Arduino15/packages/esp8266/hardware/esp8266/2.6.3/libraries/ESP8266WiFi/examples/WiFiScan/WiFiScan.ino
Using board 'generic' from platform in folder: /Users/nazanin/Library/Arduino15/packages/esp8266/hardware/esp8266/2.6.3
Using core 'esp8266' from platform in folder: /Users/nazanin/Library/Arduino15/packages/esp8266/hardware/esp8266/2.6.3
Detecting libraries used...
/Users/nazanin/Library/Arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/2.5.0-4-b40a506/bin/xtensa-lx106-elf-g++ -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ -I/Users/nazanin/Library/Arduino15/packages/esp8266/hardware/esp8266/2.6.3/tools/sdk/include -I/Users/nazanin/Library/Arduino15/packages/esp8266/hardware/esp8266/2.6.3/tools/sdk/lwip2/include -I/Users/nazanin/Library/Arduino15/packages/esp8266/hardware/esp8266/2.6.3/tools/sdk/libc/xtensa-lx106-elf/include -I/var/folders/11/fjrxwd413yl0rhm04vsfnj7c0000gn/T/arduino_build_295774/core -c -w - [... cut ...]
  File "/Users/nazanin/Library/Arduino15/packages/esp8266/hardware/esp8266/2.6.3/tools/esptool/esptool.py", line 2890, in main
    esp.connect(args.before)
  File "/Users/nazanin/Library/Arduino15/packages/esp8266/hardware/esp8266/2.6.3/tools/esptool/esptool.py", line 483, in connect
    raise FatalError('Failed to connect to %s: %s' % (self.CHIP_NAME, last_error))
esptool.FatalError: Failed to connect to ESP8266: Timed out waiting for packet header
esptool.FatalError: Failed to connect to ESP8266: Timed out waiting for packet header

_

Thank you.

priyarithadia, 2020-12-01:
hi all,
Executable segment sizes:
IROM   : 236328          - code in flash         (default or ICACHE_FLASH_ATTR) 
IRAM   : 28596   / 32768 - code in IRAM          (ICACHE_RAM_ATTR, ISRs...) 
DATA   : 1260  )         - initialized variables (global, static) in RAM/HEAP 
RODATA : 780   ) / 81920 - constants             (global, static) in RAM/HEAP 
BSS    : 25712 )         - zeroed variables      (global, static) in RAM/HEAP 
Sketch uses 266964 bytes (27%) of program storage space. Maximum is 958448 bytes.
Global variables use 27752 bytes (33%) of dynamic memory, leaving 54168 bytes for local variables. Maximum is 81920 bytes.
esptool.py v2.8
Serial port COM3
Connecting........_____....._____....._____....._____....._____....._____.....____Traceback (most recent call last):
  File "C:\Users\PRIYA\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.4/tools/upload.py", line 65, in <module>
    esptool.main(cmdline)
  File "C:/Users/PRIYA/AppData/Local/Arduino15/packages/esp8266/hardware/esp8266/2.7.4/tools/esptool\esptool.py", line 2890, in main
    esp.connect(args.before)
  File "C:/Users/PRIYA/AppData/Local/Arduino15/packages/esp8266/hardware/esp8266/2.7.4/tools/esptool\esptool.py", line 483, in connect
    raise FatalError('Failed to connect to %s: %s' % (self.CHIP_NAME, last_error))
esptool.FatalError: Failed to connect to ESP8266: Timed out waiting for packet header
esptool.FatalError: Failed to connect to ESP8266: Timed out waiting for packet header

_
problem please help me