espressif/esptool #409: ModuleNotFoundError No module named 'esptool'

mracst, 2019-02-13:
Hi,
I am new to python. I followed the installation instructions but I was not able to fix the problem.
I install python checking the Windows path environment variable modification .

* Operating system: Windows 7
* Windows path include python root and python scripts directories
* Python version: 3.7.2
* Hardware: ESP02+CPI2102

I specified the following command  inside the CMD propt
esptool.py -b 115200 -port COM15 read_flash 0x000000 0x100000 flash_1M.bin

The output was:
Traceback (most recent call last):
  File "E:\Programs_7\Python\Scripts\esptool.py-script.py", line 11, in <module>
    load_entry_point('esptool==2.7.dev0', 'console_scripts', 'esptool.py')()
  File "E:\Programs_7\Python\lib\site-packages\pkg_resources\__init__.py", line 487, in load_entry_point
    return get_distribution(dist).load_entry_point(group, name)
  File "E:\Programs_7\Python\lib\site-packages\pkg_resources\__init__.py", line 2728, in load_entry_point
    return ep.load()
  File "E:\Programs_7\Python\lib\site-packages\pkg_resources\__init__.py", line 2346, in load
    return self.resolve()
  File "E:\Programs_7\Python\lib\site-packages\pkg_resources\__init__.py", line 2352, in resolve
    module = __import__(self.module_name, fromlist=['__name__'], level=0)
ModuleNotFoundError: No module named 'esptool'

Thank's for the help.

projectgus, 2019-02-14:
It does look like some kind of Python path issue.

I take it you installed esptool using `pip install esptool`, is that right?

Is the Python installed into `E:\Programs_7\Python\` the only Python on your system?

If you run `python esptool.py -b 115200 -port COM15 read_flash 0x000000 0x100000 flash_1M.bin`, does it work then?

mracst, 2019-02-14:
I have installed the script using pip install esptool command and the Python is the only one installed in my PC.
It works using the command python esptool.py -b 115200 --port COM15 read_flash 0x000000 0x100000 flash_1M.bin.
Have you any suggestion to solve the problem with esptool.py.exe?

projectgus, 2019-02-18:
Good to hear that you've found a solution that does work.

When you run a `.py` file directly in Windows, it looks up some associations about how to run that file. Essentially, Windows does some lookups and then decides "I should run this with the `python` interpreter" and then it internally runs `python esptool.py ...`.

There's a description of this process here: https://stackoverflow.com/a/1936078 - suggest checking that all these associations match the same command which is run when you do `python esptool.py ...` on the command line.

projectgus, 2019-05-28:
Closing as the original issue is fixed, and no update on the follow-up question.

closed 2019-05-28 by projectgus