espressif/esptool #411: termios.error: (5, 'Input/output error')

sunbearc22, 2019-02-21:
I am encountering this error when the ESP32 unplug at the usb connector. How do I handle this? I would like my application to continue to run despite of the disconnection. 
 
```
Traceback (most recent call last):
  File "~/Esp32/test.py", line 291, in _check_connection
    self.esp.flush_input()
  File "~/.local/share/virtualenvs/Esp32-zoIgntnu/bin/esptool.py", line 392, in flush_input
    self._port.flushInput()
  File "~/.local/share/virtualenvs/Esp32-zoIgntnu/lib/python3.6/site-packages/serial/serialutil.py", line 584, in flushInput
    self.reset_input_buffer()
  File "~/.local/share/virtualenvs/Esp32-zoIgntnu/lib/python3.6/site-packages/serial/serialposix.py", line 595, in reset_input_buffer
    termios.tcflush(self.fd, termios.TCIFLUSH)
termios.error: (5, 'Input/output error')
```

projectgus, 2019-02-21:
I agree that pyserial should probably throw a SerialException here, but this code is part of [pyserial](https://github.com/pyserial/pyserial/) not part of esptool. esptool doesn't currently do any filtering of pyserial errors. Suggest sending the pyserial maintainers a PR.

In the meantime, you `import termios` and `except termios.error:` in the calling code as a workaround.

closed 2019-03-15 by sunbearc22

gojimmypi, 2023-12-19:
I bumped into this error today, both with the ESP-IDF and with my own, custom pything script, failing at this line:

```
ser = serial.Serial(args.port, args.baudrate)
```

I'm doing this on an ESP32-C2 which for some odd reason seems to be stuck on a monitor baud rate of 78800. Something like `putty` works just fine, but apparently 78800 is not one of the baud rates supported by [pySerial](https://pyserial.readthedocs.io/en/latest/pyserial_api.html).

Using my script:
```
gojimmypi:/mnt/c/workspace/wolfssl-master/IDE/Espressif/ESP-IDF/examples/wolfssl_test
$ ./wolfssl_monitor.py  -p /dev/ttyS49 -b 78800
Saving log to ESP32.log -p /dev/ttyS49 -b 78800
Connecting to /dev/ttyS49p /dev/ttyS49 -b 78800
Traceback (most recent call last):
  File "/mnt/c/workspace/wolfssl-master/IDE/Espressif/ESP-IDF/examples/wolfssl_test/./wolfssl_monitor.py", line 34, in <module>
    ser = serial.Serial(args.port, args.baudrate)
  File "/home/gojimmypi/.espressif/python_env/idf5.1_py3.10_env/lib/python3.10/site-packages/serial/serialutil.py", line 244, in __init__
    self.open()
  File "/home/gojimmypi/.espressif/python_env/idf5.1_py3.10_env/lib/python3.10/site-packages/serial/serialposix.py", line 332, in open
    self._reconfigure_port(force_update=True)
  File "/home/gojimmypi/.espressif/python_env/idf5.1_py3.10_env/lib/python3.10/site-packages/serial/serialposix.py", line 517, in _reconfigure_port
    termios.tcsetattr(
termios.error: (5, 'Input/output error')
```

Using `idf.py`:
```
gojimmypi:/mnt/c/workspace/wolfssl-master/IDE/Espressif/ESP-IDF/examples/wolfssl_test
$ idf.py monitor --monitor-baud 78800 --port /dev/ttyS49
/mnt/c/SysGCC/esp32/esp-idf/v5.1/tools/check_python_dependencies.py:12: DeprecationWarning: pkg_resources is deprecated as an API. See https://setuptools.pypa.io/en/latest/pkg_resources.html
  import pkg_resources
Executing action: monitor
Running idf_monitor in directory /mnt/c/workspace/wolfssl-master/IDE/Espressif/ESP-IDF/examples/wolfssl_test
Executing "/home/gojimmypi/.espressif/python_env/idf5.1_py3.10_env/bin/python /mnt/c/SysGCC/esp32/esp-idf/v5.1/tools/idf_monitor.py -p /dev/ttyS49 -b 78800 --toolchain-prefix riscv32-esp-elf- --target esp32c2 --decode-panic backtrace /mnt/c/workspace/wolfssl-master/IDE/Espressif/ESP-IDF/examples/wolfssl_test/build/wolfssl_test.elf -m '/home/gojimmypi/.espressif/python_env/idf5.1_py3.10_env/bin/python' '/mnt/c/SysGCC/esp32/esp-idf/v5.1/tools/idf.py'"...
--- idf_monitor on /dev/ttyS49 78800 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
Exception in thread Thread-2 (_run_outer):
Traceback (most recent call last):
  File "/usr/lib/python3.10/threading.py", line 1016, in _bootstrap_inner
    self.run()
  File "/usr/lib/python3.10/threading.py", line 953, in run
    self._target(*self._args, **self._kwargs)
  File "/home/gojimmypi/.espressif/python_env/idf5.1_py3.10_env/lib/python3.10/site-packages/esp_idf_monitor/base/stoppable_thread.py", line 47, in _run_outer
    self.run()
  File "/home/gojimmypi/.espressif/python_env/idf5.1_py3.10_env/lib/python3.10/site-packages/esp_idf_monitor/base/serial_reader.py", line 53, in run    self.serial.open()
  File "/home/gojimmypi/.espressif/python_env/idf5.1_py3.10_env/lib/python3.10/site-packages/serial/serialposix.py", line 332, in open
    self._reconfigure_port(force_update=True)
  File "/home/gojimmypi/.espressif/python_env/idf5.1_py3.10_env/lib/python3.10/site-packages/serial/serialposix.py", line 517, in _reconfigure_port
    termios.tcsetattr(
termios.error: (5, 'Input/output error')

gojimmypi:/mnt/c/workspace/wolfssl-master/IDE/Espressif/ESP-IDF/examples/wolfssl_test
```