espressif/esptool #412: Datasheet vs github strapping pins discrepancy

scottpage, 2019-02-24:
* ESP hardware in use: ESP32-WROOM-32

Details:
In this repository it states that GPIO4 is a strapping pin, and that information about strapping pins can be found in section 2.4.

Issues:
1) The datasheet does not mention GPIO4 as a strapping pin.
2) The datasheet does not contain a section 2.4.  Instead, strapping pins are in 2.3.

projectgus, 2019-02-25:
Hi @scottpage ,

Just to make I understand, you mean the [list of pins in the GPIO_STRAP register given on the wiki here](https://github.com/espressif/esptool/wiki/ESP32-Boot-Mode-Selection#boot-mode-message), yes? 

>     1. The datasheet does not mention GPIO4 as a strapping pin.

You're right, GPIO4 is part of the `GPIO_STRAP` register but it's not used to select any of the documented boot modes. I've added a note on the wiki clarifying that this pin can be either high or low for all documented boot modes.

>     2. The datasheet does not contain a section 2.4.  Instead, strapping pins are in 2.3.

What version of the datasheet do you have? [Current version is 2.8](https://www.espressif.com/en/support/download/documents) and it has Section 2.4 "Strapping Pins". This section was also numbered 2.4 in datasheet 1.0 (October 2016. Perhaps there is some interim version where it was changed?

projectgus, 2019-05-17:
Closing as I believe the wiki update took care of this issue.

closed 2019-05-17 by projectgus