espressif/esptool #413: A fatal error occurred: Timed out waiting for packet header

Fivefold, 2019-02-24:
* Operating system: Windows 10 Pro
* Python version: 3.7.2
* ESP hardware in use: ESP-12F / ESP8266, specifically this one: [frontside,](https://ae01.alicdn.com/kf/HTB1v0c7u8mWBuNkSndVq6AsApXaY/1PCS-ESP-12F-ESP-12E-upgrade-ESP8266-Remote-Serial-Port-WIFI-Wireless-Module-ESP8266-4M-Flash.jpg) [backside](https://ae01.alicdn.com/kf/HTB1lm3iQVXXXXcRXVXXq6xXFXXXr/1PCS-ESP-12F-ESP-12E-upgrade-ESP8266-Remote-Serial-Port-WIFI-Wireless-Module-ESP8266-4M-Flash.jpg)

# Full esptool.py command line as run:

`esptool.py --port COM3 --baud 74880 write_flash --verify -fm dio 0x00000 firmware.bin`

# Full output from esptool.py (please copy and paste all lines of output)

```
Serial port COM3
Connecting........_
Detecting chip type... ESP8266
Chip is ESP8266EX
Features: WiFi
MAC: 84:f3:eb:b4:5c:d7
Uploading stub...
Running stub...
Stub running...
Configuring flash size...

A fatal error occurred: Timed out waiting for packet header
```

# What is the expected behaviour?

Successfully flashing the image.

# Do you have any other information from investigating this?

1. Manually setting the flash size via `-fs 4MB` makes no difference.
1. Setting the baud rate to 9600 or omitting the `--baud` argument makes no difference in the output.
1. Setting the baud rate to 512000 changes the `Configuring flash size...` to `Changing baud rate to 512000` and afterwards getting the same time out error.
1. I would rule out power as an issue (like with most others that are getting this error). I use a Manson NRP-2050 lab supply to drive the ESP module that can deliver up to 5 A
1. When just connecting to the module via a serial monitor and resetting the module, I get the message: ` ets Jan  8 2013,rst cause:2, boot mode:(1,6)`, so it should be in the correct boot mode for flashing. Any other baud rates I've tried produce garbage characters.

# Is there any other information you can think of which will help us reproduce this problem?

The .bin image used is a NodeMCU image from [here](https://nodemcu-build.com/index.php) and it's size is about 632 KB.

If needed, I can give a sketch or description of my wiring.

projectgus, 2019-02-25:
Hi @Fivefold ,

Thanks for the comprehensive set of details. A couple of additional questions:

* What version of esptool.py do you have? It seemed to get trimmed from the debug log.
* If you flash with `esptool.py --no-stub ...`, does this work?
* If you run `esptool.py -p PORT flash_id`, what does it say?

Thanks,

Angus

Fivefold, 2019-02-25:
Hello @projectgus 

Thank you for the fast response.

> What version of esptool.py do you have? It seemed to get trimmed from the debug log.

My mistake, forgot to copy that line.
v2.6

>  If you flash with `esptool.py --no-stub ...`, does this work?

Unfortunately, no.

```
> esptool.py --no-stub --port COM3 --baud 74880 write_flash --verify -fm dio 0x00000 firmware.bin
esptool.py v2.6
Serial port COM3
Connecting....
Detecting chip type... ESP8266
Chip is ESP8266EX
Features: WiFi
MAC: 84:f3:eb:b4:5c:d7
Enabling default SPI flash mode...

A fatal error occurred: Timed out waiting for packet header
```

I just realized that `dio` is an SPI flash mode. I am trying to flash via UART (i.e. the Rx and Tx pins instead of MISO, MOSI and so on). Is this the issue?

> If you run `esptool.py -p PORT flash_id`, what does it say?

```
>esptool.py -p COM3 flash_id
esptool.py v2.6
Serial port COM3
Connecting...
Detecting chip type... ESP8266
Chip is ESP8266EX
Features: WiFi
MAC: 84:f3:eb:b4:5c:d7
Uploading stub...
Running stub...
Stub running...

A fatal error occurred: Timed out waiting for packet header
```

One thing I forgot to mention ealier: When esptool says `Connecting...` I have to manually reset the module by pulling down the RST pin shortly. I suppose this is correct procedure?

projectgus, 2019-02-25:
Hi,

Thanks for the extra details. The behaviour you have is consistent with bad power, a bad serial link, or a bad flash chip (or flash chip connection) on the module.

* What's connected to the module? Is it possible something is connected to one of the module pins which is shared with the flash chip?

>     4\. I would rule out power as an issue (like with most others that are getting this error). I use a Manson NRP-2050 lab supply to drive the ESP module that can deliver up to 5 A

This is a partial factor for ensuring stable power. Is the module installed on a PCB or a development board of some kind? Is there a low ESR capacitor placed close to the module on the PCB? (ie a 10uF ceramic cap or something of this kind).

If you have a photo of your setup, that would be ideal.

> One thing I forgot to mention ealier: When esptool says `Connecting...` I have to manually reset the module by pulling down the RST pin shortly. I suppose this is correct procedure?

This is the case if you're not using a dev board with an automatic reset circuit. More details here: https://github.com/espressif/esptool/wiki/ESP8266-Boot-Mode-Selection

Fivefold, 2019-03-03:
Sorry for the late response. I was quite busy this week.

> What's connected to the module? Is it possible something is connected to one of the module pins which is shared with the flash chip?

Nothing besides the FTDI USB to Serial Module: [image](https://ae01.alicdn.com/kf/HTB1QCD3SFXXXXayXpXXq6xXFXXXW/High-Quality-FT232RL-FT232-FTDI-USB-3-3V-5-5V-to-TTL-Serial-Adapter-Module-Mini.jpg)

> This is one of the factors for ensuring the module gets stable power. Is the module installed on a PCB or a development board of some kind? Is there a low ESR capacitor placed close to the module on the PCB? (ie a 10uF ceramic cap or something of this kind).
>
> If you have a photo of your setup, that would be ideal.

I soldered the module to a perfboard using small wires (maybe too small for the supply). This is connected to a breadboard via pin headers. [Here are two images of the setup.](https://imgur.com/a/0Ojp173) I included a short description to help out.

For this image I completely redid the breadboard wiring to get a better image and possibly rule out faulty breadboard contacts in the previous configuration. The error messages stayed the same. When connecting to the module via a serial monitor (in flashing mode) and resetting I now get `ets Jan 8 2013,rst cause:2, boot mode:(1,7)` (before it was (1,6)). I am not sure if this is relevant.

The ceramic cap used is only 100 nF; I don't have any larger ceramics at hand currently. I tried an electrolytic to no avail. Also the comparatively long wires might make the capacitor useless anyway. So if there are no other suggestions the next thing I would try (in this order) is: 
1. Get my hands on a 10 uF cap and solder it directly to the terminals of the module 
1. Redo the soldering and exchange the supply wires for larger ones

projectgus, 2019-03-04:
Yes, I agree that the root cause is probably lack of capacitance (or possibly bad breadboard connection or solder connection somewhere - what's under the hot glue?) If you can get a larger value cap (maybe even in parallel with the 100nF cap) and solder it directly to the module, you should find things come good.

If you're just starting out, I really recommend getting one of the little development boards with integrated USB/serial as a first step, and migrating to custom wiring once you're comfortable with that.

> resetting I now get ets Jan 8 2013,rst cause:2, boot mode:(1,7) (before it was (1,6)). I am not sure if this is relevant.

The second number is a bitmask indicating which strapping GPIO pins were high/low on reset, probably one of them is floating. Provided the first number after "boot mode:" is a 1, serial flashing should work. [Some more details here](https://github.com/espressif/esptool/wiki/ESP8266-Boot-ROM-Log).

I'm going to close this issue as it seems like this is a hardware issue, not a bug in esptool. For more pointers on wiring up an ESP8266 module on a breadboard, suggest posting on the http://bbs.espressif.com or http://esp8266.com forums.

closed 2019-03-04 by projectgus

Fivefold, 2019-07-22:
I just want to give a small update for anyone stumbling upon this issue.

Finally got around to making a proper PCB including bypass caps instead of the dead bug wiring I had before. It now works as it should, so it really was the hardware, more specifically the wiring.

DEADSEC-SECURITY, 2020-08-29:
> I just want to give a small update for anyone stumbling upon this issue.
> 
> Finally got around to making a proper PCB including bypass caps instead of the dead bug wiring I had before. It now works as it should, so it really was the hardware, more specifically the wiring.

FiveFold can you send me a copy of the ESP wiring cuz I think I got something wrong too and I just would want to check if my wiring is ok!

Fivefold, 2020-09-02:
Sure! [Here's a few images.](https://imgur.com/a/QWfxeFM) Note that this was a very quick & dirty prototype to test the functionality, so don't expect good PCB design.