espressif/esptool #414: espsecure.py: Move to python cryptography

projectgus, 2019-02-27:
Right now espsecure.py uses two pure Python libraries - pyaes and ecdsa -  for cryptography algorithms.

This is quite slow (espsecure.py needs 7 seconds to encrypt a 1MB binary on my laptop, whereas openssl can encrypt 1MB of data in 22ms). It is also difficult to add new encryption algorithms.

Goal is to move to [python cryptography](https://github.com/pyca/cryptography) for the existing cryptography functions, and to use their APIs for any future functions.

Adding this new dependency will remove the vendored libraries and require espsecure users to use "pip" (however IDF users will already have python-cryptography installed as part of IDF requirements).

darkfader, 2019-03-06:
It's funny how docs tell about "land mines, dragons, and dinosaurs with laser guns".
So, pycrypto is too old? Look like pycryptodome aims compatibility a bit and adds new stuff but still no optimized signing (isn't the most time consuming though). And cryptography has OpenSSL backend; It feels a bit heavy to me though but it can probably use AES-NI.
An alternative would be pyOpenSSL I guess.

projectgus, 2019-05-17:
@darkfader Sorry I didn't see this comment when you posted it.

Thanks for the pointers, I don't think I'd seen pycryptodome and pyOpenSSL before.

The appeal for "cryptography" module comes down to four things: supported algorithms, maturity of project, performance, and ease of install.

Pure Python is better than binary for ease of install obviously, but it misses out on performance. "cryptography" seems to do a good job of shipping a pre-compiled openssl library for Windows & MacOS, which makes it an easy pip install on most platforms (Linux users get to choose between their distros version of the package or letting pip build it locally.) pyOpenSSL might do this as well, I haven't checked.

Also I find their API quite clean, and philosophically I like their approach of putting "hazmat" around most crypto APIs to remind you that they're easy to misuse.

hroncok, 2020-04-08:
PyCA cryptography depends on the much more common and vetted openssl library for cryptography functions. This means algorithms are properly hardened against side channel attacks unlike other implementations.

As a Fedora packager of esptool, I'd very much prefer PyCA cryptography over pyaes and ecdsa.

projectgus, 2020-12-07:
Small update: in esptool 3.0 we've dropped both vendored libraries, and replaced pyaes with pyca-cryptography (see commit above).

The ecdsa module is still a dependency of esptool, for deterministic ECDSA. pyca-cryptography doesn't support this yet, and it seems openssl doesn't either so it may be some time before it's added.

projectgus, 2021-06-23:
Closing this out as it seems like `ecdsa` support will be kept for the forseeable future. The main issues were due to AES performance, and this is solved. Future cryptography features will use the cryptography package whenever possible.

closed 2021-06-23 by projectgus