espressif/esptool #415: No AP configured after uploading firmware

alessss, 2019-03-01:
(Please delete any lines which don't apply.)

* Operating system: MacOS
* Python version: (`python -V` to check this) 2.7.10
* ESP hardware in use: ESP8266EX

# Full esptool.py command line as run:
python esptool.py --port /dev/tty.wchusbserial1430 --baud 57600 write_flash --flash_size=4MB 0 ../esp8266-20180511-v1.9.4.bin --verify

# Full output from esptool.py (please copy and paste all lines of output)
esptool.py v2.7-dev
Serial port /dev/tty.wchusbserial1430
Connecting....
Detecting chip type... ESP8266
Chip is ESP8266EX
Features: WiFi
MAC: 60:01:94:7b:75:22
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Flash params set to 0x0040
Compressed 604872 bytes to 394893...
Wrote 604872 bytes (394893 compressed) at 0x00000000 in 70.3 seconds (effective 68.8 kbit/s)...
Hash of data verified.

Leaving...
Verifying just-written flash...
(This option is deprecated, flash contents are now always read back after flashing.)
Flash params set to 0x0040
Verifying 0x93ac8 (604872) bytes @ 0x00000000 in flash against ../esp8266-20180511-v1.9.4.bin...
-- verify OK (digest matched)
Hard resetting via RTS pin...
Leaving...
Hard resetting via RTS pin...

# What is the expected behaviour?
ESP AP is configured and is seen on list or networks available on Mac.

# Do you have any other information from investigating this?

# Is there any other information you can think of which will help us reproduce this problem?
Tried to change baudrate and all other solutions from point 1.7 (http://docs.micropython.org/en/latest/esp8266/tutorial/intro.html). Nothing helped.

projectgus, 2019-03-05:
Hi @alessss ,

This looks like a MicroPython problem not an esptool problem, especially if the ESP8266 was working properly with MicroPython 1.7. All esptool does is writes the .bin file to the ESP8266 flash, everything else happens once MicroPython boots. And the esptool output suggests that the .bin file was written successfully to the flash.

Only one additional suggestion. Try removing this command line argument: `--flash_size=4MB` entirely and run esptool.py without it. If your flash is actually a different size to 4MB, esptool will detect this and it may help the firmware boot. If the flash size is actually 4MB, removing this argument will make no difference.

The other thing you can try is to run a serial monitor at 74880bps to watch the early ESP8266 boot messages. This may give you some clue.

However, probably the solution to this problem will be found somewhere on the MicroPython side, not the esptool side.

alessss, 2019-03-05:
Fixed using --flash-mode dio. Thanks.

projectgus, 2019-03-06:
Good to hear!

closed 2019-03-06 by projectgus