espressif/esptool #417: esptool.FatalError: Failed to connect to ESP32 : Invalid head on packet 0x08.

sunbearc22, 2019-03-05:
I had recently wrote a python script that provides a [Linux OS GUI](https://github.com/sunbearc22/ESP32FlashWriter) to detect and connect to ESP32 and write firmware to the device flash. It uses esptool.py. I am trying to adapt it to work on Windows 10.
 
I have a connection issue. The same commands that I had used to connect a ESP32 devkitv1 with Linux don't seem to work with Windows 10. I noticed the `esptool.ESPLoader.connect()` method keeps returning 
`esptool.FatalError: Failed to connect to ESP32 : Invalid head on packet 0x08.` This message was caused [line 468](https://github.com/espressif/esptool/blob/master/esptool.py#L468). How do I resolve this issue?

sunbearc22, 2019-03-05:
Here is the link to my [work-in-progress script for Win10](https://github.com/sunbearc22/ESP32FlashWriter/blob/master/esp32flashwriter_w4_win_d1.py). esptool.FatalError arising from [line L354](https://github.com/sunbearc22/ESP32FlashWriter/blob/master/esp32flashwriter_w4_win_d1.py#L353)

sunbearc22, 2019-03-06:
I have used the solution by @MartyMacGyver mentioned in Issue #136 to resolve my connection issue. After replacing esptool.py [line 425](https://github.com/espressif/esptool/blob/master/esptool.py#L425) with `time.sleep(2.0)  # Sleep longer` and [line 432](https://github.com/espressif/esptool/blob/master/esptool.py#L432) with `time.sleep(0.5)  # Sleep longer`, the above mentioned connection issue was resolved. That is,
```
...
self._port.setDTR(False)  # IO0=HIGH
self._port.setRTS(True)   # EN=LOW, chip in reset
time.sleep(2.0)  # Sleep longer
self._port.setDTR(True)   # IO0=LOW
self._port.setRTS(False)  # EN=HIGH, chip out of reset
time.sleep(0.5)  # Sleep longer
...
```
May I know why esptool.py default `sleep` settings for DTR and RTS aren't using the values provided by @MartyMacGyver? I found that these longer sleep settings also benefitted connections between Linux OS to esp32 devices too; they shortened the connection duration.

projectgus, 2019-03-08:
This additional sleep behaviour was merged as the `esp32r0_delay` option, which happens automatically if the initial connections fail:
https://github.com/espressif/esptool/blob/master/esptool.py#L462

An explanation and a link to the same issue 136 is here:
https://github.com/espressif/esptool/blob/master/esptool.py#L398

esptool tries a "normal" connection first then switches over to the "esp32 r0 workaround".

This reason this isn't done always is that it adds a minimum of 1.6 seconds to the sync time, and is not necessary on ESP8266 at all or on ESP32 silicon revision 1, or on ESP32 revision 0 boards with enough capacitance on the EN line. I've also never heard of it helping on a non-Windows platform, although you're saying something different here.

> esp32 DevKit-V1 

Do you have a photo of this board? Is it ESP32 silicon revision 0? (esptool.py will print the silicon revision when it syncs).

sunbearc22, 2019-03-08:
> Do you have a photo of this board? Is it ESP32 silicon revision 0? (esptool.py will print the silicon revision when it syncs).

I am testing my script using a ESP32 DEVKITV1 that has a ESP32D0WDQ6(revision1). Looks the same as [device shown in this site](https://github.com/playelek/pinout-doit-32devkitv1).

> I have used the solution by @MartyMacGyver mentioned in Issue #136 to resolve my connection issue. After replacing esptool.py line 425 with time.sleep(2.0) # Sleep longer and line 432 with time.sleep(0.5) # Sleep longer, the above mentioned connection issue was resolved. That is,

With regards to this observation that I had earlier made, I think it was just fortuitous that there was an improvement. On Windows 10, connection with the DevkitV1 worked well for about an hour of testing. Thereafter, it just could not connect. Puzzling?????? The most reliable solution that I arrived at so far is to manually hold down the BOOT button on the DevKit1 and run `ESPLoader.connect()` method, or   while `ESPLoader.connect()` method is running to manually hold down the BOOT button until connection is established. I have hard coded this solution in my GUI app. See [esp32flashwriter_v4_2.py](https://github.com/sunbearc22/ESP32FlashWriter/blob/master/esp32flashwriter_v4_2.py) which works on both Linux and Window10.

Don't know why Window 10 is having such a connection issue. hmm....

sunbearc22, 2019-03-08:
> ESP32 revision 1 can't use this workaround as it relies on a silicon bug.

Can you point me to the literature explaining [line 403-404](https://github.com/espressif/esptool/blob/master/esptool.py#L403-L404)? What is this "silicon bug"?  At least now I know why the "esp32 r0 workaround" dont' connect Win10 to ESP32 DEVKITV1 well.

projectgus, 2019-03-11:
So the board is "DOIT ESP32 DEVKITV1"? Including the manufacturer in the name is useful. :)

Try adding a capacitor between the EN and GND pins of your dev board. At least 100nF is good, up to 10uF should be fine. A lot of third party dev board manufacturers seem to leave this off and it makes automatic reset unreliable. Without this capacitor, parasitic capacitance can make a difference which may be why it randomly starts/stops working.

The workaround is to hold down the BOOT button, as you're doing now.

Different OSes have different timing around USB interactions, so the automatic reset can behave differently depending on OS and even specific host hardware.

> Can you point me to the literature explaining line 403-404? What is this "silicon bug"?

See [ECO and workarounds for bugs in ESP32](https://www.espressif.com/en/support/download/documents) item V3.1. Basically, the r0 silicon resets twice so we get two chances for it to pick up the correct bootstrapping mode.

sunbearc22, 2019-03-11:
> So the board is "DOIT ESP32 DEVKITV1"? Including the manufacturer in the name is useful. :)

On the back of the board, it prints "ESP32 DEVKITV1" w/o the "DOIT" label. 
Mac: 24:0a:c4:xx:xx:xx,  Manufacturer: ff, Device:fff
Any issue with these?

> Try adding a capacitor between the EN and GND pins of your dev board. 

According to this [pic](https://github.com/playelek/pinout-doit-32devkitv1), you mean to connect pins 2 and 15 (left side) with at least 100nF to 10uF?

> Different OSes have different timing around USB interactions, so the automatic reset can behave differently depending on OS and even specific host hardware.

If so, maybe `esptool.ESPLoader._connect_attempt()` method should include a condition check for platform types and correspondingly implement appropriate `esp32r0_delay` workaround? Or is this something espressif want esptool.py users to do?

projectgus, 2019-03-12:
> prints "ESP32 DEVKITV1" w/o the "DOIT" label. Any difference?
> Mac: 24:0a:c4:xx:xx:xx, Manufacturer: ff, Device:fff
> Any issue with this device?

I guess one issue is that the board manufacturer is bad at labelling their product. :)

The flash manufacturer and device IDs shouldn't be all-FF, but this isn't causing problems with the serial link.

> According to this pic, you mean to connect pins 2 and 15 (left side) with at least 100nF to 10uF?

Yes, looks like it.

> esptool.ESPLoader._connect_attempt() method should include a condition check for platform types and correspondingly implement appropriate esp32r0_delay workaround? Or is this something espressif want esptool.py users to do?

There is no way to 100% solve this problem in software, because of the way the "automatic reset circuit" works and the fact we have to set RTS and DTR separately in two separate commands.

There is a brief window of time after `setDTR(True)` and before `setRTS(False)`, when both pins are asserted (both True) and the "automatic reset circuit" tries to boot the chip into the normal running mode. If the chip comes out of reset during this time window, it will boot into the wrong mode.

All of the timing that happens between setDTR(True) and setRTS(False) is out of esptool.py's control (part is pyserial, part is Python runtime, part is Operating System, part is hardware drivers, part is development board hardware).

Having the capacitor between EN & GND means that the chip takes longer to come out of reset, so the window of time can be longer. This is the only reliable fix.

If you want to know more, there are a ton of details here: https://github.com/espressif/esptool/issues/136

closed 2019-03-12 by projectgus

MartyMacGyver, 2019-03-12:
Funny how this all came full circle to #136. :-D

No, I wouldn't expect parallel processes to help -  even then you're still going to end up single-threading into the same driver/chipset to command to RTS and DTR.

I'd make the hardware mod @projectgus suggested and be done with it. That's the effective solution.

projectgus, 2019-03-12:
> Funny how this all came full circle

The part I find funny is that it's over 2.5 years since Espressif (and reputable dev board vendors) started adding this $0.02 part to their ESP32 boards to fix the problem, and we updated the reference designs, but there are still bargain basement dev board vendors who haven't changed anything...

Maybe everyone makes bug reports to esptool.py and noone tells the vendors to fix their board designs... :laughing:

MartyMacGyver, 2019-03-12:
Or they'd be playing a futile game of "contact the vendor"... sadly, most vendors aren't responsive at all, or the true vendor is one or more steps removed from the vendor. 😞 

My original devices were direct from Espressif though, so I knew who to bug about... bugs. 😄

sunbearc22, 2019-03-12:
I am sure I would have gotten a better device if I was more knowledgeable. Unfortunately, I am a newbie and the market seems to be filled with them. :sweat: Still, I greatly appreciate your advices. :smile: 

For the same device, I noticed that the voltage performance at the EN and GND Pins when connected to Linux can be far superior than Windows 10. I have [evidenced my observations here](https://github.com/sunbearc22/ESP32FlashWriter/blob/master/reason_for_win10_poor_connectivity_with%20esp32_devkitv1.pdf) to provide a basis to ask you if the issue I am facing is also related to the [CP210x USB to UART Bridge Virtual COM Port (VCP) drivers for Winows 10](https://www.silabs.com/products/development-tools/software/usb-to-uart-bridge-vcp-drivers) ? 

Sorry to have taken up your bandwidth on this older board that is using the esp32 chip.

projectgus, 2019-03-12:
Hi @sunbearc22 ,

It's not your fault some boards are poorly designed, I didn't mean to blame you for reporting the issue - apologies.

> For the same device, I noticed that the voltage performance at the EN and GND Pins when connected to Linux can be far superior than what Windows 10 gives. I have evidenced my observations here 

I am not sure what you're seeing here, but I think it's probably to do with different hardware? Is the USB power cleaner from the Linux computer, or grounding relative to the scope is different, or something like this?

sunbearc22, 2019-03-13:
> I am not sure what you're seeing here, but I think it's probably to do with different hardware? Is the USB power cleaner from the Linux computer, or grounding relative to the scope is different, or something like this?

Discovered that clean voltage signals for the Linux + ESP32DevKitV1 combination could be obtained w/ and w/o connecting the ground on the scope's probe to the GND of the ESP32 board. However, the latter practice cannot be used to measure the voltage signal for the Windows 10 + ESP32DevKitV1 combination. **I had to ensure that the ground on the scope probe is connected to the GND of the ESP32 board.** Doing so irradicated all the failings that I had observed earlier. That said, the connectivity issue between Windows 10 and ESP32DevKitV1 still persists. 

Given that the ESP32DevKitV1 has this design flaw that can be fixed by connecting the EN pin and the GND pin with a capitor, why is this solution only needed when Windows OS is used and not when Linux OS is used?  :confused: Thinking aloud here....

projectgus, 2019-03-13:
To get clean signals on an oscilloscope, you always need to put the ground clip close as possible to the thing you're probing. If you happened to get a clean signal with the ground clip anywhere other than GND on the ESP32 board then this is due to luck, or some electrical property of the thing you're probing (ie the computer's USB port).

> Given that the ESP32DevKitV1 has this design flaw that can be fixed by connecting the EN pin and the GND pin with a capitor, why is this solution only needed when Windows OS is used and not when Linux OS is used? 

Because Windows is a different operating system it has different OS-level timing and serial APIs, the drivers have different timing, the USB stack has different timing. All of these can result in small changes to the timing that USB commands packets are sent to the CP2102, which results in small changes to the timing that the RTS & DTR pins change levels. See comments and linked discussion above for more details, it's all explained there.

sunbearc22, 2019-03-13:
Thank you. 

I would like to measure the DTR voltages, which is given by GPIO0 pin. However, I dont' seem to be able to locate it in [it's layout](https://github.com/playelek/pinout-doit-32devkitv1). Can you tell me which pin it is? I did probe the "RTC_GPIO0" at Left Pin 14, but I did not see the DTR voltages as set by `esptool.ESPLoader._connect_attempt()`.

projectgus, 2019-03-13:
RTC_GPIO0 != GPIO0

GPIO0 doesn't seem to be on a dev board pin. Consult the ESP-WROOM32 datasheet or the schematics at the link you posted to find which pin on the ESP32 module is GPIO0.

DTR & RTS are pins on the CP2102 serial chip.

DTR & RTS are not wired directly to GPIO0 & EN, there is an automatic reset circuit in between. Some high level description is here: https://github.com/espressif/esptool/wiki/ESP8266-Boot-Mode-Selection#automatic-bootloader . The circuit is shown in the schematic of the page you linked to.

sunbearc22, 2019-03-13:
After testing all the board's GPIO Pins, I initially thought Pins GPIO1 (i.e. U0RXD) and GPIO3 (i.e U0TXD) were likely suspects. I found GPIO14  (i.e. EMAC_TXD2) and GPIO15( i.e. EMAC_RXD3) yielded similar signal to EN but noisier. 

I also came across [ESP32 Pin List ](http://www.smartarduino.com/esp32/esp32_chip_pin_list_en.pdf). Found on 
page 2 that: _U0* is for the signals of UART0 module_
page 6 that a mention of _Signal No.: 16, Output Signals: U0DTR_out,  Output enable of output
signals: 1'd1_,
page 9 a mention of "_Debugging Log on U0TXD During Booting_". I suspect it is saying that U0TXD will toggle True during ESP32 booting. Probably, this explains why GPIO1 toggles to LOW and HIGH after each time EN returns to HIGH from a LOW.
 
[image]

Concluding, I think the ESP32DeviKitV1 board with 30 pins do not have a pin to point to ESP32 GPIO0. However, a ESP32DeviKitV1 board with 36 pins may have such a pin.

sunbearc22, 2019-03-13:
Referring to the schematics at regions 3B, 2C, and 4C of [esp32_Schematic Prints.pdf](https://www.dropbox.com/s/jefwxxtufgwg0ex/esp32_Schematic%20Prints.pdf?dl=0), I see GPIO0 (i.e. pin 25 of the ESP32 chip) as a connection to the chip's internal flash and the DTR pin of the CP2102 module, which is UART0 of the ESP32DevKitV1 board. GPIO0 appears not to be accessible via any of the GPIO pins of the ESP32DevKitV1 board.... except through the USB...?. Don't think I can probe the GPIO0 voltages. Is this correct?

projectgus, 2019-03-13:
Yes, the schematic shows GPIO0 is pin 25 of the ESP32-WROOM module (the big metal can - not the "chip's internal flash".)

It is possible to probe GPIO0, but you have to probe the pad on the ESP32-WROOM module (or one of the other connections on the circuit board) as it's not on any of your development board's external pins. This is what I was trying to indicate in the previous comment.

This discussion has strayed a long way from any issue in esptool.py. If you want to get more electronics diagnostics help with this project then it may be worth starting a thread over on https://esp32.com/

sunbearc22, 2019-03-14:
[image]
The satisfaction of finally being able to see `esptool.ESPLoader._connect_attempt()` in its entirety. :smile:. Hope this image can also help other users appreciate `esptool.py`.

Thank you again. :thumbsup: