espressif/esptool #418: A fatal error occurred: Timed out waiting for packet header

Darkblade48, 2019-03-08:
* Operating system: Windows 10 Enterprise LTSC Version 1809
* Python version: 2.7.15
* ESP hardware in use: ESP8266(EX)

# Full esptool.py command line as run:
 `esptool.py --port COM13 --baud 115200 write_flash -fs 1MB -fm dout 0x0 sonoff.bin`

# Full output from esptool.py (please copy and paste all lines of output)
```
esptool.py v2.6
Serial port COM13
Connecting........_____....._____....._
Detecting chip type... ESP8266
Chip is ESP8266EX
Features: WiFi
MAC: 68:XX:XX:XX:XX:XX
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Compressed 534032 bytes to 365310...
Writing at 0x00008000... (13 %)
A fatal error occurred: Timed out waiting for packet header
```

# What is the expected behaviour?
Successful flash

# Do you have any other information from investigating this?
I have tested with CH340G and also a PL2303 USB TTL Serial adapter. Both result in the same errors.

Using write_flash at various baud rates (tested 9600, 19200, 38400, 57600, 115200, and leaving it blank) results in the same error, always at the same 0x00008000 address, after varying times (slower baud rates take longer to show the error). 

A complete log with the --trace flag is here:
[https://pastebin.com/3CFKFULQ](https://pastebin.com/3CFKFULQ)

Running `esptool.py --port COM13 erase_flash` results in the following:
```
esptool.py v2.6
Serial port COM13
Connecting....
Detecting chip type... ESP8266
Chip is ESP8266EX
Features: WiFi
MAC: 68:XX:XX:XX:XX:XX
Uploading stub...
Running stub...
Stub running...
Erasing flash (this may take a while)...

A fatal error occurred: Timed out waiting for packet content
```

Running `esptool.py flash_id` results in the following:
```
esptool.py v2.6
Found 1 serial ports
Serial port COM13
Connecting....
Detecting chip type... ESP8266
Chip is ESP8266EX
Features: WiFi
MAC: 68:XX:XX:XX:XX:XX
Uploading stub...
Running stub...
Stub running...
Manufacturer: ef
Device: 4016
Detected flash size: 4MB
Hard resetting via RTS pin...
```

Running `esptool.py chip_id` results in the following:
```
esptool.py v2.6
Found 1 serial ports
Serial port COM13
Connecting....
Detecting chip type... ESP8266
Chip is ESP8266EX
Features: WiFi
MAC: 68:XX:XX:XX:XX:XX
Uploading stub...
Running stub...
Stub running...
Chip ID: 0x08cd7baa
Hard resetting via RTS pin...
```

Running `esptool.py --port COM13 read_flash 0x0 0x400000 backout.txt` results in the following:
```
esptool.py v2.6
Serial port COM13
Connecting....
Detecting chip type... ESP8266
Chip is ESP8266EX
Features: WiFi
MAC: 68:XX:XX:XX:XX:XX
Uploading stub...
Running stub...
Stub running...

A fatal error occurred: Corrupt data, expected 0x1000 bytes but received 0xfff bytes
```

projectgus, 2019-03-11:
Hi @Darkblade48 ,

> ESP hardware in use: ESP8266(EX) 

Exactly what hardware are you flashing, a Sonoff or something else?

If you're using a bare module on a breadboard, you probably need to add more capacitance to the module itself - a lot of vendors seem to ship modules that don't integrate enough power capacitance and it causes brownouts during flashing and during Wi-Fi operation as well.

If your setup is something unusual, do you have a photo of it?

Darkblade48, 2019-03-11:
> Hi @Darkblade48 ,
>
>
> Exactly what hardware are you flashing, a Sonoff or something else?
>
> If you're using a bare module on a breadboard, you probably need to add more capacitance to the module itself [...]

Oops, I guess I forgot to mention that in my opening post. It is a Sonoff S31, nothing too unusual.

projectgus, 2019-03-11:
> It is a Sonoff S31

Was the Sonoff working before you went to flash it?

IF yes, my guess is that the 3.3V power supply either can't supply enough current, or there's a resistive drop in the cables you're using for power. Try using the shortest cables with the most copper (thickest copper section) for the 3.3V and GND connections in particular. A lot of cheap breadboard wires only have a tiny amount of copper, so it may be 3.3V at the power supply end and much lower at the ESP8266 end when it's drawing current.

Darkblade48, 2019-03-11:
I didn't think to test the Sonoff before flashing it, since I've flashed several S20s in the past without any problems (using the same setup!).

The only other thing I can think of at this point is sagging 3.3V or simply a bad Sonoff?

projectgus, 2019-03-13:
If the same setup works for other S20s then a bad device does seem likely, unless it's some small hardware revision change that means less capacitance or higher power draw in this one.

Darkblade48, 2019-03-14:
Oh, that is certainly most disappointing. Is there anyway to trouble shoot capacitance problems (or more directly, is there anyway to add capacitance to a particular pin to prevent brownouts)?

I checked the spec sheet for the 3.3V power supply I'm using, and it's supposed to be able to output a maximum of 700 mA. I could try changing to thicker wires and give that a shot to troubleshoot potential current issues.

projectgus, 2019-03-14:
Yes, thicker/shorter wires are probably worth a try. You could try soldering an additional capacitor between VCC & GND on the board. (Remember it'll need to still be safe once mains voltage is reattached.)

You could also try `esptool.py -b 9600 --no-stub -p PORT write_flash ...` and see if this helps. This flashes slowly and with compression disabled, so flashing will be less "bursty" on power.

Angus

Darkblade48, 2019-03-14:
Strangely enough, when I run `esptool.py --port COM13 --baud 9600 --no-stub write_flash -fs 1MB -fm dout 0x0 sonoff.bin` I end up with the following error:

```
esptool.py v2.6
Serial port COM13
Connecting........_____....._____....._____....._____....._____....._____....._____

A fatal error occurred: Failed to connect to Espressif device: Timed out waiting for packet header
```

Now interestingly, when I change the baud rate to 19200 and run the following: `esptool.py --port COM13 --baud 19200 --no-stub write_flash -fs 1MB -fm dout 0x0 sonoff.bin` it seems to detect the chip fine, but then times out with the following error:

```
esptool.py v2.6
Serial port COM13
Connecting........_____....._____....._____.....
Detecting chip type... ESP8266
Chip is ESP8266EX
Features: WiFi
MAC: 68:XX:XX:XX:XX:XX
Enabling default SPI flash mode...
Configuring flash size...
Erasing flash...

A fatal error occurred: Timed out waiting for packet header
```

I haven't tried the hardware tweaks yet, but with regards to the capacitor between VCC & GND, I take it that I'm supposed to short the two together, with a capacitor? What voltage/capacitance would be good? I think I have some spare electrolytic 50V 1 uF caps lying around, but they're not exactly ideal as decoupling capacitors, I would assume.

projectgus, 2019-03-14:
You could soldering the 50V 1uF capacitor between VCC and GND, can't hurt. Ideally get a ceramic cap with a voltage rating at least 6V, and the highest capacitance you can find.

(If soldering electrolytics, make sure to check the polarity - electrolytics only work one way and will go "pop" otherwise.)

Darkblade48, 2019-03-17:
I tried putting a 50V 10 uF capacitor between VCC and GND, and tried `esptool.py --port COM13 --baud 115200 write_flash -fs 1MB -fm dout 0x0 sonoff.bin` and `esptool.py --port COM13 --baud 19200 --no-stub write_flash -fs 1MB -fm dout 0x0 sonoff.bin` to the same error messages :(

At this point, I'm really at a lost...if it's hardware related, is there any actual way I can test? I can't even seem to load any other precompiled BIN file.

Darkblade48, 2019-03-24:
So an update, I also tried a CP2102 USB to serial converter, and still have the same problems as detailed above...

Does anyone else have any clue/ideas?

projectgus, 2019-03-25:
@Darkblade48 Sorry, I can't think of any recommendations beyond checking as much of the hardware as you can.

I'm going to close this because I don't think it's an issue with esptool.py. If you do find some other way to flash the device which works, but esptool.py does not, then please let me know and I'll reopen.

Best of luck flashing your Sonoff.

closed 2019-03-25 by projectgus

juan98mg, 2019-05-22:
@Darkblade48 Hello, hey friend I figured out the solution. http://arduinoamuete.blogspot.com/2015/12/reiniciar-nodemcu-en-windows.html on this link you can see the solve problem it worked for me, just remember while is started flash you need to press down flash button and keep it down until the program finish. and then your board is gonna work. I promisse that it will work if you have an esp8266 board

Darkblade48, 2019-05-23:
@juan98mg Thank you for the link. I will give it a shot when I have some time and let you know if it worked

tusharm16, 2019-07-01:
Hey @Darkblade48 : Any luck? I am stuck with the same problem as well. Tried everything and losing hope.

projectgus, 2019-07-01:
Anyone who is failing to flash a sonoff might find this comment useful: https://github.com/espressif/esptool/issues/432#issuecomment-506675955

(I can't explain why this is necessary for sonoff hardware, but seems like it helped at least one person.)

Darkblade48, 2019-07-01:
@tusharm16 Unfortunately, I haven't had the time to look at the solution that @juan98mg posted, but it didn't seem like it would solve my problem.

There was a blog article that someone wrote where they had mentioned that it might be due to a bad USBTTL chip, hence my trying with with multiple chips (CH340G, PL2303, CP2102), but they all return the same error.

jiapei100, 2019-08-23:
Here the same:

```
longervision-GT72-6QE% esptool.py --port /dev/ttyUSB0 erase_flash                                                                     
esptool.py v2.8-dev
Serial port /dev/ttyUSB0
Connecting........_____....._____....._____....._____....._____....._____....._____

A fatal error occurred: Failed to connect to Espressif device: Timed out waiting for packet header
```

mirospacek, 2019-10-29:
This worked for me at the "A fatal error occurred: Timed out waiting for packet header" .
Hold down the Flash button while downloading from arduino ide. Then release the button after the downloading is done., then unplug the device, then plug it in again. Now the blink.ino program uploaded starts.

AHafsy, 2019-12-16:
Hello,
I'm working on ESP8266 and I had the same error that is reported before: "A fatal error occured: Timed out waiting for packet header". I've tried all the proposition in order to solve this problem but it doesn't help me an I still have it. 
When I run esptool.py --port /dev/ttyUSB0 flash_id, I got the chipid and some other information without any connection problem. Taht's why I don't understand why when I try to flash the bunary files on the board the fatal error occurs.
Can anyone help me to solve this problem please ?

projectgus, 2019-12-16:
Hi @AHafsy,

Suggest opening a new issue. Include the command and full output that it working, and the command and the full output that is not working. Answer all the questions that the issue template asks for, and give as much detail as you can.

AHafsy, 2019-12-17:
Hi @projectgus,

You find here [https://github.com/espressif/esptool/issues/493#issue-538954922](url) the new issue.

Thank you!

SharathN25, 2020-03-11:
Hello guys, I have faced the similar issue with ESP32 DEVKIT V1. Just press the "boot" button the dev kit when it shows "connecting.....____.....". This will resolve it.

grodrm00, 2020-03-28:
Hi All,

I have the same problem... I have several days trying, reading all the forums and I still have the same errors. I have check the connections, tried several Baudrates, 

I have an arduino Mega2560 board.
My WIFI-Module is ESP8266-01

The connections I had made are following this tutorial.
https://programarfacil.com/podcast/como-configurar-esp01-wifi-esp8266/

I think the problem is in the bootloader, I have read carefully this article and try all it is said
https://github.com/espressif/esptool/#entering-the-bootloader

Can anyone help me with this?
Thanks you in advance

projectgus, 2020-03-29:
@grodrm00 Using an Arduino board's integrated USB/serial converter seems to be unreliable in a lot of situations. If you can get a dedicated USB/serial adapter board then it's more likely to succeed.

Otherwise, I'd recommend double checking all of the wiring shown in the page you linked - in particular that the buttons are correctly pulling the right signals to GND, and that the RESET<->GND link on the Arduino Mega is in place.

rohan-kekatpure, 2020-08-30:
I'm working with a Nodemcu ESP8266 board trying to flash MicroPython binary. The instructions at the back of my board said to use a Baud Rate of 9600, which gave me the "Timeout" error. Leaving out the `--baud` argument solved it.

MohammadsM, 2022-02-09:
> @Darkblade48 Hello, hey friend I figured out the solution. http://arduinoamuete.blogspot.com/2015/12/reiniciar-nodemcu-en-windows.html on this link you can see the solve problem it worked for me, [...]

I had the same issue with "NodeMCU devkit v1.0". 
[image]

But the problem has solved with:
1- Pressing down the `flash` button.
2- Press and release the `reset` button, while `flash` button is pressed.
3- Execute `make flash monitor` again.

Thank you for the tip!

g1ra, 2023-10-04:
I had a same problem and nothing helped me . Solution was : The board in the past , ONCE was connected to a 5VOLT serial adapter . This burned out the ESP8266EX chip (ESP07) . 
Once I did replace the ESP8266EX with another new module everything worked well .