espressif/esptool #419: blk3 contents sequence change

YuchiRoo, 2019-03-11:
V2.6 version change the blk3 contents sequence 
BLK3                   Variable Block 3                                  
  = 00 00 00 00 00 00 00 42 31 42 32 43 45 33 4a 33 44 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 R/W 

V2.5 and below it's look like:

BLK3                   Variable Block 3                                  
  = 00 00 00 00 00 00 00 44  33 4a  33 45 43 32 42 31 42  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 R/W 

why would you change the sequence?

projectgus, 2019-03-11:
@YuchiRoo yes, this is mentioned in the release notes:
https://github.com/espressif/esptool/releases/tag/v2.6

The commit, with some explanation, is here:
https://github.com/espressif/esptool/commit/fde8dd2bcf6a081bdb6cce78a76a917568088968

> Reformat espefuse.py summary output (includes: read-only efuses now display value ? not 0, byte order of keys shown for BLK1,BLK2 are now shown in software order not the reversed byte order used by hardware cryptographical modules.)

The reason is that the old ordering only makes sense when storing data for the hardware cryptographic modules, and usually when using these modules the efuse block is made read-protected.

However, I don't think that the order should be exactly what you give above. If you have this in V2.5:
```
= 00 00 00 00 00 00 00 44 33 4a 33 45 43 32 42 31 42 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 R/W
```
(This is last word of efuse first, with MSB of each word printed first.)

Then you should see this in V2.6:
```
= 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 42 31 42 32 43 45 33 4a 33 44 00 00 00 00 00 00 00 R/W
```
(This is first word of efuse first, with LSB of each efuse printed first - ie same layout as software sees in the registers, if they were copied to a byte array)

If you see something different, it may be a bug.

`espefuse.py dump` output should be the same between both versions (it prints each 4 bytes of efuse as a word.)

YuchiRoo, 2019-03-11:
yes, we write some information into blk3, when we update the esptool, we found our manufacture tools work fail.

closed 2019-03-15 by YuchiRoo