espressif/esptool #421: /dev/ttyUSB0 failed to connect

GeorgeFlorian, 2019-04-02:
* **Operating system:** Linux Mint 19.1
* **Python version:** Python 2.7.15rc1
* **ESP hardware in use:** ESP Wroom-32

## Issue
I couldn't upload any sketch to the ESP32 and it began to run very hot. I thought that by erasing the flash I will make it work again.

## Full esptool.py command line as run:
 `esptool.py erase_flash`

## Full output from esptool.py :
```
esptool.py v2.6
Found 2 serial ports
Serial port /dev/ttyUSB0
Connecting........_____....._____....._____....._____....._____....._____....._____
/dev/ttyUSB0 failed to connect: Failed to connect to Espressif device: Timed out waiting for packet header
Serial port /dev/ttyS0
Connecting........_____....._____....._____....._____....._____....._____....._____
/dev/ttyS0 failed to connect: Failed to connect to Espressif device: Timed out waiting for packet header

A fatal error occurred: All of the 2 available serial ports could not connect to a Espressif device.
```
## What is the expected behaviour?
Shouldn't it erase the flash ? I kept the boot button pressed, I tried with it unpressed. I've always uploaded sketches while the button was pressed.

## Do you have any other information from investigating this?

Here is the whole story !
https://github.com/espressif/arduino-esp32/issues/2630

richard-scott, 2019-04-02:
@GeorgeFlorian - Do you have D0 connected to RST to enable **ESP.deepSleep**?

I had the same issues you describe after I erase_flash'd a unit. I had to disconnect my jumper cable on the RST/D0 pins to get the flashing to work. After the initial flash after erase_flash I could reconnect my cable and future flashing of firmware worked just fine.

GeorgeFlorian, 2019-04-03:
> @GeorgeFlorian - Do you have D0 connected to RST to enable **ESP.deepSleep**?
>
> I had the same issues you describe after I erase_flash'd a unit. I had to disconnect my jumper cable on the RST/D0 pins [...]

I have nothing connected to the ESP32 right now. It's just the USB Cable.

GeorgeFlorian, 2019-04-08:
Nothing ? No one ?

richard-scott, 2019-04-08:
Has it ever worked?
Do you have permissions to read/write /dev/ttyUSB0?
You can add them with this:
```
$ sudo usermod -a -G dialout $USER
```
Log out and back in again to activate the new permissions.

GeorgeFlorian, 2019-04-08:
> Has it ever worked?
> Do you have permissions to read/write /dev/ttyUSB0?
> You can add them with this:
> 
> ```
> $ sudo usermod -a -G dialout $USER
> ```
> Log out and back in again to activate the new permissions.

Tried that, did not work. Thanks !

Of course it worked ! I uploaded tons of code to it using Arduino IDE and PlatformIO IDE. And the port was always /dev/ttyUSB0.

projectgus, 2019-04-10:
Hi @GeorgeFlorian,

The error message seems to indicate that the ESP32 is not responding to attempts to reset it into bootloader mode.

* You haven't said exactly what dev board you have, but if it has a "BOOT" button then try holding this down while flashing.
* Otherwise, if you open a serial terminal at 115200bps and reset your ESP32 (press and release "EN" button if it has it), do you see some output? [Some reference info here](https://github.com/espressif/esptool/wiki/ESP32-Boot-Mode-Selection#boot-mode-message). You can run `miniterm.py /dev/ttyUSB0 115200` to open a terminal on the command line (use Ctrl-] to exit).
* If you don't see any output when resetting the ESP32, the hardware is faulty in some way.
* There is a general list of troubleshooting tips to go through here, also: https://github.com/espressif/esptool/#troubleshooting

projectgus, 2019-04-10:
OK I just saw you posted boot messages and a schematic on the linked issue:

```
 rst:0x10 (RTCWDT_RTC_RESET),boot:0x33 (SPI_FAST_FLASH_BOOT)
 flash read err, 1000
 ets_main.c 371 
 ets Jun  8 2016 00:22:57
```

There's an explanation of this message at the at the link above about boot messages.

If you pull GPIO0 to ground, you should be able to have the chip reset into UART bootloader mode instead and it will talk to esptool.py.

GeorgeFlorian, 2019-04-10:
> OK I just saw you posted boot messages and a schematic on the linked issue:
>
> ```
>  rst:0x10 (RTCWDT_RTC_RESET),boot:0x33 (SPI_FAST_FLASH_BOOT)
>  flash read err, 1000
>  ets_main.c 371
>  ets Jun  8 2016 [...]

The board is a ESP-Wroom-32 DevKitV1
https://i1.wp.com/randomnerdtutorials.com/wp-content/uploads/2018/08/ESP32-DOIT-DEVKIT-V1-Board-Pinout-30-GPIOs-Copy.png?w=966&ssl=1

I have the 30 pins board and it has no GPIO0.

I've always pressed BOOT when uploaded code.

projectgus, 2019-04-11:
> I have the 30 pins board and it has no GPIO0.
> I've tried grounding GPIO12 and then connected it to the 3V3 pin, but to no avail.
>
> I've always pressed BOOT when uploaded code.

If you hold the BOOT button and look at the serial console output, what do you get?

You can check for bit 0x10 set in the `boot:0xNN` message to check if the chip is reading GPIO0 as high or low. Full list of bits here: https://github.com/espressif/esptool/wiki/ESP32-Boot-Mode-Selection#boot-mode-message

If BOOT button is held down but value 0x10 is set in the boot:0xNN message, your development board is faulty and the button is not working. You'll need to either repair the board or get a new one.

GeorgeFlorian, 2019-04-11:
> If you hold the BOOT button and look at the serial console output, what do you get?
>
> You can check for bit 0x10 set in the `boot:0xNN` message to check if the chip is reading GPIO0 as high or low. [...]

How do I check those ? I've tried using esptool just once, and I find it very difficult to use.

projectgus, 2019-04-12:
> How do I check those ?

Use a regular serial terminal program to look at the serial console output, similar to whatever you used to get the previous boot log captures you posted to the other issue.

projectgus, 2019-05-28:
Hi @GeorgeFlorian ,

Did you make any progress investigating this problem?

Angus

GeorgeFlorian, 2019-05-29:
> Hi @GeorgeFlorian ,
> 
> Did you make any progress investigating this problem?
> 
> Angus

Unfortunately, no ! I threw it away and bought a new one, because it was taking too long to fix it and I wanted to go on with my projects. 
Thank you for your help !

closed 2019-05-29 by GeorgeFlorian