espressif/esptool #422: esptool write_flash: error: argument <address> <filename>: Address "dout" must be a number

ubuntupunk, 2019-04-10:
(Please delete any lines which don't apply.)

* Operating system: Ubuntu 16.04

* Python version: (`python -V` to check this)
Python 2.7.12
* ESP hardware in use:
NodeEMU devkit

# Full esptool.py command line as run:
/home/<user>/esptool/esptool.py --port /dev/ttyUSB1 --baud 115200 write_flash -fm qio dout 0x00000
 
# Full output from esptool.py (please copy and paste all lines of output)
esptool write_flash: error: argument <address> <filename>: Address "dout" must be a number
# What is the expected behaviour?
doesn't appear to be related to any known problem

# Do you have any other information from investigating this?
no
# Is there any other information you can think of which will help us reproduce this problem?
Arduino 1.89

projectgus, 2019-04-11:
`qio` and `dout` are two different flash modes (Quad I/O and Dual Output). You need to pass one or the other, not both.

After this option, you need an address and then a filename. The filename in your case seems to be missing.

You need one of these two command lines:

```
esptool.py --port /dev/ttyUSB1 --baud 115200 write_flash -fm qio 0x0 filename.bin
```
OR
```
esptool.py --port /dev/ttyUSB1 --baud 115200 write_flash -fm dout 0x0 filename.bin
```

For more information, consult the esptool.py README: https://github.com/espressif/esptool#write-binary-data-to-flash-write_flash

closed 2019-04-11 by projectgus

ubuntupunk, 2019-04-17:
home/afrodeity/esptool/esptool.py --port /dev/ttyUSB1 --baud 115200 write_flash dout 0x00000 /tmp/arduino_build_834877/NodeMCU_hello_world_.ino.bin 
usage: esptool write_flash [-h] [--erase-all]
                           [--flash_freq {keep,40m,26m,20m,80m}]
                           [--flash_mode {keep,qio,qout,dio,dout}]
                           [--flash_size FLASH_SIZE]
                           [--spi-connection SPI_CONNECTION] [--no-progress]
                           [--verify] [--encrypt]
                           [--ignore-flash-encryption-efuse-setting]
                           [--compress | --no-compress]
                           <address> <filename> [<address> <filename> ...]
esptool write_flash: error: argument <address> <filename>: Address "dout" must be a number
esptool write_flash: error: argument <address> <filename>: Address "dout" must be a number

projectgus, 2019-04-17:
Add `-fm ` before `dout`, ie `-fm dout`.

For reference look at the command line in previous comment I posted.