espressif/esptool #424: A fatal error occurred: Timed out waiting for packet header

badbyt3, 2019-04-16:
* Operating system: Ubuntu 2019
* Python version: 2.7.15
* ESP hardware in use: Sonoff S20
# Full esptool.py command line as run: esptool.py --port /dev/ttyUSB0 write_flash -fm dout -fs 128m -ff 40m 0x00000 ./bin/0x00000.bin 0x10000 ./bin/0x10000.bin

# Full output from esptool.py (please copy and paste all lines of output)
esptool.py v1.2-dev
Connecting...
Running Cesanta flasher stub...
Flash params set to 0x0390
Writing 32768 @ 0x0... 
A fatal error occurred: Timed out waiting for packet header

# What is the expected behaviour? I have upgraded the flash with winbond w25q128fv it is pin compatible, yet I get the above mentioned error message.

projectgus, 2019-04-17:
Hi @badbyt3 ,

* The version of esptool.py you have is quite old, it may be worth trying the current version.
* What is the output of `esptool.py --port PORT flash_id`?

If the flash_id values are garbage (all 0xFF or all 0x00) it means the flash chip is not responding to the ESP8266. This would be a hardware issue not a compatibility issue (either the flash chip is not connected correctly, there is a soldering issue, or the flash chip is dead/fake/etc).

badbyt3, 2019-04-17:
The outout of `esptool.py --port /dev/ttyUSB0 flash_id` is:
esptool.py v1.2-dev
Connecting...

A fatal error occurred: Timed out waiting for packet header

projectgus, 2019-04-17:
@badbyt3 It looks like the  ESP8266 didn't reset into bootloader mode this time. It did the first time (Connecting succeeded), so you must have done something differently.

More tips here: https://github.com/espressif/esptool#Troubleshooting

badbyt3, 2019-04-17:
You may be right. Is there a way how to flash directly the chip?

projectgus, 2019-04-17:
> You may be right. Is there a way how to flash directly the chip?

You can use a SPI serial programmer device to directly flash the chip, if you have one.

Note that if esptool.py can't talk to the flash chip via the ESP8266, the ESP8266 also won't be able to boot from this flash chip - as the same SPI connection and commands are used for both things.

projectgus, 2019-05-28:
@badbyt3 Did you have any luck getting output from the `flash_id` command?

projectgus, 2019-11-28:
Closing due to lack of updates.

closed 2019-11-28 by projectgus